openscad/openscad #1495: Difference with highlighted object makes cut faces transparent

thehans, 2015-11-19:
When subtracting highlighted geometries:  X-#Y, the new faces cut into X are given the color of Y.  This ends up putting transparent faces on otherwise solid colored objects if Y is highlighted.

After some discussion in IRC kintel suggests a solution of changing this line:
https://github.com/openscad/openscad/blob/master/src/OpenCSGRenderer.cc#L110
from
`if (j_obj.flag & CSGTerm::FLAG_HIGHLIGHT) {`
to
`if (highlight && j_obj.flag & CSGTerm::FLAG_HIGHLIGHT) {`

This will makes it so the faces cut by highlighted objects stay the same color as they would be when not highlighted.

This will change the expected test results for all colorscheme logo tests and a couple other highlight related tests.

Before
[image]
After
[image]

One side effect seems to be that faces cut into a highlighted object (#X - Y) will be drawn first with non-highlighted color, then drawn over with highlighted.

thehans, 2015-11-19:
One case that this still doesn't address is X-Y where Y is set to alpha < 1 using the color function and not highlight.  This will still make a transparent face in a solid colored object.  I'm thinking it would be good to also force alpha=1 when drawing cutouts.

This can be done by inserting 
`if (colormode == COLORMODE_CUTOUT) basecol[3] = 1;`
before line 77 here:
https://github.com/openscad/openscad/blob/master/src/renderer.cc#L77

nophead, 2015-11-20:
If the cut face is normally coloured by the object making the cut it seems
logical it should be made transparent if it is cut by a transparent object.
However it has always seemed a little odd to me that cut faces are anything
different from the solid they are part off.

thehans, 2015-11-20:
Ideally I think it should work this way, for a difference X-Y
if Y is colored then the cut faces get the color of Y, but the alpha of X
else if only X is colored the cut faces get the color of X, and the alpha of X
else the cut faces get default cut color from colorscheme

Highlighting Y should make Y transparent and red, but not affect the color of faces it cuts.

I'm trying some things to see if I can make such a scheme work.  The problem is when Y cuts more than one object of different color.  Say you have objects  A and B.  Y is in a position that would cut into both of them.  OpenCSG will render (A - Y), drawing to the depth buffer, then we fill in a color for Y on all faces that match that depthbuffer.  Then OpenCSG draws the depths for (B-Y) into the same buffer, and we fill in a possibly different color for Y this time.  The problem is it will still match the depths that were put in for the render pass on object A, so those color get overwritten.

Here is a test script I've been playing with.
```
$fn=100;
$vpr = [80, 0, sin(360*$t) * 75];
//echo($vpr[0]);
difference() {
  union() {
    // red
    color([1,0,0,1]) translate([-20,0]) cube(10, center=true);
    // green
    color([0,1,0,1]) translate([-7.5,0]) cube(10, center=true);
    // blue
    color([0,0,1,0.5]) translate([7.5,0]) cube(10, center=true);
    // default
    translate([20,0]) cube(10, center=true);
  }
  translate([0,-5,3]) rotate([0,90,0]) cylinder(r=1,h=60,center=true);
  color([0,1,1,0]) translate([0,-5,1]) rotate([0,90,0]) cylinder(r=1,h=60,center=true);
  #translate([0,-5,-1]) rotate([0,90,0]) cylinder(r=1,h=60,center=true);
  #color([1,0,1,1]) translate([0,-5,-3]) rotate([0,90,0]) cylinder(r=1,h=60,center=true);
}
```

I've altered the color assigning code to match what I talked about above.  You can see that the last cube rendered uses the default colorscheme.  So the top cuts in the other cubes end up with that color also. 
[image]

If we clear the depth buffer for each call to OpenCSG, then we can put a different color for each cut.  The problem with that is then objects no longer occlude eachother.
[image]

If OpenCSG can be called in such a way that it renders to two different depth buffers, one cumulative(as it is now), and one that gets cleared every pass(we would use this one to shade our colors to), then I think this could theoretically work for all cases.

thehans, 2015-12-04:
Making a note for reference.  Discussion of color of subtracted faces relevant to  #501.

kintel, 2015-12-20:
Fixed by #1498

closed 2015-12-20 by kintel