openscad/openscad #1496: Transparent objects render faces out of order

thehans, 2015-11-19:
When viewing a transparent object, the preview has artifacts due to the faces being drawn in uncontrolled order.

This shows up as a strange seam in cylinders for example:
[image]

It looks much better if the back faces are drawn before the front faces.  This can be accomplished with `glEnable(GL_CULL_FACE);` and `glCullFace` calls.  Drawing the object with the front culled first, then again with the rear culled.
[image]

The downside to this is that it could possibly make preview rendering slower.  I haven't found any noticeable slowdowns based on my subjective interactive testing, but if there are any suggested performance tests, I would be willing to try them.

To mitigate some of this potential slowdown the rear faces can be only drawn when color alpha < 1.  And only draw the front faces when when not rendering cutouts.

kintel, 2015-11-20:
I'm slightly concerned about performance degradation.
Otoh, rendering lots of transparent objects probably isn't a major workflow.

thehans, 2017-12-04:
Fixed by #1497

closed 2017-12-04 by thehans