openscad/openscad #1502: No ambient light in preview when edges shown

thehans, 2015-11-22:
I noticed that when edges are shown, all the faces look a little darker, and under some orientations go completely black.  The edges use a shader while no edge rendering uses the fixed function pipeline.  The difference is that the fixed function version has ambient lighting enabled.  The default ambient shading component in this mode is (0.2,0.2,0.2,1.0)

So, not sure this is worth a pull request for a one line change, but 

Changing this line in the vertex shader code:
https://github.com/openscad/openscad/blob/master/src/GLView.cc#L275
`      "  shading = abs(dot(normal, lightDir));\n"`
to 
`      "  shading = 0.2 + abs(dot(normal, lightDir));\n"`

Makes the faces match exactly between the two modes.

t-paul, 2015-11-25:
We could try https://help.github.com/articles/editing-files-in-another-user-s-repository/ :-).

kintel, 2015-11-25:
cool, I didn't know about that feature.

kintel, 2015-12-05:
@thehans Could you give an example where this happens? I'm not able to casually reproduce this issue..

thehans, 2015-12-05:
Its specifically in Preview mode with "Show Edges" checked.  If you toggle Show edges on and off you can see the shading change.  
[image]

kintel, 2015-12-05:
ah, I somehow thought you meant the shading of the wireframe. thx!

kintel, 2015-12-20:
Fixed by #1509

closed 2015-12-20 by kintel