openscad/openscad #1504: Add way to compute model volume

Harvie, 2015-11-24:
Since OpenSCAD is meant for making solid manifold models, i guess there is way to compute volume of such model. I am not sure how hard it's to compute volume of complex objects (like boolean intersection of sphere, cube and cone), but there's probably at least some numeric way for doing this. It doesn't have to be perfect (i know that eg. circle in OpenSCAD is in fact the polygon, so it can't be 100% accurate). But such feature can be quite handy if you will manage to achieve some level of accuracy suitable for practical uses.

Everybody knows how to calculate volume of cube, but calculating more complex shapes can be tricky and i think that OpenSCAD is really perfect tool for such case.

There should be two volumes calculated...
1.) volume of used material (cavities/bubbles buried inside model are subtracted from volume = can be used for calculating weight of model if you know density of used material).

2.) volume of model ignoring the cavities/bubbles = exact volume of water that will be displaced when fully submerged (water will not fill internal cavities).

DrDub, 2015-12-01:
Stochastic computation of volumes via sampling should be doable and it is not that different that solving numerically the closed forms for the shapes.

That'll take care of (1). (2) might need a different method, maybe similar to raytracing.

nophead, 2015-12-01:
IIRC it can be calculated exactly for a convex polyhedron by tessellating the facets with triangles and joining each triangle vertex to an arbitrary point in the centre to form many tetrahedra, for which there is a formula for the volume. A concave polyhedron can be decomposed into multiple convex ones.

kintel, 2015-12-01:
@nophead Your first solution also works for concave polyhedrons. If the faces are planar, tessellating them isn't even necessary.

MichaelPFrey, 2018-03-01:
related to/duplicate of #433

jordanbrown0, 2023-11-18:
Duplicate of #433.

closed 2023-11-18 by jordanbrown0