openscad/openscad #1507: Debian / Ubuntu instructions need updating on website

WZ9V, 2015-12-01:
The current documented process for Debian/Ubuntu install did not work on Linux Mint (Ubuntu).  These steps installed 2015.03

$ sudo add-apt-repository -y ppa:openscad/releases

$ sudo apt-get update

$ sudo apt-get install openscad

t-paul, 2015-12-01:
What exactly did not work? The latest update 2015.03-2 was just released a couple of hours ago, so the PPA still has 2015.03-1. Packaging needs some time, but the updates should hopefully be available soon.

WZ9V, 2015-12-01:
When I did what was on the web I got the 2014 version, default for Mint is the 2013 version.  Following the steps above I got the original 2015.03 version. I didn't expect the newer 2015.03-02 at this time.

t-paul, 2015-12-01:
Ok, so it's about the need for `apt update`? I guess we can add this and also link to the Debian documentation.

WZ9V, 2015-12-01:
Yes, might just be a Mint 17.2 thing because some of its stuff is horribly out of date.  I don't recall it happening on Ubuntu Mate 15.10 when I tried it.  The extra step probably wouldn't hurt though since I think it just refreshes the database before the install.

osj1961, 2016-01-09:
I just followed the web instructions (on ubuntu 15.04) and got the 2014 version.  Then saw this issue and tried again with the "apt-get update" before the install and that got me the March 2015 version.

WZ9V, 2016-01-09:
The maintainer of the PPA in Ubuntu has not updated it since 2015.03-1 so that is the latest packaged version.

https://launchpad.net/~openscad/+archive/ubuntu/releases

closed 2016-09-18 by t-paul