openscad/openscad #1510: Segfault on initialization (glewInit())

Doeme, 2015-12-06:
Hi

I'm experiencing a rather odd segfault.

Im operating on x86_64 Gentoo Linux, with openscad 754cb0f75079098b16de94b5605bd00dc1c56f7c (I've encountered the error with every release until openscad-2013.01, where I stopped trying. I'm sure that at least one of these had worked sometime in the past.)

## glxinfo ##
```
OpenGL vendor string: nouveau
OpenGL renderer string: Gallium 0.4 on NVE7
OpenGL core profile version string: 4.1 (Core Profile) Mesa 11.0.6
OpenGL core profile shading language version string: 4.10
OpenGL core profile context flags: (none)
```
## Libraries/Tools ##
* media-gfx/opencsg-1.3.2-r1
* media-libs/glew-1.10.0-r2
* dev-qt/qtcore-4.8.6-r2
* sys-devel/gcc-4.9.3
* sci-mathematics/cgal-4.3
* dev-libs/boost-1.56.0-r1
* dev-libs/mpfr-3.1.3_p4
* x11-libs/libX11-1.6.3
* virtual/glu-9.0-r1
* virtual/opengl-7.0-r1
* dev-libs/gmp-6.0.0a
* dev-qt/qtgui-4.8.6-r4
* dev-qt/qtopengl-4.8.6-r1
* sci-libs/mpfi-1.5.1
* dev-cpp/eigen-3.1.3
* x11-libs/qscintilla-2.8.4-r1

## Stacktrace ##
```
#0  0x0000039bb76fa78f in XQueryExtension () from /usr/lib64/libX11.so.6
#1  0x0000039bb76e9846 in XInitExtension () from /usr/lib64/libX11.so.6
#2  0x0000039bb7d06c45 in ?? () from /usr/lib64/libGL.so.1
#3  0x0000039bb7d00df4 in glXQueryVersion () from /usr/lib64/libGL.so.1
#4  0x0000039bbb5c30d5 in glxewContextInit () from /usr/lib64/libGLEW.so.1.10
#5  0x0000001a33cf9c05 in QGLView::initializeGL (this=0x1a384ba660) at src/QGLView.cc:104
#6  0x0000039bb9649c37 in QGLWidget::glDraw() () from /usr/lib64/qt4/libQtOpenGL.so.4
#7  0x0000039bb81930b9 in QMetaObject::activate(QObject*, QMetaObject const*, int, void**) ()
   from /usr/lib64/qt4/libQtCore.so.4
#8  0x0000001a3422924f in Preferences::requestRedraw (this=0x1a38721e40)
    at objects/moc_Preferences.cpp:236
#9  0x0000001a33cdeef8 in Preferences::apply (this=0x1a38721e40) at src/Preferences.cc:739
#10 0x0000001a33e616ab in MainWindow::MainWindow (this=0x1a384a9db0, filename=...) at src/mainwin.cc:468
#11 0x0000001a33e40df4 in gui (inputFiles=std::vector of length 1, capacity 1 = {...}, 
    original_path=..., argc=1, argv=0x3df9fc99988) at src/openscad.cc:757
#12 0x0000001a33e42d5f in main (argc=1, argv=0x3df9fc99988) at src/openscad.cc:941
```

```
(gdb) frame 5
#5  0x0000001a33cf9c05 in QGLView::initializeGL (this=0x1a384ba660) at src/QGLView.cc:104
104	  GLenum err = glewInit();
```

([There](https://github.com/openscad/openscad/blob/master/src/QGLView.cc#L104))

I first thought the problem might be that `glewInit()` gets called twice, but the breakpoint only hits once.

I think it might be suspicious for `glewInit()` getting called from the `Preferences` class, but I don't know the normal programflow.

GilesBathgate, 2015-12-06:
Preferences.cc emits the signal ```requestRedraw()``` which in tern is connected to qglview slot ```UpdateGL()```. UpdateGL should just mark the view needs to be refreshed, and so ```PaintGL()``` will get called by the event loop. ```glewInit()``` seems to only be called in ```InitialiseGL()```. Its certainly very odd, maybe a race condition. What happens if you remove/comment out this line of code: http://git.io/vRVPA

Doeme, 2015-12-06:
Still segfaults, this time in 
```
#0  0x00000379f141778f in XQueryExtension () from /usr/lib64/libX11.so.6
#1  0x00000379f1406846 in XInitExtension () from /usr/lib64/libX11.so.6
#2  0x00000379f1a23c45 in ?? () from /usr/lib64/libGL.so.1
#3  0x00000379f1a1ddf4 in glXQueryVersion () from /usr/lib64/libGL.so.1
#4  0x00000379f52e00d5 in glxewContextInit () from /usr/lib64/libGLEW.so.1.10
#5  0x0000007158895c05 in QGLView::initializeGL (this=0x715d253130) at src/QGLView.cc:104
#6  0x00000379f3366c37 in QGLWidget::glDraw() () from /usr/lib64/qt4/libQtOpenGL.so.4
#7  0x0000007158a10e73 in MainWindow::setColorScheme (this=0x715d242a90, scheme=...)
    at src/mainwin.cc:2720
#8  0x00000071589fd700 in MainWindow::MainWindow (this=0x715d242a90, filename=...) at src/mainwin.cc:471
#9  0x00000071589dcdf4 in gui (inputFiles=std::vector of length 1, capacity 1 = {...}, 
    original_path=..., argc=1, argv=0x39c1b790728) at src/openscad.cc:757
#10 0x00000071589ded5f in main (argc=1, argv=0x39c1b790728) at src/openscad.cc:941
```

GilesBathgate, 2015-12-06:
I think that rules out ```preferences.cc``` causing the problem.

Doeme, 2015-12-06:
Nope, I haven't. I'm already running the free nouveau-drivers.
Also, forcing the software rasterizer with `LIBGL_ALWAYS_SOFTWARE=y ./openscad` (which I think/hope should rule out nouveau being at fault) didn't change the behaviour.

Also, I'd rather not install the horrid binary blob from NVIDIA, but I'll try to downgrade mesa. 
Maybe this will help.

Doeme, 2015-12-06:
```
OpenGL vendor string: nouveau
OpenGL renderer string: Gallium 0.4 on NVE7
OpenGL core profile version string: 3.3 (Core Profile) Mesa 10.6.9
OpenGL core profile shading language version string: 3.30
```

Still has the exact same behaviour.
Next I'll try it on a different machine with a similar setup, but I'll postpone that until tomorrow.

GilesBathgate, 2015-12-06:
Good, the command was to purge nvidia, it caused me troubles in the past.

GilesBathgate, 2015-12-06:
Could it be the same as this: http://sourceforge.net/p/avogadro/bugs/654/

"This bug can be reproduced if qt-gui and qt-opengl compiled with egl support. Without egl support avogadro starts and works very well."

Doeme, 2015-12-06:
Might very well be. qt-gui and qt-opengl are compiled with the egl-flag enabled.
I'll try and recompile them without.

Doeme, 2015-12-06:
Ok, the answer is yes, it is the same problem.
Thank you very much for digging that out!

I do wonder where that problem arises...

GilesBathgate, 2015-12-07:
That makes sense, if "egl" means OpenGL ES, then I think its a known issue that GLES doesn't play well with glew.

t-paul, 2016-09-18:
Closing as this is not really an OpenSCAD issue. Note that GLEW 2.0 is supposed to support EGL, so maybe that opens up some more possibilities soon.

closed 2016-09-18 by t-paul