openscad/openscad #1511: Add update number to version shown in GUI window status bar

MichaelAtOz, 2015-12-06:
Just now I was testing between 2015.03-1 & 2015.03-2 and noticed that the -1 & -2 was not shown.
It is in Help/About.

t-paul, 2015-12-06:
I think that's intentional... @kintel ?

MichaelAtOz, 2015-12-06:
well it makes it hard when testing...

kintel, 2015-12-06:
It should show in help-about. But perhaps we didn't add it to the window title?

MichaelAtOz, 2015-12-06:
I mean bottom right
[image]

I suppose you could add it to the About window title, but that is a bit redundent.

kintel, 2015-12-06:
Right, I left it out of the bottom right to avoid showing too much info to people. 2015.03 would define the feature set, the patchlevel release should just be about bugfixes.

closed 2018-04-30 by MichaelAtOz