openscad/openscad #1512: Alternate construction failed. CGAL error in CGAL_Nef_polyhedron3() - with scale() & surface()

MichaelAtOz, 2015-12-06:
Chasing down this [forum item](http://forum.openscad.org/Problem-in-openscad-script-from-Thingiverse-td14953.html), the problem is in 2015.03 (-1 & -2), but not in 2014.03 or 2013.06.

There the issue shows as:
```
PolySet has nonplanar faces. Attempting alternate construction
Geometries in cache: 19
...
   Volumes:         5
WARNING: Object may not be a valid 2-manifold and may need repair! 
Rendering finished.
WARNING: Object may not be a valid 2-manifold and may need repair! See http://en.wikibooks.org/wiki/OpenSCAD_User_Manual/STL_Import_and_Export 
```

I cut the code down to:
```
image_file = "cp.dat"; // [image_surface:100x100]

$fa = 6;
$fs = 1.5;
image_width = 40;
image_length = 40;

emboss_height = 1.5;
handle_height = emboss_height+2.5;

difference () {
	cube([110, 110, handle_height]);
	   translate([50,50,handle_height])
			scale([(image_width)/100,(image_length)/100,emboss_height])
				rotate([0,180,0])
					surface(file=image_file, center=true, convexity=5);
}
```
(cp.dat is linked at the forum - but it happens with other .dat files too)

Which gets the following error:
```
ERROR: Alternate construction failed. CGAL error in CGAL_Nef_polyhedron3(): 
CGAL ERROR: assertion violation! 
Expr: e->incident_sface() != SFace_const_handle() 
File: /opt/mxe/usr/x86_64-w64-mingw32.static/include/CGAL/Nef_S2/SM_const_decorator.h 
Line: 326 
```
remove the scale() and it doesn't error.

At first I thought it may be due to the long decimals in the .dat file (e.g. 0.433333333333 -0.103921568627), but it errors with my [Flag on Moon .dat](http://www.thingiverse.com/thing:78719/#files) which only has 2 decimals.

That's as far as I can go.

MichaelAtOz, 2016-01-07:
Changing it to `translate([50,50,handle_height+0.01])` removes the coincident face an the alternate construction.

hcatak15, 2019-02-25:
Hello ı want to merge 2 stl s. but couldnt.
it seems with f5  but not f6

$fn=150;
module move() {
  scale([0.5,0.5,0.5])
import("astronot2.stl",convexity=10);
  };

union(){
    
    rotate([-50,0,90]) translate([2,-19,20])scale([0.095,0.095,0.57])import("bahcelogo4.stl",convexity=3);
    move();
};
astronot2.stl
bahcelogo4.stl

MichaelAtOz, 2019-02-25:
Unfortunately the astronot2.stl file has problems.
If you just union it with a `cube(1);` (below is an implicit union), that causes F6 to always call CGAL
```
import("astronot2.stl",convexity=10);
translate([0,0,-10]) cube(1);
```
So with F6 you get 
```
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /opt/mxe/usr/x86_64-w64-mingw32.static/include/CGAL/Nef_S2/SM_const_decorator.h Line: 326 
```
CGAL is checking the geometry and found a problem with a surface.
It has holes in the mesh, the yellow bits.
[image]

kintel, 2024-03-26:
Status as of 2024-03-24:
* The original issue is fixed, both with CGAL and Manifold
* The astronaut issue is a bad input file and still fails with both CGAL and Manifold.

-> closing

closed 2024-03-26 by kintel