openscad/openscad #1513: list comprehensions - fully evaluated when subset requested.

GilesBathgate, 2015-12-08:
The following code locks my machine:

```c#
list = [ for(x=[1:999999]) for(y=[1:999999]) x*y ];
echo(list[5]);
```
When I need element 5 of the list (denoted ```list[5]```) then surely this should be really quick, since if I understand correctly the list comprehension is a declarative description of 'what' the list contains. Not an imperative description of 'how' to generate a list with those elements.

It seems as though what openscad is doing is imperatively making &asymp; 1 × 10<sup>12</sup> calculations, and putting the results in a giant list. Which causes the machine to lock up and run out of memory.

I rather expected that instead, the index operation would 'zip through' the for loops until the index was equal to 5, and then do a single calculation of x*y, and use it. Alternatively it could interrogate the declaration as if it were an enumerable function that yielded the result of the calculation, and keep doing so until the function had been called 6 times, and use the result.

> As a side note: this is the only way one can currently access a subset of a given list comprehension, are there any plans to allow list slicing such as ```list[3:9]```, of course one could always do ```[ for(i=[3:9]) list[i] ]``` but that would also be really slow, due to the above implementation.

nophead, 2015-12-08:
The fact it says list = []; means make an instance of a list.  And the list comprehension describes the contents to be huge. I don't see how you could interpret it otherwise.

It gives a memory error in Python. Are there any languages clever enough to only generate the elements in a list on demand?

GilesBathgate, 2015-12-08:
@nophead Well I tried this also:

```c#
echo([ for(x=[1:999999]) for(y=[1:999999]) x*y ][5]);
```

You are right about python, but that is a hybrid language not purely declarative. I understand that the equivalent in haskell would generate the elements on demand.

You say that ``list = [];``` means make it a list, but the way I see it ```list = [ for | let | if ];``` is a list comprehension, its not the same type of thing as a list. If there were such a thing as a typeof operator in openscad I would expect ```typeof([1,2,3])``` to return 'list' and ```typeof([ for(i=[0:1]) i ])``` to return something else.

I am not familiar with how its implemented behind the scenes, but I kind of expected that behind the scenes openscad would treat the declaration as a way of assigning an enumerable function to a value, semantically equivalent to:

```c#
function list() = for(x=[1:999999]) for(y=[1:999999]) yeild x*y;
```

GilesBathgate, 2015-12-08:
@nophead Actually I think you can do it in python:

```python
for i in range(0, 20):
```
vs
```python
for i in xrange(0, 20):
```

I think the later evaluates the range lazily and is thus faster.

nophead, 2015-12-08:
In Python using xrange() means it doesn't need to make the range into a list to give to the for loop to iterate through. It gets round that fact that for in Python doesn't handle ranges directly. The for loop still attempts to do all the iterations and make the massive list. xrange just avoids making a list of 1:999999 first.

I am not sure being purely declarative implies lazy evaluation. You have declared a big list and that is what you get.

GilesBathgate, 2015-12-08:
@nophead What led me to the python example was this: https://en.wikipedia.org/wiki/Generator_(computer_programming)

That's what I meant when I was talking about enumerable functions (which is .NET parlance). Either way pythons xrange() function returns what they call a 'sequence type', whereas range() returns a list.

I suppose what I was getting at really that the following gives:

```python
>>> for i in range(1,10000000000000000):
...   if i<10:
...     print i
...   else:
...     break
... 
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
MemoryError
```

Whereas with xrange() we get:
```python
>>> for i in xrange(1,10000000000000000):
...   if i<10:
...     print i
...   else:
...     break
... 
1
2
3
4
5
6
7
8
9
```

Its not to dissimilar, to the original points I was trying to make. I didn't mean to imply that purely declarative implies lazy evaluation. 

If you declare that "a is an even number for all integers", you don't expect that when asked to get the 100th integer that someone would write down all possible even integers and then give you the 100th one... perhaps you do ;)

nophead, 2015-12-08:
Yes xrange avoids making a list but I don't see how that helps, other than getting around Python's for loop not handling a range directly. I don't see how it helps you make a lazy list, other than a stand in for a list that is a range.

And generators in Python are not lists. You can't slice them or take the length.

What is the problem you are actually trying to solve?

GilesBathgate, 2015-12-08:
@nophead I didn't have a particular use case in which I needed the declaration of a 1000000<sup>2</sup> grid of x*y values. In which I was just going to pull out the 5th one.

I suppose that I just expected that list comprehensions could give me something more than an imperatively generated list of values.

Like you say, I ask for a big list and that is what I get.

closed 2015-12-21 by GilesBathgate