openscad/openscad #1514: Special camera variables should always update

thehans, 2015-12-10:
I would like to create some labels for things that would follow the camera so they are always easily readable.

Here is a simple example:
`rotate($vpr) text("Hello World", halign="center");`

If this is previewed and camera is orbited around, the text doesn't follow.

You have to go into animation mode and setup FPS and steps, and then this trick works.  I think this functionality should be available outside of animation mode as well.

kintel, 2015-12-10:
Due to how this works, we'd have to reevaluate the script for each mouse movement. That's a bit much..

thehans, 2015-12-10:
Well it could be conditionally updated on whether the $vpr variable exists in the script or not.
It just seems like if the variable is in the script then the user would want it to be used.

Also, even in animation mode, since the variable updates are not in sync with the redraw, it makes jerky movement when orbiting.

Another thing is, depending on which variable is included, it only would need to reevaluate for specific operations.
If only $vpr is included, then only reevaluate when orbiting
If only $vpt is included, then only reevaluate when panning
If only $vpd is included, then only reevaluate when zooming

thehans, 2015-12-10:
To further address performance concerns, I would suggest that this functionality could be an option in preferences, disabled by default.

t-paul, 2015-12-11:
I'm not sure it's a good idea to have the labels as part of the actual model. If we could find a way to annotate things where some label should be shown, that could be handled separately when displaying the model without interfering with the model calculation.