openscad/openscad #1515: Broken Polygons

deadsy, 2015-12-11:
/*
The following code *should* create a polygon symmetrical about the x-axis,
and yet it does not. Not in 2015.03. Not in head/master as of 12/10/15.
It's broken in both preview and rendered views.
I have other (larger) polygon test cases that are also broken.
Hopefully someone can take a look at it.
Thanks.
*/

x = [

[0.144661, -0.0251276],
[0.145121, -0.0251826],
[0.146508, -0.0252498],
[0.148822, -0.0251822],
[0.152043, -0.0248333],
[0.156137, -0.0240592],
[0.16105, -0.0227204],
[0.166714, -0.0206835],
[0.173041, -0.0178231],
[0.179933, -0.0140233],
[0.187275, -0.00917936],

[0.187275, 0.00917936],
[0.179933, 0.0140233],
[0.173041, 0.0178231],
[0.166714, 0.0206835],
[0.16105, 0.0227204],
[0.156137, 0.0240592],
[0.152043, 0.0248333],
[0.148822, 0.0251822],
[0.146508, 0.0252498],
[0.145121, 0.0251826],
[0.144661, 0.0251276]

];

polygon(x);

t-paul, 2015-12-11:
Root cause is the same as in #999 - Vertex cleaning in Clipper ([CLEANING_DISTANCE](
https://github.com/openscad/openscad/blob/b152acb9e4743b2a76be1356d3772049c0fa5e87/src/clipper-utils.cc#L54)).

deadsy, 2015-12-11:
Thanks for the fast response.

A couple of thoughts:

 1. If I'm calling polygon directly in the code then the program has no business trimming the vertices I specified. It might warn me about performance, but if I put the vertices there, then I'd like them to stay there.

 2. The vertex trimming is weird. The result doesn't approximate the desired shape.

3. What constitutes close vertices? These vertices are not close in a relative sense within that polygon, but they are close according to the arbitrary units imposed by the program. That's the wrong standard if scaling is to work properly.

 4. This is showing up in an object specified in inch units with an overall size of about 5/16". I shouldn't be hitting any sort of limitation on units accuracy. I'd be happy with 0.0001"

Attached a picture of the test case.
[image]

deadsy, 2015-12-11:
I was shocked the first time I saw a small circle turned into a small triangle, but I could fix that by specifying $fn as a function of curve radius and desired accuracy. With this polygon issue it seems my only response is to scale my dimensions to the point where the program stops messing with my polygons.

So... the principal effect of the CleanPolygon call is to a) cause me problems and b) to make me want to write code that avoids it. Sounds like a bug.

I understand the desire to simplify polygons- but the code doesn't work properly and introduces a dynamic range bug/gotcha that people have to find out about and work around.

Still... I have a path forward, x 1000

kintel, 2015-12-20:
Closing as duplicate of #999. If this use-case differs enough or is interesting to #999, please add a comment with a test case there.

closed 2015-12-20 by kintel