openscad/openscad #1517: Feature Request: mirror across an arbitrary plane

greysAcademicCode, 2015-12-11:
It would be great if the mirror() function could accept an arbitrary plane (that might not pass through the origin) across which the mirror operation takes place!

t-paul, 2015-12-11:
I think that would fit better into a user space library instead of the core language.

greysAcademicCode, 2015-12-11:
May I ask why?
It seems to me like a feature that should be present out of the box for the following reasons:
It can really make the code a lot easier to read (and write) because it prevents the need to throw in annoying offset calculations to move a part after mirroring and all the other CAD suites I'm familiar with have this feature.

nophead, 2015-12-11:
There is no standard representation of plane in OpenScad other than using the normal vector.

A user module can be written to do a pair of translation and use mirror, depending on how one wishes to represent the plane.

greysAcademicCode, 2015-12-11:
Wouldn't it be enough to just add an additional (optional) argument to the mirror() function which defaults to [0,0,0] that defines a point on the plane to be mirrored across?

t-paul, 2015-12-11:
There is no contradiction between "out-of-the-box" and a user-space library, that's basically "just" a detail of how to maintain/ship the feature.

Extending the core functionality means the interface (features used in the *.csg file format) gets bigger. Right now only FreeCAD can read *.csg files, I think. There was a project for BRL-CAD to import this format too, I don't know the state of that. Adding core features means every program reading the files needs to support those. A user space module would resolve to just the core features, adding the required translation and rotation if needed.

nophead, 2015-12-11:
I think it is just a matter for translating to the origin, mirroring and translating back again. Exactly the same as if you want to rotate around an arbitrary point. If you added it to mirror then by symmetry you would need to add it to rotate. Since it is easy to do in user space and can be wrapped up into a module, why make the core language more complicated?

greysAcademicCode, 2015-12-11:
Ah ha! I thought a user-space library meant it wouldn't be distributed with the project. Agreed then; I don't see any reason this should be super low level change.

@t-paul this is off topic here, but since you mentioned it above: from my experience, the FreeCAD .csg import feature seems to be broken currently in many cases. It doesn't seem like there is anyone on the FreeCAD end keeping this working as functions are added to OpenSCAD. For example, `hull()` crashes FreeCAD and `offset()` immediately aborts the import. I've submitted bug reports for these but I'm not hopeful they'll be addressed.

closed 2026-02-11 by github-actions[bot]