openscad/openscad #1519: --camera command line parameter not working on Windows

akaleeroy, 2015-12-14:
Sorry but the `--camera` parameter seems to be ignored completely on my machine.

    openscad.com --camera=0,0,0,55,0,25,680

Windows 7, OpenSCAD 2015.03

The reason I'm interested is that I've figured out that `680` is the distance at which objects appear life-size on my display. Pretty handy!
Maybe in the future it would be cool to add a feature for this? Either try to detect display density & size or offer a way for users to calibrate. Then you could have a button/shortcut labeled "1:1 Life-like" or something like that.

t-paul, 2015-12-14:
Command line options are only effective for non-GUI usage, e.g. to export PNG images.

There's currently no configuration option that allows to specify the default camera for the GUI. I don't think there's a cross platform way to safely detect the actual screen resolution.

The viewport camera variables are writable from the script, for details, see the [manual section about $vpX variables](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Other_Language_Features#.24vpr.2C_.24vpt_and_.24vpd)

akaleeroy, 2015-12-14:
Oh, I see, thanks that explains everything! :+1:

closed 2015-12-20 by kintel