openscad/openscad #1521: difference seems to break preview

tcurdt, 2015-12-16:
I have the following model which consists of a sweep path and two cubes that get subtracted via `difference()`. When those cubes are shown with `%` all looks fine. Simplified:

    difference() {
      sweep(...)
      % translate(...) cube(...)
      % translate(...) cube(...)
    }

[image]

When I remove the `%`

    difference() {
      sweep(...)
      translate(...) cube(...)
      translate(...) cube(...)
    }

the preview is broken:

[image]

It renders fine though!

The code is [here](https://gist.github.com/tcurdt/e90cd374c1ac44af4cf9). I first tried on stable and now are running 2015.12.02 (git 97e9742).

MichaelAtOz, 2015-12-16:
Note that F6 with the cubes removed (via '*' or '%') shows
```
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
PolySet has nonplanar faces. Attempting alternate construction
PolySet has nonplanar faces. Attempting alternate construction
...
```

Neon22, 2015-12-16:
Which sweep are you using ?
Check F12 to see if right side orientation ?

MichaelAtOz, 2015-12-17:
@Neon22 Was that directed at me? Yes F12 shows purple.
I got sweep from https://github.com/openscad/list-comprehension-demos/blob/master/sweep.scad

tcurdt, 2015-12-17:
Interesting. The rotations makes it purple. Basically you can switch the purple by toggling `left`. So it really is the side orientation. I didn't think a rotation in the transforms would cause that.

I guess I could just manually flip the side - but question is (especially if this happens more dynamic) if the sweep should take care of this.

tcurdt, 2015-12-17:
Properly reversing the shape before the sweep solves this issue for me.
But I guess this really is a problem for rotations within a sweep.
Should I open a bug for this in the list-comprehension-demos repo?

closed 2015-12-17 by tcurdt

kintel, 2015-12-18:
@tcurdt This is a weakness of sweep(). We don't have a good way of analyzing your intentions and you need to construct both the polygon and path in the "natural" order (which isn't documented). This is another issue we need to resolve before making sweep() a builtin. Opening an issue on list-comprehension-demos would make sense.