openscad/openscad #1522: OpenSCAD hangs in multi monitor scenario, if both monitors aren't set to the same scaling (Win10)

eddieparker, 2015-12-17:
I have a Dell P2715Q monitor (4K monitor) as my primary monitor, and a ViewSonic VA2246 as my secondary.  In Windows 10, the primary is set up with 150% ("Recommended") scaling, while the Viewsonic is at 1920x1080 with 100% scaling.  If I run OpenSCAD on the main monitor, it's fine.  If I move it to the secondary monitor, close it, and reopen it, it's hung forever.  

Only way to fix it is to turn off scaling, log out, log back in, start up openscad (it'll be fine) and then move it back to the main display.

Is this an issue in OpenScad not handling scaling properly?  Or is this a Windows bug?

kintel, 2015-12-20:
Is this using the latest version of OpenSCAD (2015.03-2) ?

eddieparker, 2015-12-20:
Sorry, I should've specified.  This is OpenSCAD 2015.10.17 (git fba8130).
I've tried it with 2015.03-2 and had similar results.

Notable is that this also occurs with Makerware's software.  Just these two
apps, for some reason -- could be my config, but I don't have another PC to
test it out on.

kintel, 2015-12-20:
Both OpenSCAD and Makerware uses Qt.
Could you try this: Install OpenSCAD-2014.03 from http://files.openscad.org
We switched from Qt4 to Qt5 in OpenSCAD 2015.03, so it would be interesting if Qt5 is the cause of this misbehavior.

Also, could you report this as a Makerware bug? They have at least 10x the number of developers as OpenSCAD - would be interesting to know when or how fast they can fix this.

kintel, 2015-12-20:
oh, and you (or anyone else reading) know of any other Open Source Qt-based 3D apps, that would be helpful for diagnose and debugging such issues in the future.

TakeItAndRun, 2015-12-20:
google" Open Source Qt-based 3D app" found:

https://wiki.manjaro.org/index.php?title=List_of_Qt_Applications

OpenSCAD is not listed there.