openscad/openscad #1523: preview user interaction very different from stable

tcurdt, 2015-12-17:
I am currently running 2015.12.02 (git 97e9742). The interactions with the preview are very different from the stable version - and not in a good way.

1. Doubleclick no longer centers on the object - but zooms out
2. Ctrl-drag moves the camera view - but now in a very unnatural way.
2. Zooming in can make the preview disappear.

[image]

tcurdt, 2015-12-17:
OpenSCAD Version: 2015.12.02 (git 97e9742)
    System information: Mac OS X Version 10.10.5 (Build 14F1509) MacBookAir6,2 2 CPUs 8.00 GB RAM 
    Compiler: Clang "7.0.0 (clang-700.1.76)" 64bit
    Boost version: 1_59
    Eigen version: 3.2.6
    CGAL version, kernels: 4.6.3, Cartesian, Extended_cartesian, Epeck
    OpenCSG version: OpenCSG 1.4.0
    Qt version: 5.5.1
    QScintilla version: 2.8.4
    MingW build: No
    GLib version: 2.46.1
    Application Path: /Applications/OpenSCAD.app/Contents/MacOS
    Documents Path: /Users/tcurdt/Documents
    Resource Path: /Applications/OpenSCAD.app/Contents/Resources
    User Library Path: /Users/tcurdt/Documents/OpenSCAD/libraries
    User Config Path: /Users/tcurdt/Library/Application Support/OpenSCAD
    Backup Path: /Users/tcurdt/Documents/OpenSCAD/backups
    OPENSCADPATH: /Users/tcurdt/Projects/libraries/openscad
    OpenSCAD library path:
    /Users/tcurdt/Projects/libraries/openscad
    /Users/tcurdt/Documents/OpenSCAD/libraries
    /Applications/OpenSCAD.app/Contents/Resources/libraries

    OPENSCAD_FONT_PATH: 
    OpenSCAD font path:
    GLEW version: 1.13.0
    OpenGL Version: 2.1 INTEL-10.6.33
    GL Renderer: Intel HD Graphics 5000 OpenGL Engine
    GL Vendor: Intel Inc.
    RGBA(8888), depth(0), stencil(0)
    GL_ARB_framebuffer_object: yes
    GL_EXT_framebuffer_object: yes
    GL_EXT_packed_depth_stencil: yes

    Using QOpenGLWidget

    GL Extensions:
    GL_APPLE_aux_depth_stencil
    GL_APPLE_client_storage
    GL_APPLE_element_array
    GL_APPLE_fence
    GL_APPLE_float_pixels
    GL_APPLE_flush_buffer_range
    GL_APPLE_flush_render
    GL_APPLE_object_purgeable
    GL_APPLE_packed_pixels
    GL_APPLE_pixel_buffer
    GL_APPLE_rgb_422
    GL_APPLE_row_bytes
    GL_APPLE_specular_vector
    GL_APPLE_texture_range
    GL_APPLE_transform_hint
    GL_APPLE_vertex_array_object
    GL_APPLE_vertex_array_range
    GL_APPLE_vertex_point_size
    GL_APPLE_vertex_program_evaluators
    GL_APPLE_ycbcr_422
    GL_ARB_color_buffer_float
    GL_ARB_depth_buffer_float
    GL_ARB_depth_clamp
    GL_ARB_depth_texture
    GL_ARB_draw_buffers
    GL_ARB_draw_elements_base_vertex
    GL_ARB_draw_instanced
    GL_ARB_fragment_program
    GL_ARB_fragment_program_shadow
    GL_ARB_fragment_shader
    GL_ARB_framebuffer_object
    GL_ARB_framebuffer_sRGB
    GL_ARB_half_float_pixel
    GL_ARB_half_float_vertex
    GL_ARB_instanced_arrays
    GL_ARB_multisample
    GL_ARB_multitexture
    GL_ARB_occlusion_query
    GL_ARB_pixel_buffer_object
    GL_ARB_point_parameters
    GL_ARB_point_sprite
    GL_ARB_provoking_vertex
    GL_ARB_seamless_cube_map
    GL_ARB_shader_objects
    GL_ARB_shader_texture_lod
    GL_ARB_shading_language_100
    GL_ARB_shadow
    GL_ARB_sync
    GL_ARB_texture_border_clamp
    GL_ARB_texture_compression
    GL_ARB_texture_compression_rgtc
    GL_ARB_texture_cube_map
    GL_ARB_texture_env_add
    GL_ARB_texture_env_combine
    GL_ARB_texture_env_crossbar
    GL_ARB_texture_env_dot3
    GL_ARB_texture_float
    GL_ARB_texture_mirrored_repeat
    GL_ARB_texture_non_power_of_two
    GL_ARB_texture_rectangle
    GL_ARB_texture_rg
    GL_ARB_transpose_matrix
    GL_ARB_vertex_array_bgra
    GL_ARB_vertex_blend
    GL_ARB_vertex_buffer_object
    GL_ARB_vertex_program
    GL_ARB_vertex_shader
    GL_ARB_window_pos
    GL_ATI_separate_stencil
    GL_ATI_texture_env_combine3
    GL_ATI_texture_float
    GL_ATI_texture_mirror_once
    GL_EXT_abgr
    GL_EXT_bgra
    GL_EXT_blend_color
    GL_EXT_blend_equation_separate
    GL_EXT_blend_func_separate
    GL_EXT_blend_minmax
    GL_EXT_blend_subtract
    GL_EXT_clip_volume_hint
    GL_EXT_debug_label
    GL_EXT_debug_marker
    GL_EXT_draw_buffers2
    GL_EXT_draw_range_elements
    GL_EXT_fog_coord
    GL_EXT_framebuffer_blit
    GL_EXT_framebuffer_multisample
    GL_EXT_framebuffer_multisample_blit_scaled
    GL_EXT_framebuffer_object
    GL_EXT_framebuffer_sRGB
    GL_EXT_geometry_shader4
    GL_EXT_gpu_program_parameters
    GL_EXT_gpu_shader4
    GL_EXT_multi_draw_arrays
    GL_EXT_packed_depth_stencil
    GL_EXT_packed_float
    GL_EXT_provoking_vertex
    GL_EXT_rescale_normal
    GL_EXT_secondary_color
    GL_EXT_separate_specular_color
    GL_EXT_shadow_funcs
    GL_EXT_stencil_two_side
    GL_EXT_stencil_wrap
    GL_EXT_texture_array
    GL_EXT_texture_compression_dxt1
    GL_EXT_texture_compression_s3tc
    GL_EXT_texture_env_add
    GL_EXT_texture_filter_anisotropic
    GL_EXT_texture_integer
    GL_EXT_texture_lod_bias
    GL_EXT_texture_rectangle
    GL_EXT_texture_sRGB
    GL_EXT_texture_sRGB_decode
    GL_EXT_texture_shared_exponent
    GL_EXT_timer_query
    GL_EXT_transform_feedback
    GL_EXT_vertex_array_bgra
    GL_IBM_rasterpos_clip
    GL_NV_blend_square
    GL_NV_conditional_render
    GL_NV_depth_clamp
    GL_NV_fog_distance
    GL_NV_light_max_exponent
    GL_NV_texgen_reflection
    GL_NV_texture_barrier
    GL_SGIS_generate_mipmap
    GL_SGIS_texture_edge_clamp
    GL_SGIS_texture_lod

kintel, 2015-12-18:
Thanks, I hadn't noticed. This should definitely be looked into.

t-paul, 2016-01-09:
The double click issue is probably caused by https://bugreports.qt.io/browse/QTBUG-43127
I'm getting the same error code 1282 for `glReadPixels()` as mentioned there.

kintel, 2016-01-10:
A less error-prone way of doing this could be to simply pick the CSG tree (ray-intersect all objects and pick the first visible surface). I wrote some code for this in JavaScript (http://codepen.io/kintel/pen/eprzyx) - shouldn't be much harder in C++ :)