openscad/openscad #1524: non-standard folder locations on OSX

tcurdt, 2015-12-17:
The current locations on OSX show:

    Documents Path: /Users/tcurdt/Documents
    User Library Path: /Users/tcurdt/Documents/OpenSCAD/libraries
    User Config Path: /Users/tcurdt/Library/Application Support/OpenSCAD
    Backup Path: /Users/tcurdt/Documents/OpenSCAD/backups

These locations are not very OSX like. Applications should leave the `Documents` folder alone. (Out of all apps I have I think only Adobe and Microsoft don't care.) At least the `Backup Path` and the `User Library Path` I would rather point to somewhere in `Library/Applicaiton Support/OpenSCAD` instead. So I am suggestion to use:

    Documents Path: /Users/tcurdt/Documents
    User Library Path: /Users/tcurdt/Library/Application Support/OpenSCAD/libraries
    User Config Path: /Users/tcurdt/Library/Application Support/OpenSCAD
    Backup Path: /Users/tcurdt/Library/Application Support/OpenSCAD/backups

or maybe

    Documents Path: /Users/tcurdt/Documents
    User Library Path: /Users/tcurdt/.openscad
    User Config Path: /Users/tcurdt/Library/Application Support/OpenSCAD
    Backup Path: /Users/tcurdt/Library/Application Support/OpenSCAD/backups

kintel, 2015-12-18:
The challenge with putting user libraries in _~/Library/Application Support_ is that _~/Library_ is hidden by default on OS X, which poses a usability challenge. So far, we've decided to follow Arduino, which puts user libraries under _~/Documents/_.

I agree that backups could be stored somewhere else.

tcurdt, 2015-12-18:
Arduino is one of the least OSXish apps out there and is not a good reference ;) Plus even Arduino let's you specify the location in the preferences. So my sketches and libraries are somewhere else.

I agree that the hidden _Library_ folder is a bit of a UX problem (that Apple annoyingly introduced in 10.7) but so are all those _~/.foo_ folders or using env variables (like `OPENSCADPATH`) if you follow that line of reasoning.

I guess the question is what the Audience is. Every developer will be fine with _~/.foo_ or _~/Library/foo_ - only a little bit annoyed by `OPENSCADPATH` (because you cannot easily set those for GUI apps on OSX. This helps https://github.com/hschmidt/EnvPane )

If you want to cater for inexperienced then I am out of suggestions because _~/Documents_ is for users not for applications. Which arguably could be OK though - because if backups are out of the way the user would have to create the _~/Documents/OpenSCAD/libraries_ folder for the user libraries. Right now it's just the backup that creates the folder.

I guess what I would do:

1. Move the Backups into `/Users/tcurdt/Library/Application Support/OpenSCAD/backups` (so no auto-creation of the folder in _~/Documents_)
2. Leave the default locations as is
3. Allow modifications of the `User Library Path` and `Documents Path` through either preferences or just NSUserDefaults

.

    defaults write org.openscad.OpenSCAD UserLibraryPath '~/Somewhere/Else'
    defaults write org.openscad.OpenSCAD DocumentsPath '~/Somewhere/Else'

What do you think?

kintel, 2015-12-18:
@tcurdt Your suggestions sound fine to me. One question: Are any of those two keys you suggest standard keys?

tcurdt, 2015-12-18:
@kintel with 'org.openscad.OpenSCAD' (the app's bundle identifier) you define the namespace. Everything in that namespace is up to the application. There is no standardization within that namespace (if that is the question). So using 'UserLibraryPath' and 'DocumentsPath' should be fine.

kintel, 2015-12-18:
In that case, we should probably just access them using QSettings, and we'll get the same functionality for the other platforms for free.

tcurdt, 2015-12-18:
Indeed. Where should it go? "Advanced Settings" or a separate tab?

kintel, 2015-12-18:
If we can access it from the cmd-line, e.g. using _defaults_, that would be a start. GUI can come later. The _Advanced_ preferences tab is already a bit messy, so perhaps we should clean that up. Off the top of my head I sounds like paths and other environment settings do belong in a separate tab.

tcurdt, 2015-12-18:
I am more than fine with just using _defaults_ for now - but that's not cross-platform and not _QSettings_ as you suggested.  Frankly speaking I am not sure how to even use _NSUserDefaults_ from C++. I've only ever used C++ from Obj-C - not the other way around.

kintel, 2015-12-18:
I was thinking perhaps QSettings and defaults write to the same .plist file?
Anyway, see here for how we access such resources using Obj-C in OpenSCAD: https://github.com/openscad/openscad/blob/master/src/PlatformUtils-mac.mm

tcurdt, 2015-12-18:
Really no idea about how QSettings does things under the hood - but I don't see any of the _OpenSCAD_ settings in my _NSUserDefaults_ for _OpenSCAD_

    {
            "3dview.colorscheme" = Cornfield;
            NSNavLastRootDirectory = "~/Desktop";
            NSNavPanelExpandedSizeForOpenMode = "{933, 602}";
            NSNavPanelExpandedSizeForSaveMode = "{712, 576}";
            "NSWindow Frame SUUpdateAlertFrame" = "432 305 851 451 0 0 1440 877 ";
            SUEnableAutomaticChecks = 1;
            SUFeedURL = "http://files.openscad.org/appcast.xml";
            SUHasLaunchedBefore = 1;
            SULastCheckTime = "2015-12-18 19:29:11 +0000";
            "advanced.cgalCacheSize" = 104857600;
            "advanced.enable_opencsg_opengl1x" = 1;
            "advanced.mdi" = 1;
            "advanced.openCSGLimit" = 100000;
            "advanced.polysetCacheSize" = 104857600;
            "advanced.reorderWindows" = 0;
            "design.autoReload" = 1;
            "editor.editortype" = "QScintilla Editor";
            "editor.fontsize" = 12;
            "editor.indentationWidth" = 2;
            "editor.tabWidth" = 2;
            lastOpenDirName = "/Users/tcurdt/Projects/libraries/openscad/tcurdt";
            "launcher.showOnStartup" = 1;
            recentFileList =     (
                "/Users/tcurdt/Projects/libraries/openscad/MCAD/motors/motors.scad"
            );
            "view.hideConsole" = 0;
            "view.hideEditor" = 0;
            "view.hideToolbar" = 0;
            "view.orthogonalProjection" = 0;
            "view.showAxes" = 1;
            "view.showEdges" = 0;
            "view.showScaleProportional" = 0;
            "window.position" = "@Point(21 23)";
            "window.size" = "@Size(1419 845)";
            "window.state" = <000000ff 00000000 fd000000 02000000 00000002 14000003 38fc0200 000001fb 00000014 00650064 00690074 006f0072 0044006f 0063006b 01000000 00000003 38000000 7a00ffff fe000000 03000003 76000000 abfc0100 000001fb 00000016 0063006f 006e0073 006f006c 00650044 006f0063 006b0100 00021500 00037600 00000a00 ffffff00 00037600 00028c00 00000100 00000200 00000100 000002fc 00000000>;
        }

So my theory would be it's not using the same plist.

Given that Obj-C++ is being used already (yay!) implementing access to _NSUserDefaults_ should be quite straightforward. Only question is whether it's worth introducing if we could just cover all platforms by just using _QSettings_.

kintel, 2015-12-18:
I'd say we should avoid writing too much platform-specific code like this, so QSettings is preferable.

episanty, 2017-04-16:
+1. If this is low-hanging, then surely it can be implemented easily?

Particularly re [#1322](https://github.com/openscad/openscad/issues/1322) - the $HOME/Documents is a personal place, and users may or may not want OpenSCAD to set up shop there no matter what. Please allow the user to change the location of this folder.

tcurdt, 2017-04-16:
@episanty I guess if you have the build going the actual changes should probably simple to implement. It just needs someone to sit down and do it.

kintel, 2017-04-16:
We usually use the "Low Hanging" tag to help new contributors to choose issues where the can get their feet wet.

peternlewis, 2017-04-28:
Hi all.  I'm new to OpenSCAD, but very much enjoying it.  I've made a few typo corrections on the documentation and started looking at building it.  I'm been a Mac programmer for 25 years, so if I get involved in OpenSCAD I'd probably like to start by improving the Mac feel (which hopefully also benefits other platforms as well).

@tcurdt’s original suggestion was good.  Putting things in ~/Documents unilaterally is not good behaviour on the Mac, and generally the ~/Library/Application Support/OpenSCAD folder would be the right place.

Adding preferences is rarely the best solution to problems.  In this case, I'd recommend the solution to @kintel’s issue with the ~/Library folder being hidden is to add a menu item in OpenSCAD to open the Config folder (and optionally others for opening the Libraries and Backups folder).  This has the advantage that it would be beneficial to the other platforms as well, making for easy access directly from the application.

kintel, 2017-04-28:
@peternlewis That would make a lot of sense!

peternlewis, 2017-04-28:
Well, I haven't got OpenSCAD compiling yet, but if I do I'm happy to attempt this as the first small change, though I imagine any patch I made might be incomplete since I haven't the foggiest how to open a folder on Windows/Linux.  

But I don't know when I'll get up and running enough to do anything, so if someone else wants to do it that's obviously fine too.

kintel, 2017-05-15:
@peternlewis Anything in particular you're struggling with?

peternlewis, 2017-05-15:
I don't like installing things globally (this is my primary development machine, so I try not to pollute it), so I don't have MacPorts or Homebrew installed, so each bit that doesn't behave requires some investigation. The gmplib web site was offline for a while, so that slowed things down.  Installed a local copy of cmake which took a few goes to make work.  Then pkg-conf, same deal (needed  --with-internal-glib for some reason).  Currently stuck on libtoolize whatever that is.  Ideally it be nice if macosx-build-dependencies.sh figured it all out, but I understand how difficult it is to make this stuff "just work" (is there a reason it couldn't just install cmake and pkg-conf if they aren't already there?).

kintel, 2017-05-15:
`macosx-build-dependencies.sh` only builds run-time dependencies. We recommend using other tools to get a development system up and running.
FYI: Neither Homebrew or MacPorts pollute your global environment. They're both strictly sandboxed to their own install folders.

jottr, 2020-11-10:
I'd like to rekindle the conversation on this issue.

According to the [documentation](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Libraries#Library_Locations) openSCAD libraries that should be globally available must be placed in the `Documents`-folder, i.e.`$HOME/Documents/OpenSCAD/libraries`.

1. This pollutes the users `Documents` folder 
2. Does not follow the commonly used MacOS configuration path, see Apple's official [documentation](https://developer.apple.com/library/archive/documentation/MacOSX/Conceptual/BPFileSystem/Articles/WhereToPutFiles.html)

I would like to suggest that openSCAD loads global libraries in the following order on MacOS:

```
$HOME/Library/Preferences/openSCAD/libraries
$HOME/.local/share/OpenSCAD/libraries # match Linux XDG path for portability
$HOME/Documents/OpenSCAD/libraries # for backwards compatibility
$HOME/.openSCAD/libraries # follow common dotfiles convention
```

tcurdt, 2020-11-10:
`$HOME/Library/Preferences` should be for preferences - not libraries and things like that.
The natural location for those would be in `$HOME/Library/Application Support`.
But really anything is better than `$HOME/Documents`.

peternlewis, 2020-11-11:
Agreed, $HOME/Library/Application Support/OpenSCAD/Libraries/ would be the best location.

Note that Apple is moving to ban applications from accessing the Documents folder except from a user initiated action (such as opening or saving a file). Access to the Documents folder requires specific permissions from the user.

t-paul, 2020-11-11:
If Apple continues banning stuff while being hostile to open source development, we'll have to see how MacOS support can work in the future. The now announced move to ARM will not make things easier in that respect...

jottr, 2020-11-11:
There are millions of x86 Macs out there **right now** that run openSCAD just fine.
This bug is stale since 2015 and no meaningful effort has been made to fix it.

This is an open source project and thus should be inclusive of all users.
This bug should be easy to fix, once we find an agreement on which directories should be parsed in which order.

**Getting back on topic:**

It seems that loading global libraries even from `$HOME/Documents/OpenSCAD/libraries` doesn't work as expected. 
The `.scad`-file needs to be placed on the top level of `libraries` directory or it won't be included by `openSCAD`.

It seems that `openSCAD` does not recurse into the subfolders placed within the `libraries` directory, which forces users to place each libraries main `.scad`-file on the same directory level. 

Example for the [BOLTS](https://github.com/boltsparts/BOLTS) library:

This works:

```
~/Documents/OpenSCAD ❯ tree -L 2
.
└── libraries
    ├── BOLTS.scad
    ├── BOLTS_ANSI.scad
    ├── BOLTS_ASME.scad
    ├── BOLTS_DIN.scad
    ├── BOLTS_DINEN.scad
    ├── BOLTS_DINENISO.scad
    ├── BOLTS_DINISO.scad
    ├── BOLTS_EN.scad
    ├── BOLTS_ISO.scad
    ├── LICENSE
    ├── base
    ├── common
    └── tables

4 directories, 10 files
```

This doesn't:

```
~/Documents/OpenSCAD ❯ tree -L 3
.
└── BOLTS
    └── libraries
        ├── BOLTS.scad
        ├── BOLTS_ANSI.scad
        ├── BOLTS_ASME.scad
        ├── BOLTS_DIN.scad
        ├── BOLTS_DINEN.scad
        ├── BOLTS_DINENISO.scad
        ├── BOLTS_DINISO.scad
        ├── BOLTS_EN.scad
        ├── BOLTS_ISO.scad
        ├── LICENSE
        ├── base
        ├── common
        └── tables

5 directories, 10 files
```

Can someone confirm my observations?

t-paul, 2020-11-11:
We certainly try to provide builds for as much platforms as possible, but If there's no developer with access to MacOS hardware it's getting ever more difficult to support platform specific issues. I'm open to suggestions which could help with that.

Apart from that, it's very helpful if there's a common agreement on what the folder locations should be (while still supporting backward compatibility where needed), thanks for the clarification discussion.

OpenSCAD has currently no support for looking into subfolders, it will only look though the default list and in addition though the path specified in OPENSCADPATH. The list of folders is shown in `HELP`->`Library Info`.

tcurdt, 2020-11-11:
I think we are mixing up a couple of things here.

1. There is also Linux on ARM. I guess having a CI/CD pipeline that also supports ARM is the limiting factor. But frankly speaking that support is overdue. And from what I heard about Rosetta2 I am not too worried.
2. Sandboxed apps need to ask for permission to access files outside their container. This is not new. OpenSCAD is not sandboxed - so no problem. If Apple forces all software to run from a sandbox it becomes a story.
3. I have worked around the folder issue by setting `OPENSCADPATH` - but it doesn't change the fact that this needs fixing.

jordanbrown0, 2020-11-11:
> OpenSCAD has currently no support for looking into subfolders, it will only look though the default list and in addition though the path(s) specified in OPENSCADPATH (unix style list of multiple path entries). The list of folders is shown in HELP->Library Info.

And it **should not** automatically recurse into subfolders.  A big point of a hierarchy is namespace management, so that you can have `foo.scad`, `lib1/foo.scad`, and `lib2/foo.scad`, and they are all independent of one another and you can reliably refer to them without confusion.

peternlewis, 2020-11-12:
> 2. Sandboxed apps need to ask for permission to access files outside their container. This is not new. OpenSCAD is not sandboxed - so no problem. If Apple forces all software to run from a sandbox it becomes a story.

Security & Privacy control over the Documents folder is in addition to sandboxing. It requires an additional user permission even for non-sandboxed applications.

tcurdt, 2020-11-13:
Well, I guess I was mixing up things, too :) 
But in the end asking for the permission once does not really contribute to a pro or con of that location I guess.

I think what is needed for now is the agreement for a different folder search order. 
I my mind something like this would make sense:

```
$OPENSCADPATH
defaults read org.openscad.OpenSCAD LibraryPath
$HOME/Library/Application Support/OpenSCAD
$HOME/.local/share/OpenSCAD
$HOME/.OpenSCAD
$HOME/Documents/OpenSCAD # for backwards compatibility
```

marbles-pi, 2022-10-21:
Still no way to change the documents-path?

tcurdt, 2022-10-21:
I guess the first step would be to agree on the folder search order.
Then it can be implemented.

@t-paul @kintel  Would [this order](https://github.com/openscad/openscad/issues/1524#issuecomment-726979039) be acceptable?

t-paul, 2022-10-23:
Why so many locations? Otherwise the order looks good. That comment covers only the library path, I suppose the backup path should be changed too as you proposed in your [initial post](https://github.com/openscad/openscad/issues/1524#issue-122710093).

When you write `$HOME/Library/Application Support/OpenSCAD` you mean `.../Application Support/OpenSCAD/`**`libraries`**?

tcurdt, 2022-10-23:
Yeah, I am not so keen on all those paths either.
If `defaults` is supported, any advanced user could configure as wanted anyway.

Bringing this all together I see it as this:

    Documents Path: /Users/tcurdt/Documents

	defaults read org.openscad.OpenSCAD DocumentsPath
	$HOME/Documents/OpenSCAD # for backwards compatibility

    User Library Path:

	$OPENSCADPATH
	defaults read org.openscad.OpenSCAD LibraryPath
	$HOME/Library/Application Support/OpenSCAD/libraries
	// $HOME/.local/share/OpenSCAD/libraries # really needed?
	// $HOME/.OpenSCAD/libraries # really needed?
	$HOME/Documents/OpenSCAD/libraries # for backwards compatibility

    User Config Path:

	defaults read org.openscad.OpenSCAD UserConfigPath
	$HOME/Library/Application Support/OpenSCAD

    Backup Path:

	defaults read org.openscad.OpenSCAD BackupPath
	$HOME/Library/Application Support/OpenSCAD/backups
	$HOME/Documents/OpenSCAD/backups # for backwards compatibility

And the paths could all be configured with:

    defaults write org.openscad.OpenSCAD DocumentsPath   '~/Somewhere/Else'
    defaults write org.openscad.OpenSCAD UserLibraryPath '~/Somewhere/Else'
    defaults write org.openscad.OpenSCAD UserConfigPath  '~/Somewhere/Else'
    defaults write org.openscad.OpenSCAD BackupPath      '~/Somewhere/Else'

t-paul, 2022-10-23:
**Documents Path:**
```
$HOME/Documents/OpenSCAD # for backwards compatibility
```
That's not the current value? This would either not change or only get the `defaults` as override?

**User Library Path:**
```
// $HOME/.local/share/OpenSCAD/libraries # really needed?
// $HOME/.OpenSCAD/libraries # really needed?
```
As far as I can tell, those are not supported right now, so I don't see much reason to change that.

**Backup Path:**
```
$HOME/Documents/OpenSCAD/backups # for backwards compatibility
```
It's used for writing, to I don't see how that backward compatibility would work. New versions will just write to the new folder (or the one from `defaults`).

tcurdt, 2022-10-23:
> That's not the current value? This would either not change or only get the defaults as override?

True. Sorry, my comment was misleading.

> As far as I can tell, those are not supported right now, so I don't see much reason to change that.

👍

> It's used for writing, to I don't see how that backward compatibility would work. New versions will just write to the new folder (or the one from defaults).

OpenSCAD could check if the folder exists - but if just moving them is fine I am all for that.

--

So it basically ends being:

    Documents Path:

        defaults read org.openscad.OpenSCAD DocumentsPath || $HOME/Documents/OpenSCAD

    User Library Path:

        $OPENSCADPATH
        defaults read org.openscad.OpenSCAD LibraryPath
        $HOME/Library/Application Support/OpenSCAD/libraries
        $HOME/Documents/OpenSCAD/libraries # for backwards compatibility

    User Config Path:

        defaults read org.openscad.OpenSCAD UserConfigPath || $HOME/Library/Application Support/OpenSCAD

    Backup Path:

        defaults read org.openscad.OpenSCAD BackupPath || $HOME/Library/Application Support/OpenSCAD/backups
    

With the question about the library path:
Search all locations? Or is that also more of an "OR" where one location wins?

t-paul, 2022-10-23:
Yes, library locations need to be all searched in the defined order using the first match. That's how `$OPENSCADPATH` works and so should all the other fixed locations. Having the fallback to the old location last makes sense, we could even think about adding a warning making it deprecated. That way people could get the information to move the files and the fallback could be removed (a number of releases) later.

tcurdt, 2026-09-26:
So - did/can we agree on this? @t-paul @kintel 

Documents Path:

    defaults read org.openscad.OpenSCAD DocumentsPath || $HOME/Documents/OpenSCAD

User Library Path:

    $OPENSCADPATH
    defaults read org.openscad.OpenSCAD LibraryPath
    $HOME/Library/Application Support/OpenSCAD/libraries
    $HOME/Documents/OpenSCAD/libraries # for backwards compatibility

User Config Path:

    defaults read org.openscad.OpenSCAD UserConfigPath || $HOME/Library/Application Support/OpenSCAD

Backup Path:

    defaults read org.openscad.OpenSCAD BackupPath || $HOME/Library/Application Support/OpenSCAD/backups

because then I might try/have to give that a shot.

Why?
Annoyingly macOS27 made it even harder (impossible without a wrapper?) to set `OPENSCADPATH` (or any env vars for GUI apps).