openscad/openscad #1525: fails to render

tcurdt, 2015-12-17:
I have thead build with sweep

[image]

I am not seeing anything particular problematic with it but on render I am getting

    Rendering Polygon Mesh using CGAL...
    PolySet has nonplanar faces. Attempting alternate construction
    PolySet has nonplanar faces. Attempting alternate construction
    ERROR: CGAL error in CGALUtils::applyBinaryOperator union: CGAL ERROR: assertion violation! Expr: itl != it->second.end() File: /Users/kintel/code/OpenSCAD/openscad-2015.03-2/../libraries/install/include/CGAL/Nef_3/SNC_external_structure.h Line: 1102

with stable and 2015.12.02 (git 97e9742).

Any idea what could be causing this?

tcurdt, 2015-12-17:
When the thread does not intersect with itself I am still getting "PolySet has nonplanar faces. Attempting alternate construction" - but at least it renders.

[image]

I still don't get why either is problem though.

nophead, 2015-12-17:
Looks like its constructed from quads. They need to be split into two
triangles to avoid the warning.

tcurdt, 2015-12-17:
@nophead not sure I can follow. It's using [sweep](https://github.com/openscad/list-comprehension-demos/blob/master/sweep.scad) - so it should all be just a polyhedron. No?

nophead, 2015-12-17:
If all the facets passed to polyhedron are triangles you don't get the
warning because a triangle has to be planar. If you use higher polygons all
the points need to be exactly ibn a plane, but that is rarely possible with
floating point representation.

tcurdt, 2015-12-17:
Looks OK to me, @nophead:

[image]

nophead, 2015-12-17:
Yes but that is after the warning that converts non-planar facets to
triangles. I haven't looked at the code but it looks like it constructs the
polyhedron with four point facets (known as quads). If the four points are
not exactly coplanar it will give the warning. If instead each quad is
added to the polyhedron as two triangles it avoid the warning.

tcurdt, 2015-12-17:
Hm - but those edges are from the preview and the message is given on render. Does the preview also turn quads into triangles - without giving a message?

Maybe @kintel could provide some insights on the sweep implementation?

nophead, 2015-12-18:
Actually, maybe it is the ends of the thread that cause the warning. They
seem to be un-triangulated polygons.

And a self intersecting polyhedron will cause problems because it isn't
manifold.

tcurdt, 2015-12-18:
> Actually, maybe it is the ends of the thread that cause the warning. They seem to be un-triangulated polygons.

Hm. How should they look like triangulated? They do consist of triangles.

> And a self intersecting polyhedron will cause problems because it isn't
manifold.

That's good to know. A bummer - but good to know.

nophead, 2015-12-18:
After playing with a simple polyhedron I think the following is true:

* Preview silently triangulates non-planar faces.

* Render triangulates non-planar faces but only sometimes gives the warning about alternate construction. For a simple case like a cube made from squares with one point off it silently triangulates.

Very confusing. Basically non-planar faces have to be triangulated so I don't think it merits a warning.

kintel, 2015-12-18:
Some notes on the above:

* Sweep build quads along the sides and large polygons for the end caps.
* OpenGL cannot render polygons; we have to triangulate. This is done for preview rendering. This is what you see in the wireframe.
* For geometry evaluation, we'd like to preserve the boundary of the original polygons passed to OpenSCAD. To do this, we pass the original polygons to CGAL, in order to partially reconstruct these polygons after performing CSG operations. Unfortunately, quads can be non-planar, especially when using sweep, so we resort to triangulating these on demand (when CGAL complains; hence the warning).
* CGAL doesn't support self-intersecting polyhedrons. It would be interesting to be able to support locally manifold self-intersecting polyhedrons and "union" it with itself, but we'd need another CSG algorithm for that. This is one of the things we'd like to resolve before making sweep() a builtin module, as building such threads is likely to be a common usage of that module.
* To build such threads today, I'd recommend either keeping a gap and adding a cylinder or smth. in the and, or start with a cylinder and remove the thread.

tcurdt, 2015-12-18:
@kintel interesting insights, thanks!

A few questions:
- Couldn't we split those quads into polygons right away in the sweep? Wouldn't that solve part of this?
- Just out of curiosity (probably totally over my head but...) where can I find those CSG algorithms that are being used atm?
- The gap works fine for now - but I am not quite sure what you mean by adding (or removing from) a cylinder. Isn't that a separate (and unrelated) union/difference operation?

kintel, 2015-12-18:
We could split it when building it. The code is here: https://github.com/openscad/list-comprehension-demos/blob/master/sweep.scad#L41

The core CSG stuff is here: https://github.com/openscad/openscad/blob/master/src/cgalutils-applyops.cc#L80 ..although the main code is in the CGAL library (https://github.com/CGAL/cgal)

Yes, by using a cylinder I mean using a separate union/difference operation.

tcurdt, 2015-12-18:
Thanks for the links!

I am adding cylinders when I use the thread. So you are saying a later union/difference operation will make the self-intersection problem go away? Because then it's a total no-problem (for me) - at least in this case.

kintel, 2015-12-18:
It won't make the self-intersection problem go away - you still need to make sure all your polyhedrons are not self-intersecting. It would simply close the gaps you leave (ref. your 3rd screenshot).

tcurdt, 2015-12-19:
OK - that makes more sense then :)
That's also how I am doing it right now.

kintel, 2015-12-20:
I suggest we just keep this around as a feature request for self-intersecting polyhedra, as that could come up again later.

TheBubbleWorks, 2016-02-25:
Hi, 

I'm hitting the same error but using linear_extrude() not smooth().

The attachments is from an only forum post on coils rendering speed, and works ok just using one coil. 
I want to copy and rotate it, which that is when the error occurs.

[Coil_OldEx20 copy.scad.txt](https://github.com/openscad/openscad/files/146442/Coil_OldEx20.copy.scad.txt)

WayneKeenan, 2016-02-25:
Hi, 

I'm hitting the same error but using linear_extrude() not smooth().

The attachments is from an only forum post on coils rendering speed, and works ok just using one coil. 
I want to copy and rotate it, which that is when the error occurs.

[Coil_OldEx20 copy.scad.txt](https://github.com/openscad/openscad/files/146442/Coil_OldEx20.copy.scad.txt)
[Coil_OldEx20 copy.scad.txt](https://github.com/openscad/openscad/files/146444/Coil_OldEx20.copy.scad.txt)

stefano2734, 2026-09-20:
Is this solved or improved after 10 years of development?

jordanbrown0, 2026-09-20:
> Is this solved or improved after 10 years of development?

No.