openscad/openscad #1526: Keyboard shortcut for "View all"/zoom extents

korpx, 2015-12-17:
I am using"View all" (zooming the extents) extensively. A keyboard shortcut would be greatly appreciated!

http://forum.openscad.org/Request-keyboard-shortcut-for-quot-View-all-quot-zoom-extents-td13526.html

kintel, 2015-12-18:
Remember to document here: https://github.com/openscad/openscad/wiki/Keyboard-Shortcuts and/or here: https://github.com/openscad/openscad/wiki/tables-of-keyboard-shortcut

korpx, 2016-01-07:
Suggest Ctrl + ",", placing it on the num pad?

thehans, 2016-01-07:
Something else to consider which I just noticed, the existing shortcuts to zoom in or out don't work either:  Ctrl+[ and Ctrl+] seem to be overridden by Qscintilla shortcuts which appear to go to the previous/next "paragraph" (line with text preceded by whitespace-only lines).

Whichever shortcut is decided, it might be nice to put the view all next to the zoom in and zoom out keys.  Not sure if zoom in/out should be given new shortcuts or the Qscintilla ones should be disabled.  Or some other solution where OpenSCAD pays attention to different shorcuts dependent on whether focus is on editor or view pane.

MichaelAtOz, 2016-01-07:
Doesn't everyone have a scroll wheel (or equiv thing on the side of a touch pad) for scrolling/zoom these days? 

Note re Ctrl+"," is mentioned in Qt doco as shortcut for Preferences on OS/X only (??). Does this do anything for Mac users?

Not all keyboards have num pad these days.

Yes the Ctrl+[&] is listed twice in the above openscad doco links, once as the View/zoom and then in the following section as para up/down. An oversight with the Qscintilla implementation. They work with the standard editor selected.

As I mentioned, I'd save keys "<," and ">." pair for something which goes up/down "< >". Perhaps they could replace Ctrl-[&], if people see paragraph movement as relevant (I don't).

kintel, 2016-01-08:
FYI: Cmd-"," is a standard shortcut for Preferences in OS X apps.

korpx, 2016-01-08:
Okay, maybe not so good then. Other suggestions? CTRL + "." / Command-Period (.) ?

Lenbok, 2016-02-09:
Instead of / as well as a dedicated key binding, it would probably make sense to add a menu accelerator, so that you could, e.g. ALT-v a, in the same way that you can ALT-v t currently.

amarjeetkapoor1, 2016-03-13:
I have added CTRL+ALT+v for view all in pull request #1601 . If someone disagree please share else it could be added as shortcut in Key in openscad.

korpx, 2016-05-14:
Excellent! Workflow is much better. No objections to the key combo?

doug-moen, 2016-05-14:
Unfortunately, ctrl+alt+letter combinations shouldn't be used on Windows
and Linux because they conflict with the AltGr input method for non-English
characters or on international keyboards. We just fixed a related bug where
ctrl+alt+f was being used.

amarjeetkapoor1, 2016-05-14:
But here ctrl+shift+v is used instead of ctrl+alt+v. so, I don't think there would be any conflict.

kintel, 2016-05-15:
This was merged a few weeks ago

closed 2016-05-15 by kintel