openscad/openscad #1528: Crash in lookup() function when array is undefined

txoof, 2015-12-22:
OpenSCAD version 2015.03-2 
Macintosh:
  OS X El Capitan V 10.11.2 (15C50)
  Model Name:	iMac
  Model Identifier:	iMac14,2
  Processor Name:	Intel Core i5
  Processor Speed:	3.2 GHz
  Number of Processors:	1
  Total Number of Cores:	4
  L2 Cache (per Core):	256 KB
  L3 Cache:	6 MB
  Memory:	16 GB
  Boot ROM Version:	IM142.0118.B12
  SMC Version (system):	2.15f7
 
Crash occurs when loading a file that calls lookup() function with an undefined table.  This crash occurs every time and happens as soon as the file is loaded.  The user does not need to hit F5 or F6.

```
m12 = [
  [boltDia, 12],
  [headHexThick, 8],
  [boltNut, 19],
  [boltNutMax, 22.1],
  [washerDia, 24],
  [washerThick, 2.5],
  [socketHeadThick, 12],
  [socketDia, 18],
  [socketSize, 10]
];

fastnerList = [m12]

// fastnerType = fastnerList[0]; // this code should crash every time; uncomment this line for normal load 

boltDia1 = lookup(boltDia, fastnerType);
headHexThick1 = lookup(headHexThick, fastnerType);
boltNut1 = lookup(boltNut, fastnerType);
echo (fastnerType);
echo (boltDia1);
echo (headHexThick1);
echo (boltNut1);
```

TakeItAndRun, 2015-12-22:
This does not crash on windows with OpenSCAD <http://www.openscad.org>
version 2015.03-1

but displays this error directly after loading:

ERROR: Parser error in line 18: syntax error

ERROR: Compilation failed!

t-paul, 2015-12-22:
I can reproduce the crash with `a = lookup(3, undef);` with the nightly build on Linux (2015.12.14.nightly (git f0a935d)).

Crash happens with 2015.03-2 too, no crash with 2015.03-1.

closed 2015-12-22 by kintel

TakeItAndRun, 2015-12-22:
Ooops, there was a missing ; in line:
fastnerList = [m12];

It runs fine on windows with OpenSCAD <http://www.openscad.org> version
2015.03-1.
Just warnings about undef variables.

2015-12-22 15:35 GMT+01:00 Peter Falke <stempeldergeschichte@googlemail.com>
:

txoof, 2015-12-22:
@kintel Wow! Thanks for fixing this so quickly.  I love how active you guys are.

MichaelAtOz, 2015-12-22:
> This crash occurs every time and happens as soon as the file is loaded. The user does not need to hit F5 or F6.

@txoof I presume you have Automatic Reload and Preview set. It's good to turn that off when debugging...