openscad/openscad #1531: Apply operators on objects generated from lists: Unpacking operator ?

bertrand-caron, 2015-12-27:
Hi,

I am fairly new to openSCAD, its syntax and its types but I am fairly experienced with Python and I was wondering if there was an operator similar to Python's `*` [(Python Doc)](https://docs.python.org/2/tutorial/controlflow.html#unpacking-argument-lists) that would allow me to do something that would write in pseudo-code:

```
sizes = [1,2,3];

union(){
    *[ for (i = sizes) cube([1,2,i]) ]
}
```

If not, is there a more openSCAD-ish way to apply operators on objects generated from lists ?

nophead, 2015-12-27:
```
sizes = [1,2,3];

union(){
     for (i = sizes) 
          cube([1,2,i]);
}
```
Does what you want but is not very interesting as the result is just the biggest cube.

bertrand-caron, 2015-12-27:
Thanks, that is exactly what I was looking for !

I guess it would make more sense in an example such as this one, for example for parametrising a series of hole:

```
difference(){
    cube([10,10,10], center=true); // Make a 10x10x10 cube
    union(){for (i = [-10:2:10]) translate([i,0,0]) cylinder(r=0.5, h=10.+1.0, center=true);} // Make a series of holes (cylinders) in the x direction separated by 2mm.
 };
```

In my opinion it would be great to mention in the documentation that operators can accept list comprehensions, and show such an example (feel free to use this one if you think its any good).

nophead, 2015-12-27:
There isn't a list comprehension in your example. [-10:2:10] is a range, not a list.

The key things are that for() will accept either a range or a list. The list can be constructed with a list comprehension.
```
for(i = [for (i = [1:5]) i * i])
   echo(i);
```

ECHO: 1
ECHO: 4
ECHO: 9
ECHO: 16
ECHO: 25

nophead, 2015-12-27:
The first for is a for statement. The second one is a list comprehension.

closed 2016-01-08 by t-paul