openscad/openscad #1535: rendering failure of transformed surface with "PolySet has nonplanar faces. Attempting alternate construction"

mnottale, 2016-01-05:
The following fails to render (model in attached file). If the transformations are removed it renders fine.

[model.data.txt](https://github.com/openscad/openscad/files/78896/model.data.txt)

:

!union()
{
    h = 30;
r = 30;
e = 3;
     cylinder(h=h, r=r+e, $fn=100);
     translate([0,0,15]) rotate(-125, [0,0,1])
          translate([0.5,0,0])
          rotate(-90, [1,0,0])
          rotate(90, [0,1,0]) scale([0.154, 0.109, 1])
            surface(file = "model.data", center = true, invert = false);
}

mnottale, 2016-01-05:
Some more tests seem to show that the downscaling is what causes the issue.

MichaelAtOz, 2016-01-07:
Potentially the same as #1512 so I tried removing the coincident faces (top/bottom of cylinder & the surface).

By 
```
cylinder(h=h+10, r=r+e, $fn=100);
	translate([0,0,20]) 
```
So the wolf/dog is centered. But...
[image]
the surface disappears when rendered
[image]

```
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
PolySet has nonplanar faces. Attempting alternate construction
Geometries in cache: 8
Geometry cache size in bytes: 166141072
CGAL Polyhedrons in cache: 2
CGAL cache size in bytes: 538784
Total rendering time: 0 hours, 1 minutes, 31 seconds
   Top level object is a 3D object:
   Simple:        yes
   Vertices:      200
   Halfedges:     600
   Edges:         300
   Halffacets:    204
   Facets:        102
   Volumes:         2
Rendering finished.
```
Something weird.

Postulation: scale() with small numbers of surface(), which already has values <1.0 is causing issues similar to that discussed http://forum.openscad.org/Unnecessary-triangulation-in-linear-extrude-td15429.html

But that doesn't explain the disappearing wolf...

MichaelAtOz, 2016-01-07:
strike "which already has values <1.0 is causing issues similar to" ... not sure what I was thinking...oh Lithophanes...

MichaelAtOz, 2016-01-07:
BTW, the lack of an error in the console was due to caching, I changed the scale factor and got the same error as the OP, hence why the wolf disappeared...

MichaelAtOz, 2016-01-07:
Gets the same error with a cube() instead of cylinder().
I then embedded a cube(1), the only other item with the rotated/scaled surface(), totally inside the surface geometry, so it has no coincident surfaces but forces CGAL use. Gets the same error, so it is likely the number formats getting non planner as per the forum discussion referenced above. ??

MichaelAtOz, 2016-01-07:
Simplified for render time, and memory use.
[image]
```
!union()
{
h = 30;
r = 30;
e = 3;
//translate([0,15,-0.5])
//rotate([90,0,0])
cylinder(h=h, r=r+e, $fn=20);
translate([0,0,15]) 
rotate(-125, [0,0,1])
translate([0.5,0,0])
rotate(-90, [1,0,0])
rotate(90, [0,1,0]) 
scale([0.154, 0.109, 1])
surface(file = "model3.data", center = true, invert = false);
}
```
model3.data is just lines 1-90 of the original. Renders in ~30s to get the error, ~2m in the case below, on my PC.

The above causes the OP error.
Uncommenting the two lines and commenting out the lines between cylinder() and scale() gives, in effect, the same geometry without the error (or the PolySet has nonplanar faces msg).
[image]

Thus it is the rotation/translation of the scaled geometry which is stuffing up the faces, not the scale().

Modified to work with 2014.03
```
h = 30;
r = 30;
e = 3;
union()
{

//translate([0,15,-0.5])
//rotate([90,0,0])
cylinder(h=h, r=r+e, $fn=20);

translate([0,0,15]) 
rotate(-125, [0,0,1])
translate([0.5,0,0])
rotate(-90, [1,0,0])
rotate(90, [0,1,0]) 

scale([0.154, 0.109, 1])
surface(file = "model3.data", center = true, invert = false);
}
```
Crashes. So presumable the 'non planar faces' bit was introduced to fix an underlying problem which is still there??

kintel, 2024-03-27:
Status as of 2024-03-24: Works both using CGAL (21 minutes) and Manifold (9 seconds).

closed 2024-03-27 by kintel