openscad/openscad #1539: creating vector

mirecta, 2016-01-06:
concat is a nice unction but interpereter behavior not 

i still cant  build vector dynamicaly  , its a basic feature (recursion is not good solution for bigger vectors)
a = [[0,0]];

for (e = [0:10]){
    
    b=[[e,e]];
    a = concat(a,b);
    echo(a);
}

echo (a);

i cant create curves ,  surfaces etc...

t-paul, 2016-01-06:
You can get around the recursion limitation by using a recursive form that allows tail-recursion. But you probably want to look at the list comprehension expressions:
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/List_Comprehensions

For some more complex examples see: https://github.com/openscad/list-comprehension-demos

closed 2016-01-06 by t-paul

mirecta, 2016-01-06:
Thx this is really big magic :D  , but it is tool for my need