openscad/openscad #1541: Confusing result with rotation of 2D shapes

t-paul, 2016-01-08:
```
module cut() {
	square([3, 12], center=true);
	translate([0, -3, 0]) square([6, 3], center = true);
}

difference() {
	square([100, 55], center = true);
	translate([0, 24, 0])
		cut();

	translate([0, -24, 0])
		rotate([180, 0, 0])   // <==
			cut();
}
```

The `rotate([180, 0, 0])` around X axis produces different results in preview and render.

See http://www.thingiverse.com/groups/openscad/topic:4328

nophead, 2016-01-08:
Using a user mode rotate fixes it, see http://forum.openscad.org/Inaccurate-rotates-tt15537.html, although I don't understand why.

nophead, 2016-01-08:
Possibly the Z coordinates become non-zero with the broken rotate. Adding a Z value to the translate makes the cut vanish in preview. Z values should be ignored in 2D operations but they seem to be used in preview but not F6.

txoof, 2016-01-08:
Indeed avoiding z coordinates solves this issue.

nophead, 2016-01-08:
Using rotate([0, 0, 180]) or simply rotate(180) fixes the problem. There are several issues here:
* rotate should only be around z when operating on 2D objects. So the code is actually invalid.
* rotate by 90 and 180 should be numerically exact because all the transformation matrix values are 0 or +/- 1. All conversions from degrees to radians should be removed from the source code except in a set of trig functions that take degrees with special cases for 0, 30, 45, 60 and 90. These should then be used everywhere trig is needed. Currently this is only done for the user trig functions.
* z values in 2D objects should be ignored in both F5 and F6. Currently they give a Z offset in F5 and an implicit projection in F6.

TakeItAndRun, 2016-01-08:
rotate([0,0,xx])square(3);

should trough an error

nophead, 2016-01-08:
Why?

Possibly rotate with an X or Y value should throw an error, at least they should be ignored, likewise translate with a Z value.

kintel, 2016-01-09:
I agree that the difference between preview and render is confusing.
In OpenSCAD today, 2D objects are considered to live in 2D space, meaning any 3D transforms are considered invalid - we just don't enforce this.
However, another way of looking at this would be that 2D object live in 3D space, making 3D transforms perfectly valid. To make this usefule, we'd just have to introduce/extend operators which allow for building manifold surfaces from said 2D objects. As an example, this could be very convenient for building convex hulls.

I believe RapCad is experimenting with allowing the "2D objects in 3D space" concept.

Neon22, 2016-01-09:
If you decide to go this route then I'd also vote for Wires.
    I.e wires as paths. so we can see them, one node per sample.
    (possibly also with an upvector to indicate orientation)
    This would make visualising easier as we start using splines etc...
    
    
    On 1/9/2016 4:16 PM, Marius Kintel
      wrote:
    
    
      I agree that the difference between preview and render is
        confusing.
        In OpenSCAD today, 2D objects are considered to live in 2D
        space, meaning any 3D transforms are considered invalid - we
        just don't enforce this.
        However, another way of looking at this would be that 2D object
        live in 3D space, making 3D transforms perfectly valid. To make
        this usefule, we'd just have to introduce/extend operators which
        allow for building manifold surfaces from said 2D objects. As an
        example, this could be very convenient for building convex
        hulls.
      I believe RapCad is experimenting with allowing the "2D objects
        in 3D space" concept.
      —
          it on GitHub.
      
        
          
          
        
        
      
      No virus
        found in this message.
        Checked by AVG - www.avg.com
        Version: 2016.0.7294 / Virus Database: 4489/11358 - Release
        Date: 01/08/16

nophead, 2016-01-09:
If you allow 2D shapes to be moved into 3D space then they should no longer be classed as regular 2D, instead they are a new class 2.5D. Further 2D operations on them should fail because they are out of the 2D plane and F6 should fail if they are still present at the top level because the implicit projection makes no sense. About the only things that do make sense is further transforms, hull or an explicit projection back to 2D. 

I think full 3D operations on 2.5D would open a massive can of worms and cause lots of edge cases and user confusion like this.

SarenT, 2023-06-11:
I would like to have a square in XZ plate and extrude it. I tried to rotate a square but I have been facing these issues. Why is this such a problem? In other CAD (e.g. FreeCAD) sketches are 2D and can appear on any plane (even tilted ones).