openscad/openscad #1543: needless linking against libqt5opengl when using QOpenGLWidget

chrysn, 2016-01-08:
when building with `CONFIG+=qopenglwidget`, `libQt5OpenGL.so` seems not to be technically required any more: when absent, the compilation only halts at the linker stage, when the linker is invoked with `-lQt5OpenGL`, which then fails.

being relatively unfamiliar with the qmake build system, i couldn't track this down much further (i only found mentions of `/qtopengl/i` in tests / cmake, no literal mention of `/qt5opengl/i`, and abundant mention of opengl in general) -- if you have the impression that openscad's scripts don't cause this and rather some qmake automatism is in error, i'll happily report this bug there against.

for the time being, i have resorted to keeping the Qt5OpenGL library around for building (next debian packages are likely to be built with QOpenGLWidget to combat http://bugs.debian.org/806670 / issue #1045).

(by the way, possibly related: the same shlibdeps mechanism that warns me about "could avoid a useless dependency if ... was not linked against libQt5OpenGL.so.5" also warns about libQt5Concurrent and libQtPrintSupport; if it's related, that might help, otherwise ignore it for this issue)

t-paul, 2016-01-08:
That should be https://github.com/openscad/openscad/blob/master/openscad.pro#L119

The `concurrent` is needed for Qt4, I think.
The `opengl` could probably then move as else part into `qopenglwidget`

kintel, 2016-01-08:
@t-paul I believe concurrent is still needed

kintel, 2016-01-13:
Update: We're using QtConcurrent, but the API's we're currently using happens to be header-only in the current version of Qt. Since the docs doesn't specify this, I think it's safer to keep linking with it in case this changes in the future.

kintel, 2016-01-13:
Closing. Please open separate issues for concurrent or printsupport if that needs special care

closed 2016-01-13 by kintel