openscad/openscad #1545: ld error - building from source

osj1961, 2016-01-09:
Trying to build on a 32 bit system running ubuntu 15.04.  All the prereqs in 
./scripts/check-dependencies.sh say OK.  Compiling of all the .o files seems to go fine 
(a smattering of warnings), but linking fails with:

/usr/bin/ld: objects/OffscreenContextGLX.o: undefined reference to symbol 'XGetErrorText'
/usr/lib/i386-linux-gnu/libX11.so.6: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
Makefile:678: recipe for target 'openscad' failed
make: *** [openscad] Error 1

kintel, 2016-01-09:
It could be our latest OpenGL fixes.
Could you try to ```git checkout d4bfcdf373c1cfbd4492ae79e3db4e129d731b97``` and see if that works?

osj1961, 2016-01-10:
Hi again,

That did it.  The main welcome screen says "OpenSCAD 2016.01.09" (today's date, so I assume that's because I just compiled it).  I don't see any way to get a version number besides that, but the version I got from ppa says "OpenSCAD 2015.03" in that location.

kintel, 2016-01-10:
@osj1961 FYI: The master branch builds a development snapshot with the build date as version. If you want to build a particular version of OpenSCAD, either check out the corresponding tag (e.g. "openscad-2015.03-2"), or download the release tarball. Those will have a hardcoded version number.

commit d8f7102c (2016-01-10): Link to GLU and X11 also when using QOpenGLWidget (fixes #1545).
commit f460ca23 (2016-01-10): Merge pull request #1550 from openscad/issue1545

Link to GLU and X11 also when using QOpenGLWidget (fixes #1545).
closed 2016-01-10 by kintel