openscad/openscad #1546: Build error with boost 1.60.0 - 'lexertext' was not declared in this scope

akien-mga, 2016-01-09:
Hi there,

I packaged openscad for Mageia and we're currently having a build issue with 2015.03-2 since boost 1.60.0 was introduced in Cauldron, our development release.

The build fails with:
```
g++ -c -O2 -g -pipe -Wformat -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fstack-protector --param=ssp-buffer-size=4 -fomit-frame-pointer -march=i586 -mtune=generic -fasynchronous-unwind-tables -fno-strict-aliasing -frounding-math -DEIGEN_DONT_ALIGN -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/harfbuzz -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/freetype2 -I/usr/include/libpng16 -I/usr/include/freetype2 -I/usr/include/libpng16 -I/usr/include/libxml2 -I/usr/include/freetype2 -I/usr/include/libpng16 -O2 -g -pipe -Wformat -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fstack-protector --param=ssp-buffer-size=4 -fomit-frame-pointer -march=i586 -mtune=generic -fasynchronous-unwind-tables -fno-strict-aliasing -DPIC -fPIC -std=gnu++0x -D_REENTRANT -Wall -W -Wno-unused-local-typedefs -fPIC -DOPENSCAD_VERSION=2015.03-2 -DOPENSCAD_SHORTVERSION=2015.03 -DOPENSCAD_YEAR=2015.0 -DOPENSCAD_MONTH=03.0 -DOPENSCAD_DAY=.0 -DDEBUG -DENABLE_CGAL -DENABLE_OPENCSG -DUSE_SCINTILLA_EDITOR -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_PRINTSUPPORT_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_CONCURRENT_LIB -DQT_CORE_LIB -I. -Isrc -I/usr/lib/qt5/include -Isrc/libtess2/Include -I/usr/lib/qt5/include/QtOpenGL -I/usr/lib/qt5/include/QtPrintSupport -I/usr/lib/qt5/include/QtWidgets -I/usr/lib/qt5/include/QtGui -I/usr/lib/qt5/include/QtConcurrent -I/usr/lib/qt5/include/QtCore -Iobjects -isystem /usr/include/eigen3 -I/usr/lib/qt5/mkspecs/linux-g++ -o objects/lexer_lex.o lexer_lex.cpp
src/lexer.l: In function 'int yylex()':
src/lexer.l:188:30: error: 'lexertext' was not declared in this scope
 {UNICODE}               { parser_error_pos -= strlen(lexertext) - 1; stringcontents += lexertext; }
                              ^
src/lexer.l:189:29: error: 'lexertext' was not declared in this scope
 \\x[0-7]{H}             { unsigned long i = strtoul(lexertext + 2, NULL, 16); stringcontents += (i == 0 ? ' ' : (unsigned char)(i & 0xff)); }
                             ^
src/lexer.l:190:24: error: 'lexertext' was not declared in this scope
 \\u{H}{4}|\\U{H}{6}     { char buf[8]; to_utf8(lexertext + 2, buf); stringcontents += buf; }
                        ^
src/lexer.l:191:21: error: 'lexertext' was not declared in this scope
 [^\\\n\"]  { stringcontents += lexertext; }
                     ^
src/lexer.l:202:30: error: 'lexertext' was not declared in this scope
 {UNICODE}               { parser_error_pos -= strlen(lexertext) - 1; }
                              ^
src/lexer.l:209:30: error: 'lexertext' was not declared in this scope
 {UNICODE}               { parser_error_pos -= strlen(lexertext) - 1; }
                              ^
Makefile:7126: recipe for target 'objects/lexer_lex.o' failed
```

You can see the full build log here: http://pkgsubmit.mageia.org/uploads/failure/cauldron/core/release/20160109134741.akien.valstar.28680/log/openscad-2015.03.2-3.mga6/build.0.20160109135654.log

And the spec file used for the build (which lists among others the build dependencies used): http://svnweb.mageia.org/packages/cauldron/openscad/current/SPECS/openscad.spec?revision=915546&view=markup

t-paul, 2016-01-09:
I've uninstalled the boost dev packages on my Debian system and compiled against a manually built boost-1.60 without any issues except CGAL complaining it likes to have boost-1.58.

Is there an easy to use live image to test this on?

On a totally unrelated note, it seems to build a debug version. Please consider doing a release build (`qmake CONFIG-=debug`), if possible, to prevent issue #1045.

akien-mga, 2016-01-09:
> Is there an easy to use live image to test this on?

Sadly not for the moment, we haven't been able yet to spin live images due to a regression in RPM 4.13.0 RC1. It should still be possible to install Mageia Cauldron using a network install or upgrade from Mageia 5, but it's probably overkill.

Here's a list of the dependencies installed for the above build: http://pkgsubmit.mageia.org/uploads/failure/cauldron/core/release/20160109134741.akien.valstar.28680/log/openscad-2015.03.2-3.mga6/rpm_qa.0.20160109135654.log
GCC 5.3.1 might be a potential culprit too.

> On a totally unrelated note, it seems to build a debug version. Please consider doing a release build (qmake CONFIG-=debug), if possible, to prevent issue #1045.

Thanks, I'll do that!

t-paul, 2016-01-09:
My Debian has: gcc version 5.3.1 20160101 (Debian 5.3.1-5)

akien-mga, 2016-01-09:
Thanks, I'll continue to investigate what might be wrong on Mageia's side then.

kintel, 2016-01-20:
@akien-mga Any news? Was this a build system issue on your side?

joequant, 2016-02-14:
I fixed the mageia build, and the problem doesn't seem to have anything to do with the new boost.

What's happening is that the /usr/bin/flex command that is generated by the qmake command is missing the --prefix=lexer command line option.  I can make the build work by adding it manually.

vtols, 2016-04-10:
I have the same problem. Some time before (nearly one month ago), all was OK. But now buid fails with this messages. My system is Arch Linux. Month ago, when build was successful, I had boost 1.60.0-2, gcc 5.3.0-5 and flex 2.6.0-2. Now I have the same. Why build fails now? How to fix it?

t-paul, 2016-04-10:
As the earlier comments show, we'd need some help with reproducing this. Are there now distribution images available which can be run in a VM?

joequant, 2016-04-10:
The fix should be runnable on a Mageia cauldron machine.

t-paul, 2016-04-10:
Well, yes, so far this issue seems to be happening only on Arch (maybe due to some library or tool update). So please someone who knows Arch provide a reasonable way to install a ready to use development environment (like Live-CD as mentioned by @akien-mga).

I'm not seeing the issue on any of the distributions covered by the [nightly builds](https://build.opensuse.org/package/show/home:t-paul/OpenSCAD) (Debian, Ubuntu, OpenSuSE, Fedora).

vtols, 2016-04-10:
Maybe I edited Makefile not properly, but runninng flex manually with --prefix=lexer resolves the problem.

kintel, 2016-04-10:
@vtols That's odd. It's hard to fix unless we can reproduce it. As @t-paul mentioned, having access to a VM image where this fails would be awesome. Alternatively, if you provide a full build log, including the generated makefiles it might be possible to hunt it down.

joequant, 2016-04-10:
Something you may want to check is if the build is overwriting the Makefile
or is using the Makefile.  I seem to remember that this was an issue when I
tracked down the problem.  qmake will overwrite the Makefile which meant I
had to modify the Makefile after it was generated.

Here is the buildspec from Mageia

%build
%qmake_qt5 VERSION=%{uversion} \
           PREFIX=%{_prefix} \
           CONFIG-=debug
# Work around bug
# https://github.com/openscad/openscad/issues/1546
sed -i -e 's!/usr/bin/flex!/usr/bin/flex --prefix=lexer!' Makefile
%make_build

kintel, 2016-04-11:
@joequant Did you look into _why_ you have to modify the makefile. This looks like a workaround for a bug somewhere..

joequant, 2016-04-11:
Yes.  IIRC, the reason for the change is that the calling arguments for
flex changed at some point.

t-paul, 2016-04-11:
The `flex.pri` is supposed to add the `-P` option while running `qmake`.

Can you please:
* Post the version number of flex (using 2.6.0 here)
* Check if it works with manual call of `flex -P lexer` (instead of the `--prefix`)
* Post the output of `qmake -d -d CONFIG=-debug 2>&1 | grep flex`

Salamandar, 2016-05-24:
I still got this bug.

Here is the output you asked. There is `flex -P`. But there isn't any `*.lexer.cpp` file…

     ❯ qmake -d -d CONFIG=-debug 2>&1 | grep flex                                                      [11:22:05]
    DEBUG 2: literal "flex"
    DEBUG 2: evaluated expression => flex
    DEBUG 1: /usr/lib/qt/mkspecs/common/unix.conf:7: QMAKE_LEX := flex
    DEBUG 2: literal "flex.pri"
    DEBUG 2: evaluated expression => flex.pri
    DEBUG 1: /home/salamandar/tmp/aur-openscad-git/src/openscad/common.pri:7: calling built-in include(flex.pri)
    DEBUG 1: visiting file /home/salamandar/tmp/aur-openscad-git/src/openscad/flex.pri
    DEBUG 1: /home/salamandar/tmp/aur-openscad-git/src/openscad/flex.pri: entering block
    DEBUG 2: hashed literal "flex.name"
    DEBUG 2: evaluated expression => flex.name
    DEBUG 1: /home/salamandar/tmp/aur-openscad-git/src/openscad/flex.pri:2: flex.name := Flex ${QMAKE_FILE_IN}
    DEBUG 2: hashed literal "flex.input"
    DEBUG 2: evaluated expression => flex.input
    DEBUG 1: /home/salamandar/tmp/aur-openscad-git/src/openscad/flex.pri:3: flex.input := FLEXSOURCES
    DEBUG 2: hashed literal "flex.output"
    DEBUG 2: evaluated expression => flex.output
    DEBUG 1: /home/salamandar/tmp/aur-openscad-git/src/openscad/flex.pri:4: flex.output := ${QMAKE_FILE_PATH}/${QMAKE_FILE_BASE}.lexer.cpp
    DEBUG 2: hashed literal "flex.commands"
    DEBUG 2: evaluated expression => flex.commands
    DEBUG 2: literal "flex"
             DEBUG 2: evaluated expression => flex -P ${QMAKE_FILE_BASE} -o${QMAKE_FILE_PATH}/${QMAKE_FILE_BASE}.lexer.cpp ${QMAKE_FILE_IN}
             DEBUG 1: /home/salamandar/tmp/aur-openscad-git/src/openscad/flex.pri:5: flex.commands := flex -P ${QMAKE_FILE_BASE} -o${QMAKE_FILE_PATH}/${QMAKE_FILE_BASE}.lexer.cpp ${QMAKE_FILE_IN}
    DEBUG 2: hashed literal "flex.CONFIG"
    DEBUG 2: evaluated expression => flex.CONFIG
    DEBUG 1: /home/salamandar/tmp/aur-openscad-git/src/openscad/flex.pri:6: flex.CONFIG := target_predeps
    DEBUG 2: hashed literal "flex.variable_out"
    DEBUG 2: evaluated expression => flex.variable_out
    DEBUG 1: /home/salamandar/tmp/aur-openscad-git/src/openscad/flex.pri:7: flex.variable_out := GENERATED_SOURCES
    DEBUG 1: /home/salamandar/tmp/aur-openscad-git/src/openscad/flex.pri:8: condition "silent" is false
    DEBUG 1: /home/salamandar/tmp/aur-openscad-git/src/openscad/flex.pri:8: taking 'else' branch
    DEBUG 1: /home/salamandar/tmp/aur-openscad-git/src/openscad/flex.pri:8: finished 'else' branch
    DEBUG 2: literal "flex"
    DEBUG 2: evaluated expression => flex
    DEBUG 1: /home/salamandar/tmp/aur-openscad-git/src/openscad/flex.pri:9: QMAKE_EXTRA_COMPILERS := flex
    DEBUG 1: /home/salamandar/tmp/aur-openscad-git/src/openscad/flex.pri:12: condition "unix" is false
    DEBUG 1: /home/salamandar/tmp/aur-openscad-git/src/openscad/flex.pri:12: AND
    DEBUG 1: /home/salamandar/tmp/aur-openscad-git/src/openscad/flex.pri:12: skipped condition "linux*"
    DEBUG 1: /home/salamandar/tmp/aur-openscad-git/src/openscad/flex.pri:12: taking 'else' branch
    DEBUG 1: /home/salamandar/tmp/aur-openscad-git/src/openscad/flex.pri:12: finished 'else' branch
    DEBUG 1: /home/salamandar/tmp/aur-openscad-git/src/openscad/flex.pri:18: condition "freebsd*" is false
    DEBUG 1: /home/salamandar/tmp/aur-openscad-git/src/openscad/flex.pri:18: taking 'else' branch
    DEBUG 1: /home/salamandar/tmp/aur-openscad-git/src/openscad/flex.pri:18: finished 'else' branch
    DEBUG 1: /home/salamandar/tmp/aur-openscad-git/src/openscad/flex.pri:22: condition "netbsd*" is false
    DEBUG 1: /home/salamandar/tmp/aur-openscad-git/src/openscad/flex.pri:22: taking 'else' branch
    DEBUG 1: /home/salamandar/tmp/aur-openscad-git/src/openscad/flex.pri:22: finished 'else' branch
    DEBUG 1: /home/salamandar/tmp/aur-openscad-git/src/openscad/flex.pri:22: leaving block, okey=true
    DEBUG 1: done visiting file /home/salamandar/tmp/aur-openscad-git/src/openscad/flex.pri
    DEBUG 1: /home/salamandar/tmp/aur-openscad-git/src/openscad/bison.pri:10: QMAKE_EXTRA_COMPILERS := flex bison
    DEBUG 1: /home/salamandar/tmp/aur-openscad-git/src/openscad/bison.pri:17: QMAKE_EXTRA_COMPILERS := flex bison bison_header
    DEBUG 1: /usr/lib/qt/mkspecs/features/uic.prf:14: QMAKE_EXTRA_COMPILERS := flex bison bison_header uic
    DEBUG 1: /usr/lib/qt/mkspecs/features/resources.prf:108: QMAKE_EXTRA_COMPILERS := flex bison bison_header uic rcc
    DEBUG 1: /usr/lib/qt/mkspecs/features/moc.prf:51: QMAKE_EXTRA_COMPILERS := flex bison bison_header uic rcc moc_header
    DEBUG 1: /usr/lib/qt/mkspecs/features/moc.prf:63: QMAKE_EXTRA_COMPILERS := flex bison bison_header uic rcc moc_header moc_source
    DEBUG 2: EscapeFilePath: flex.pri -> flex.pri

MarPiRK, 2016-07-03:
I've had the same issue on Fedora 24 (with boost 1.60.0). Changing to `-P lexer` in Makefile fixed that.

t-paul, 2018-11-01:
Closing as there's no more feedback and no progress. Official builds are now at boost versions later than 1.60 without showing issues.

closed 2018-11-01 by t-paul