openscad/openscad #1548: Display Font sample preview in OpenSCAD Font List

clothbot, 2016-01-10:
Enhancement request:

Please add a font "sample preview" in the Help - Font List window.

- Add field to specify 't' value; default to "OpenSCAD"

- Have 'Copy to Clipboard' generate a working text(t="OpenSCAD",font="blah"); line.

- Include 'Advanced' field-per-parameter mode to construct+preview whole, functional 'text()' line to copy+paste, instead of just the 'font' parameter value.

For Reference: https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Text

t-paul, 2016-01-10:
Could you please add this to #1395? I think it makes sense collect those changes in one issue.

t-paul, 2016-01-10:
Oh, on second thought, let me extend the original text there :-)....

clothbot, 2016-01-10:
Ah good, didn't spot that in the existing issues when I did a cursory scan for similar.

droidmunkey, 2016-03-22:
Where IS the font list now? I don't have a menu at all in the latest build.

t-paul, 2016-03-22:
on Ubuntu? -> https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#I.27m_not_getting_any_menubar_when_running_OpenSCAD_in_Ubuntu.2C_how_can_I_get_it_back.3F

droidmunkey, 2016-03-22:
Awesome thanks! That fixed it!   There is none on the screenshot from
http://www.openscad.org/ either so i wasn't sure if it was removed ;-)

--Robert

t-paul, 2016-03-22:
Hehe, right, that's OSX. Apple knows how to get the menu bar to the top of the screen. Seems Ubuntu not so much.

closed 2025-09-01 by jordanbrown0