openscad/openscad #1554: Add Mesh Decimation options to STL import

clothbot, 2016-01-12:
When operating on high poly-count STL mesh sources, often times it's not necessary to use the exact mesh to achieve the design goals.

Hypothetical Example: Creating a negative-space mold for casting.

difference() {
  minkowski() {
    import("big_mesh.stl",decimation_triangles=1000,decimation_error=0.01);
    sphere(r=4*min_wall_th,$fn=8);
  }
  import("big_mesh.stl");
}

Potentially useful code:

http://kmamou.blogspot.ca/2015/06/a-simple-c-class-for-3d-mesh-decimation.html
https://github.com/kmammou/MeshDecimation
- BSD 3-clause licensed; compatible with OpenSCAD.

doug-moen, 2016-01-12:
This is a slow operation. The paper says 15 seconds to reduce the Stanford
bunny to 1000 triangles. You'd likely be better off performing this
operation offline using a separate tool, and then importing the result.
Then you wouldn't incur the cost each time you render or preview the model.

The best reason I can think of for performing this operation inside
OpenSCAD is if the decimation parameter is part of the model.

Trygon has previously proposed using an error tolerance variable, $fe, for
parameterizing models that contain spheres, cylinders and other curved
surfaces, as an alternative to $fn, $fa.
http://forum.openscad.org/fe-Tolerance-based-arc-approximation-td14212.html

I've tried using $fe, and I like it. Nice if it was built in. This proposal
also uses an error tolerance as a decimation parameter. So if you are
already using $fe to parameterize your model, then you could also use it to
parameterize decimation of an imported stl. That's the use case that occurs
to me, where there's a benefit to having decimation built in.

clothbot, 2016-01-12:
Of interest/use, in Khaled's v-hacd project he's seeing orders-of-magnitude speed-up via OpenCL:

http://kmamou.blogspot.ca/2015/01/opencl-acceleration-for-v-hacd.html

Something to bare in mind across the board.

clothbot, 2016-01-12:
@doug-moen - From the paper: "This data was collected on an SGI Indigo2 with a 195 MHz R10000 processor and 128 Mbytes of memory." Also, a lot of Khaled's own work is being developed in the context of real-time applications for things like bullet3 physics engine and Unreal gaming engine usage: http://kmamou.blogspot.ca/2015/04/unreal-engine-4-switches-to-v-hacd.html

...so "slow" is relative. :-)

Neon22, 2016-01-13:
Decimation is one way to simplify the object but general purpose polygon reduction strategies often work better and preserve the shape better than simple decimation. Alas its a lot of tricky code (from personal experience). Hughes hoppe is the guy with the papers for this.
2006 Poisson surface reconstruction
2005 Fast exact and approximate geodesics on meshes
http://research.microsoft.com/en-us/um/people/hoppe/