openscad/openscad #1555: export to binary STL

EastEriq, 2016-01-14:
For a smaller file size than ASCII. Or is it just me not finding the right place to control it?
My current workflow involves exporting as ASCII stl, and reopening the stl file in MeshLab to resave it as binary.

t-paul, 2016-01-14:
No, binary STL is not supported yet. This also depends on #649 providing a way to specify export options.
In case there's scripting in the workflow, there's also `admesh` which allows converting on command line.

EastEriq, 2016-01-14:
I see. Thanks for the admesh tip, looking into.

TLC123, 2016-02-17:
why not bundle admesh with openscad and have some piping trough it in the export dialog that'll solve stl repair issue

t-paul, 2016-02-17:
I'm not sure general repair on export is reliable enough and it would cover only a single output format. We probably can add the library version of admesh for optional repair at STL import.

doug-moen, 2016-02-17:
If the requirement is "a smaller file size than ASCII STL"  then we can
satisfy this by supporting compressed AMF, which is on average half the
size of binary STL.

>

EastEriq, 2016-02-17:
> If the requirement is "a smaller file size than ASCII STL" 

For me it is. Our workshop accepts only STL and clouding 200MB chunks is not always practical. But see #1557...

kintel, 2016-02-17:
@TLC123 Admesh doesn't perform the type of repair we need. Also, the admesh code for binary STL export is not ideally suited for integration; it's probably less code to implement our own exporter than to write the glue to integrate admesh. See #1042 for more info.

TLC123, 2016-02-17:
Somewhat off topic. If some one may clarify. if no CSG is used, say we used
just plain polyhedron, couldn't export just bypass CGAL?

kintel, 2016-02-17:
@TLC123 yes, if no CSG or minkowski operations are used, we typically won't run it through CGAL. That doesn't influence this discussion though (repair issues should be discussed in an appropriate issue).

donbright, 2017-12-13:
the maintenance headaches of this far outweigh the potential benefits. STL ascii vs binary was designed decades ago when the difference mattered for speed. nowdays that speed question is irrelevant because computers and storage have gotten so much cheaper and faster. modern computers have gigabytes of storage inside gadgets that are smaller than a fingernail and cost 5 dollars. there is no good reason to increase the complexity of a program in order to save a little bit of space on output.

Here is the main problem. ASCII decimal floating point and binary floating point are not the same set of numbers. You cannot convert from one back to the other without high risk of losing information. classic example, 0.6 decimal cannot be converted exactly to a finite binary floating point number.  

```
don@serebryanya:~/src/openscad/binmaster$ cat t.cc
#include <iostream>
#include <iomanip>
main() {
std::cout << std::setprecision(25) << 6.0/10.0 << "\n";
}
don@serebryanya:~/src/openscad/binmaster$ g++ t.cc
don@serebryanya:~/src/openscad/binmaster$ ./a.out 
0.5999999999999999777955395

```

People (like past version of me) don't understand this. Most aren't taught it in school and it is very unpleasant to think about (cognitive dissonance related to the social view that computers are infallible math machines). Search "calculator is broken" you will find hundreds of pages and videos of people thinking their calculator is broken - Search "excel rounding error" you will find out that millions of dollars are lost because people dont know this stuff. Their calculator is not broken, it is their educational system that is broken. 

Imagine the bug reports and people confused "oh, after 20 hours of debugging, i figured out if i export in binary it works, but in ASCII it didnt.. why is OpenSCAD broken fuuuUUuuuUUUU  :( :(". its happened before on such topics.

AMF has the right idea - if people want to cut down on storage, they can compress the ASCII using something like zip. AMF doesnt have binary format.

jtoy, 2017-12-13:
I read your whole post. So which is better than in this case for a more accurate representation, binary stl or ascii stl?

doug-moen, 2017-12-13:
In theory, ASCII STL is more accurate, because it can represent 64 bit
floats, whereas binary is restricted to 32 bit floats. In practice, this
depends on both the generator and the consumer of  the STL preserving 64
bit accuracy during reads and writes. OpenSCAD throws away most of its
accuracy when writing an STL file (last time I checked) due to truncating
the ASCII representation to 6 significant digits, although this is fixable.

EastEriq, 2017-12-14:
I'm not involved in the development of openscad so you may decide
differently, but:

>nowdays that speed question is irrelevant because computers and storage
have gotten so much cheaper and faster.

yeah - till the moment you have to submit your STL to a 3d printshop which
has size limitations. Or you have to send GB over the net.

>ASCII decimal floating point and binary floating point are not the same
set of numbers.

Moot point, I presume the program works internally with machine binaries.
Besides, respectable producers and consumers on the toolchain should be
capable to compensate for truncation errors, e.g with options for fixing
solids not closed by epsilon.

Doug clarified the point below.

nsoft, 2019-11-01:
https://formats.kaitai.io/stl/index.html may be useful

nsoft, 2019-11-01:
Be sure to use a newer version though: https://github.com/kaitai-io/kaitai_struct/issues/612

t-paul, 2019-11-01:
Nice format description (although we already read binary STL, so the format is sort-of known). It's not very likely we will use that much infrastructure for exporting STL though.

rcolyer, 2020-08-02:
Feature complete with merged PR.

closed 2020-08-02 by rcolyer