openscad/openscad #1556: options GUI, or at least document OpenSCAD.conf

EastEriq, 2016-01-14:
I can find no such GUI in 2015.03, and I don't know what would be the legal options in ~/.config/OpenSCAD/OpenSCAD.conf . An earlier version I tried, I think 2014.01, at least had a GUI for some of them, accessible from the menu bar in the main window, which seems gone.

t-paul, 2016-01-14:
The GUI is still there and was only extended a bit. I assume this is on Ubuntu? This messes with menu entries and breaks applications that way. See the [FAQ entry](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#I.27m_not_getting_any_menubar_when_running_OpenSCAD_in_Ubuntu.2C_how_can_I_get_it_back.3F).

EastEriq, 2016-01-14:
Very right, Ubuntu, and found the FAQ just a minute ago; removed appmenu-qt5 and enjoying. I think you can close, sorry for the noise...

closed 2016-01-14 by EastEriq