openscad/openscad #1557: import and export openCTM

EastEriq, 2016-01-14:
I just discovered http://openctm.sourceforge.net/?page=about . Compared to STL, even binary, size compression is really impressing. A pipeline via ctmconv is possible, but maybe native support is worth considering.

t-paul, 2016-01-14:
Is that used in any application?
Also the page suggests the file format specification is still "work in progress".

I'd suggest looking at AMF which has at least some support by now and also supports extended features to STL and a compressed format (which is not widely supported yet).

EastEriq, 2016-01-14:
For sure MeshLab reads the format, and I understand the author developed a plugin for blender too.

Yes, I see that it is an one man project, probably dormant since 2010, with minimum later followup. I wast just throwing in...

EastEriq, 2016-01-14:
Quick test with one mesh of mine:

openscad -->stl ascii: 26.1MB
openscad -->amf ascii: 26.1MB
stl ascii --> stl binary (meshlab, admesh buffer overflows in my setup): 8.1MB
stl binary --> ctm (ctmconv): 148kB!

MichaelAtOz, 2016-01-15:
ya know they sell 8TB drives now...

bobc, 2016-04-11:
AMF just specifies ZIP to create a binary AMF file. It is hard to justify an application specific compression format when ZIP is so effective and ubiquitous.

t-paul, 2016-04-11:
I'd suggest to close this issue. The spec is still marked beta, the latest release of openCTM is 6 years old, and I can't see any code changes in the last 4 years.

In case the lib/format ever gets picked up again, we can always just reopen the issue and have another look.

donbright, 2017-12-13:
Three votes for Close.

MichaelAtOz, 2017-12-14:
+1, makes 5. Closing...

closed 2017-12-14 by MichaelAtOz