openscad/openscad #1559: Accept new list comprehension features as a stable feature

kintel, 2016-01-19:
Accept the experimental **else** and **each** list comprehension features as stable.

TODO:
* [ ] Evaluate the 'each' syntax
* [ ] Cheat sheet
* [ ] Documentation
* [ ] Optional: Add or update example

t-paul, 2016-01-24:
@doug-moen while tinkering with the parser, I found that `each` is a bit of an exception when looking at the other list comprehension expressions as it is the only one that has no `(` `)`. I guess that would make it more complicated to have a unified syntax. How is that looking in the OpenSCAD2 parser definition?

doug-moen, 2016-01-25:
The implementation of EACH in the OpenSCAD2 grammar is trivial, but I do
need to make it a keyword.

https://github.com/doug-moen/openscad2/blob/master/parser.y

kintel, 2016-11-05:
Regarding `each`: Should we allow `each each` constructs?

Suppose we have a nested array and want to flatten this into an array of value. `each each` sounds like the right solutions since it will accept either an array or a value and unpack it if it's an array:

```
array = [
  [ [1,2], [3,4], [5,6] ],
  [ [7,8], [9,10], [11,12] ]
];

res = [ for (a=array) each each a ];
echo(res);
```
=> `ECHO: [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12]`

..now, if the initial array isn't evenly nested:

```
array = [
  [ [1,2], [3,4], [5,6] ],
  [ [7,8], [9,10], [11,12] ],
  [ 13,14 ]
];

res = [ for (a=array) each each a ];
echo(res);
```
=> crash (see #1851)

Of course, we could rewrite the latter flattening expression as:
`[ for (a=array, b=a) each b ]`.

If we want to support `each each` we need to be more robust, or provide an error if the arguments have uneven nesting. Any ideas what the best solution would be for us?

doug-moen, 2016-11-06:
In my openscad2 prototype, each takes an expression as an argument, but
'each expr' is a generator, not an expression. So my code reports a "not an
expression" error for 'each each expr'. This makes sense if you consider
'each x' to be shorthand for 'for (i=x) i'.

Also, if the argument to 'each' evaluates to something that isn't a list
value, then a "not a list" error is reported. I see that the OpenSCAD
'each' behaves differently.

In my system, 'concat' with a single argument flattens a list by one level.
So 'concat([[1,2],[3,4]])' -> [1,2,3,4]. To flatten an an array by two
levels, you apply concat twice. Using the short form function call syntax
in OpenSCAD2, that would be:
   concat concat array
which is shorter and simpler than
   [ for (a=array) each each a ]

In fact, I think your proposed semantics for
each each a
are identical to
each concat a
in my system.

I think you get more benefit from a single argument 'concat' primitive (or
call it 'flatten' if you prefer) than you get from each each, which to me
seems more obscure. So maybe that's a counterproposal.

Finally, in my system, concat([a,b,c]) reports an error if a, b or c are
not lists. The OpenSCAD convention that "nothing is an error" drives me
crazy, so my system is strict about error reporting. I realize that this is
a personal preference, but for me, detecting and reporting errors is
important for debuggability, which I put a high value on. Right now, I get
a detailed stack trace when an error occurs, but soon you will be dumped
into an interactive debugger.

>

yunfan, 2017-05-14:
why not add the python style command `map`, `reduce`, `filter`

doug-moen, 2017-05-14:
Map, reduce and filter require function values to be passed as arguments.
OpenSCAD doesn't support function values.

t-paul, 2018-10-14:
Manual: [list comprehension](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/List_Comprehensions)

t-paul, 2018-12-04:
Closed via #2522.

closed 2018-12-04 by t-paul