openscad/openscad #1562: Add subvolumes and decompose features to render()

clothbot, 2016-01-20:
Feature request time!

Please add render(subvolumes=true) and render(decompose=true) options where:

render(subvolumes=true) - performs render() and returns separate volumes that can be iterated over via children() if passed into a module:

module process_volumes() {
  for(i=[0:$children-1]) {
    echo(str("Child ",i));
    children(i);
  }
}

process_volumes() {
  render(subvolumes=true) difference() {
    cylinder(r=1,h=10,center=true);
    cube([10,10,1],center=true);
  }
}

...would see two children, the top piece of the cylinder and the bottom piece of the cylinder.

render(decompose=true) - performs render(), then 'convex_decomposition_3' on the resulting blob.  Hypothetical use-model would include using the results as inputs to piece-wise hull() operations where you want more of the 'holes' preserved.

See http://doc.cgal.org/latest/Convex_decomposition_3/index.html

Filing this as an issue because I haven't quite figured out how to get render() to 'push back' the resulting sub-geometries.

t-paul, 2016-01-20:
With the implicit union, I think that's currently not possible. See the #350 branch that might have the framework for returning a list of geometries.

I'm not sure how the mentioned use-case would work as it requires either knowledge of the order of pieces returned by CGAL or some kind of additional ordering after the decomposition.

doug-moen, 2016-01-20:
Please don't overload the render() function like this. Render is a "pragma"
that gives hints to the implementation to speed up preview. It has no
semantic effect, and calls to render can be removed without affecting the
final geometry.

What you are requesting should be packaged as two new operations, eg
subvolumes() and decompose().

Subvolumes might be a poor name, since this should also work in 2D. Maybe
call them disjoint_decomposition() and convex_decomposition().

>

clothbot, 2016-01-20:
I'll fake-together an example for render(decompose=true) to show how it could be used for beveling the intersection of two objects without losing the holes.  The 'children(0)' input would be arbitrary geometry and the remaining 'children(1:N-1)' parts would be the convex parts, similar in concept to how difference() works.

...and yes, I'm very much ok with not overloading render().  That was the first thing that came to mind because it 'solidifies' its children(), which would be a necessary first-step before partitioning and/or decomposing could happen.

t-paul, 2016-01-20:
Maybe have a look at `minkowski()` which already uses CGAL 3d convex decomposition.

https://github.com/openscad/openscad/blob/master/src/cgaladv.cc#L181
https://github.com/openscad/openscad/blob/master/src/GeometryEvaluator.cc#L1055
https://github.com/openscad/openscad/blob/master/src/cgalutils-applyops.cc#L209

nophead, 2016-01-20:
> Render is a "pragma" that gives hints to the implementation to speed up
preview. It has no semantic effect, and calls to render can be removed
without affecting the final geometry.

The final geometry is not the only output of OpenScad. It can also be used
to make images of the model, with colour and transparency, etc. render()
has the semantic meaning of computing the geometry and rendering it as
polyhedra rather than synthesising it with pixel tricks. That is the only
way to get clean assembly views because otherwise when you put objects
close together the invisible negative volumes used to construct them
interfere with each other and create z buffer fighting.

doug-moen, 2016-01-20:
Chris said: "The final geometry is not the only output of OpenScad. It can
also be used to make images of the model, with colour and transparency,
etc. render() has the semantic meaning of computing the geometry and
rendering it as polyhedra rather than synthesising it with pixel tricks.
That is the only way to get clean assembly views because otherwise when you
put objects close together the invisible negative volumes used to construct
them interfere with each other and create z buffer fighting."

"render" is a hack to work around bugs and limitations of the current
preview implementation. The situation you describe is just another example
of this. It's a bug that preview is unreliable and sometimes doesn't
display the geometry correctly. It's conceivable that we could fix some of
these bugs, and eliminate some of the use cases for render.

By contrast, "convex_decomposition" is a bona-fide geometric operation.

clothbot, 2016-01-20:
Here you go, example of a faked convex_decomposition; I meant filleting operation, not beveling operation:

https://github.com/clothbot/ClothBotCreations/blob/master/examples/convex_decomposition_faked.scad

[image]

GilesBathgate, 2016-02-23:
Instead of introducing new modules to access the ```subvolumes()``` or presumably a whole set of modules ```disjoint_*```, could the ```children()``` module be overloaded to provide access to the children?

```
children() decompose() {  cube(10); cylinder(10); }
```

doug-moen, 2016-02-23:
@GilesBathgate I don't understand what problem you are solving by putting this functionality into `children()`. What are the benefits?

The problems I see are:
* It's not compatible with OpenSCAD2, which deprecates `children()` (new code isn't supposed to use it).
* It doesn't play well with the 'lazy union' extension of OpenSCAD. With this extension, a module can return a group of objects, and this group can be supplied as the children to another module call. The standard example is `intersection() for(..)..`, but `mymodule() decompose() foo()` would be another example.
* It's more complicated than just defining these operations as modules.

GilesBathgate, 2016-02-23:
@doug-moen I guess I was thinking aloud.  I've had a ```decompose()``` module for a long while in rapcad, and I don't know what to do with it, since the resultant parts cannot be individually accessed. 

I don't have the implicit union problem in rapcad e.g. ```intersection() for(...)...``` works as expected.

I was thinking along the lines of ```children(1) decompose() { ... }``` returning the 2nd part of the decomposition, but then why not just have an index as a parameter on the decompose module? Well because I don't want to have to call decompose multiple times to get at each part.

GilesBathgate, 2016-02-23:
@doug-moen Perhaps a usage case would help explain what I would like to be possible:

```c#
module explode(){
  for(i=[0:$children])
     translate([0,0,i*10])children(i);
}

explode() decompose() { ... }
```

doug-moen, 2016-02-23:
In rapcad, can't you write
```
mymodule() decompose() foo()
```
and then access the individual pieces of the decomposition within my module
using children?

In OpenSCAD2 you can assign a shape (or group) to a variable:
```
parts = decompose() foo();
```
and then use `len(parts)` or `parts[i]`.

I'm not sure if decompose() is directly usable in OpenSCAD without first
implementing some of these planned language extensions.

GilesBathgate, 2016-02-23:
@doug-moen Yes of course! That's a good idea now I just need to make that work. but I still wouldn't be able to do:

```scad
mymodule() difference() { cube(1); cube(2) }
```

mymodule would only have access to the result of the difference :( but if I could do:

```scad
mymodule() children() difference() { cube(1); cube(2) }
```

or from within mymodule itself:

```scad
module mymodule() {
   children() children();
}
```
```

doug-moen, 2016-02-23:
Are you saying that
```
mymodule() children() difference() { cube(1); cube(2) }
```
is equivalent to
```
mymodule() { cube(1); cube(2) }
```
? I don't think I understand what your `children() X` extension is supposed to do, or how it is relevant to `decompose()`.

GilesBathgate, 2016-02-23:
@doug-moen Yes, isn't that what @clothbot wanted in the OP?

clothbot, 2016-02-23:
The 'convex_decomposition()' operator takes an arbitrary solid input (could be composed of more than one discrete volume) and breaks it down into multiple convex parts.  As such:

convex_decomposition() difference() { cube(2,center=true); cube(1,center=true); }

...would break apart into at least six convex parts - one per side of the hollow cube.

The 'subvolumes()' could reduce 'difference() { cube(2,center=true); cube(1,center=true); }' into two volumes, where one is the outer cube(2) and the other is the inner cube(1) empty-space, but I know I would find that behaviour/side-effect confusing.  I would treat it as an optional behaviour, not a default behaviour.

GilesBathgate, 2016-02-23:
@clothbot So ```subvolumes() difference(){ cube(1,center=true); cube(2,center=true); } ``` is identical to
```scad
group() { cube(1,center=true); cube(2,center=true); }
```
(assuming that group doesn't do an implicit union). 
With the optional feature that you could get the inverse of one volume 
```scad
group() { cube(1,center=true); compliment()cube(2,center=true); }
```

doug-moen, 2016-02-23:
@clothbot said The 'subvolumes()' could reduce 'difference() { cube(2,center=true); cube(1,center=true); }' into two volumes, where one is the outer cube(2) and the other is the inner cube(1) empty-space, but I know I would find that behaviour/side-effect confusing. I would treat it as an optional behaviour, not a default behaviour.

That's not what I would expect. I think about OpenSCAD models in terms of the mathematical solids themselves, and not in terms of their mesh representation. It's true that a single continuous solid can be have more than one contiguous boundary if it contains interior voids, but I do not expect an operation named "subvolumes" to dissect a single contiguous solid into its boundaries. Maybe if it was called `boundaries()`.

For me, a good use case for `subvolumes` is to dissect an STL model into multiple disjoint solids. If you create an OpenSCAD model containing multiple disjoint solids that do not touch or overlap, save them to an STL file, then import the STL file, you get a single object. `subvolumes` could be used to recover the original disjoint solids as separate objects. Eg, `subvolumes() import("foo.stl")`. In this use case, if one of the solids has interior voids, then it is treated as a single object by `subvolumes`.

clothbot, 2016-02-23:
@GilesBathgate - those examples of yours are only equivalent in the sense that difference() { small(); big(); } gets you null(); ;-)  I wouldn't expect primitives to be preserved thru the difference() operator, into the subvolumes() operator, only the result(s) of the difference() operation passed.

@doug-moen - agreed. Once we get rid of the implicit union 'feature' the 'import()' operators could be enhanced to explicitly 'subvolume()' the imported geometry.  It would be up to the designer to union() import(); to treat it as a single entity.

Side note: My preference for a better, more explicit operator for accessing interior voids (versus overloading subvolumes()) could be just that - 'voids()'.  I've lost track of how often I've wished I could get a handle on the negative space without explicitly difference-cutout inspecting it.

GilesBathgate, 2016-02-23:
@doug-moen Ahh, at least I now understand @clothbot 's feature now. I thought he just wanted to get access to the children of the union, difference, intersection modules. What he actually wants is to split multiple non intersecting parts. Yes that makes much more sence to have a ```disjoint_decomposition()``` module. Personally I would prefer it if you could come up with something a bit more terse, and easier for the layman to understand. 

Suggestions:

- split()
- parts()
- seperate()
- break_appart()

clothbot, 2016-02-23:
@GilesBathgate - Yeah, it's sort of two features in one issue.  Removing the implicit union behaviour will immediately obsolete the subvolumes() part of this request.  That would leave the convex_decomposition() available in CGAL (hence the name choice to match the implementation+documentation) which is where the fun *really* begins.

Side note: If 'convex_decomposition()' were implemented sooner and not subject to implicit union-ing, I'd be happy with that, since 'convex_decomposition() subvolumes() blob();' would be equivalent to 'convex_decomposition() blob();' in output.

GilesBathgate, 2016-02-23:
@clothbot There is still a difference between separating things that do not touch, and decomposing a concave solid into convex parts though. That's why @doug-moen  suggested disjoint_decomposition() and convex_decomposition().

doug-moen, 2016-02-23:
@clothbot I endorse your idea of an explicit operator for accessing interior voids, called `voids()`.

clothbot, 2016-02-23:
@GilesBathgate - I think we are in agreement... 'separating things that do not touch' is what I mean by 'subvolumes()' operator.  Removing implicit union behaviour would result in 'things that do not touch are separate' without the designer needing to do anything special.

I would recommend against using 'disjoint_decomposition()' as a name, at least the 'decomposition' part of it.  Other references I can find to 'disjoint decomposition' (e.g. https://www.researchgate.net/figure/221357131_fig1_Figure-1-Decomposition-of-a-mesh-into-four-disjoint-partitions-indicated-by-different) look more like CGAL's Surface Segmentation features:
 http://doc.cgal.org/latest/Surface_mesh_segmentation/index.html#Chapter_3D_SurfaceSegmentation

GilesBathgate, 2016-02-23:
@clothbot

>  Removing implicit union behaviour would result in 'things that do not touch are separate' without the designer needing to do anything special.

Yes it would, except for incoming STL's, 3mf etc...

disjoint is the correct term mathematically for sets. But I think we can find a better module name.

doug-moen, 2016-02-23:
@clothbot You said "Once we get rid of the implicit union 'feature' the 'import()' operators could be enhanced to explicitly 'subvolume()' the imported geometry. It would be up to the designer to union() import(); to treat it as a single entity."
* To completely eliminate the need for an explicit `subvolumes` operation, you'd also need `difference` and `intersection` and maybe `polyhedron` to run `subvolumes` on their output.

That seems to make sense for STL, modulo performance concerns--how big a hit is that? If the `subvolumes()` operation is slow, and I'm guessing it is, maybe we don't want to have it always run implicitly.

It makes less sense for AMF files, which have the ability to describe multiple separate objects. It's a feature that I already use with Slic3r. I'd like `import` to preserve the internal structure of an AMF file when importing it, to the extent that this makes sense. (I don't have a fully worked out proposal for AMF multi-object import right now.)

With what I know right now, I'm more comfortable with `subvolumes()` being a separate operation.

GilesBathgate, 2016-02-23:
@doug-moen Yeah you are right about amf and 3mf having separate objects. I guess someone could have an amf/3mf with a two non-intersecting volumes in a single object, or would that be non-conformant to the spec? I guess it would still be useful for repairing files created by broken implementations though.

For non intersecting solids I don't think splitting them would be a slow operation. CGAL NefPolyhedron3 provides these volumes via the ```Volume_const_Iterator```, accessed through ```volumes_begin()```. But that assumes that you've already got a Nef3.

The Slic3r implementation for splitting STL's is fast though.

t-paul, 2016-02-23:
How is the order of the subvolumes defined? If we end up with fillets being the only sensible use case, maybe abstraction level for the feature is at the wrong place?

GilesBathgate, 2016-02-23:
@t-paul Would a spacial sort work? http://doc.cgal.org/latest/Spatial_sorting/index.html#secspatial_sorting

t-paul, 2016-02-23:
I guess with no other means to probe the result any defined sort order would do to at least give stable results. I don't know if indexed access is generally useful as it's probably hard to write a script that anticipates the result for importing different STL files for example.

clothbot, 2016-02-23:
For sorting subvolumes, order could be as simple as sorting by minZ point(s), then minY point(s) then minX point(s).  You could do a coarser/faster bounding-cubesort, but those wouldn't have unique solutions.

Side note: Sorting the output of convex_decompostion() by bounding-cube should be trivial.

GilesBathgate, 2016-02-23:
@t-paul yeah I see your point. I actually intended on creating an auxiliary module for rapcad that would display the indexes visually. But It would still be a two step process e.g.

```scad
index$()decompose()mything();
```
look at results then:

```scad
module mymodule() children(1);

my()decompose()mything();
```

(Although that's a bit clunky I think I still like the children syntax)
```scad
children(1) decompose()mything();
```

GilesBathgate, 2016-02-23:
@t-paul BTW I think fillet (or bevel) should be a built-in module. I think it could work by resizing all the facets using offset, and joining up the edges.

doug-moen, 2016-02-23:
@GilesBathgate Now that you've explained that a Nef_Polyhedron_3 automatically keeps track of disjoint subvolumes, I like the idea of taking advantage of this feature in OpenSCAD. The proposal is that OpenSCAD will also automatically keep track of disjoint subvolumes after all operations, and permit modules to iterate over them.

This feature interacts with other proposed features, such as:
* multi-material support, where a single contiguous volume is decomposed into disjoint subvolumes at the material boundaries, with a material tag attached to each subvolume.
* tagged multi-part models, where you can attach names to subobjects as metadata, and the names are used during export: eg, export each named subobject to a separate STL file, or export an AMF file containing multiple named objects.

So I'm still trying to envision how all this fits together, and what the data model will look like for compound geometric objects.

clothbot, 2016-02-23:
On my drive home I could think of several cases where convex decomp output might share bounding cube min x,y,z coordinate, so I take back the 'trivial' comment.

Throwing a related idea out there: It would be Really Useful to have a selection operator to filter 'children()':

// selected if bounding cube of mask_object fully contains bounding cube of subvolume();
// - use this one for speed and/or trivial cases.
select(filter="bounds") {
  mask_object();
  subvolumes() blobs();
}

// selected if intersection() { mask_object(); subvolume(); } != null();
// - performs 'bounds' filter to remove trivial cases.
// - more computationally intensive but selected when a single 'point inside mask_object()' is detected.
select(filter="intersects") {
  mask_object();
  subvolumes() blobs();
}

// selected if intersection() { mask_object(); subvolume(); } == subvolume();
// - most computationally intensive.
select(filter="contains") {
  mask_object();
  subvolumes() blobs();
}

GilesBathgate, 2016-02-23:
@doug-moen Yeah actually, I do this slightly differently in rapcad, I keep the complete evaluation tree for every operation. I store the two inputs in a QList called children under CGALPrimitive, and return the parent. Its easier to explain by example so take a look at the source: https://github.com/GilesBathgate/RapCAD/blob/master/src/cgalprimitive.cpp#L336
(I assume from your previous comments you are familiar with C++)

This is how rapcad supports multiple materials. I also have just added some code which does the same for the decompose module, but its still WIP: https://github.com/GilesBathgate/RapCAD/blob/master/src/cgalprimitive.cpp#L385

But in principle its certainly possible to do the features you mention, by utilising the Volumes under Nef. See: http://doc.cgal.org/latest/Nef_3/classCGAL_1_1Nef__polyhedron__3_1_1Volume.html

MichaelAtOz, 2016-02-26:
I presume preview performance of all this will be woeful. Injecting some real life...

clothbot, 2016-02-26:
Not necessarily. You get some hooks to subvolumes for free at Render/F6 time (in order to report 'Volumes:'), and convex_decomposition() is a preparation step for at least one form of 3D Minkowski -  http://doc.cgal.org/latest/Minkowski_sum_3/index.html#Minkowski_sum_3Decomposition - so it'll definitely be faster than actually performing the Minkowski on complex geometry.

doug-moen, 2016-02-26:
@MichaelAtOz said "I presume preview performance of all this will be woeful."

Yes, but it should still be faster than preview for minkowski sum (which uses convex decomposition internally). In other words, so what? If preview is too slow, which can easily happen even without these new operations, then you may need to change your code. Eg, try using render().

At this point, I don't think we should be rejecting new shape operators on the basis that they might preview slowly if given complex arguments. As long as we aren't slowing down existing scripts, it's okay.

kintel, 2016-02-26:
Any of my scepticism towards accepting the suggested operations would most likely come from a fear of growing too dependent on CGAL. CGAL's Nef polyhedrons are currently the single worst offender in terms of OpenSCAD performance, and I'm actively seeking alternatives.

GilesBathgate, 2016-02-26:
@kintel You need an implementation of convex_decomposition for your fast minkowski implementation, which as you described to me as decomposing polyhedra into convex pieces, computing pair-wise hull of the convex pieces, and uniting the pair-wise sums. I don't know how you implement it, but if you move away from CGAL it might be worth having an implementation of it anyway.

As for the disjoint decompostion, you just pick a triangle, and walk all connected triangles until all connected triangles have been visited. This is volume one. Then pick an unvisited triangle and repeat the walk. All volumes have been found when the entire list of triangles has been exhausted.

clothbot, 2016-02-26:
@kintel +1 what @GilesBathgate wrote. Choosing the CGAL implementation is a convenience, not a requirement.  The 'subvolumes()' could be obsoleted/redundant if 'render()' output wasn't implicit union'd.

GilesBathgate, 2016-02-27:
@kintel https://github.com/openscad/openscad/issues/1562 has reminded me that you would have to test for voids in the disjoint decompostion. I did just think of the implementation off the top of my head, I imagine the Slic3r source would be a good reference for something more concrete.

thehans, 2021-03-05:
I was recently thinking how useful a `convex_decomposition()` operation would be, but there are still problems with how to access the individual results of such an operation.  Or really just any operation were you may want to individually modify separate geometries.  This issue also happens to generally limit the usefulness of `lazy-union`.

The problem is, with the way `$children` and `children()` currently work, they can't really be used for this purpose to count results of geometric operations.  (see #3143 for specifics) 
Summary: all expressions are basically evaluated placed directly into CSG Tree *before* generation of geometry (expressions include variable lookups such as `$children` needed to instantiate loops!).  After evaluating all expressions, this leaves only builtin module instantiations with literal values for **all** their parameters (the ".csg" format).  Because of this, `$children` has to be based entirely on the count of AST nodes in the source script, not CSG nodes or # of Geometry in a list of geometries, etc.  

There were also some comments as a forum discussion thread leading up to PR #3555 (and comments on the PR itself).  From what I gathered, it sounds like people would prefer not to change the behavior of children *even if* we had some way to do so.  So instead of `children()` I think we will need a whole other terminology and method of access.

The only thing I can come up with, which I *think* could help workaround these sort of issues, would be adding a new type of loop statement:
```openscad
for_geometry() { 
  // geometry to loop over
  convex_decomposition() ...; 
} do {
  // "body" of loop
  perform_some_operations() ...
     geometry();    // reference this iteration's geometry with "geometry()" instead of "children()"
}
```

The `do` section could set a `count` and `index` variable (maybe as special $ variables?), which could be referenced inside the loop.  I'm still not sure if adding variables like that would work with the current flow of evaluation, it would still mean that there are unresolved variables/expressions remaining in the CSG, but at least it provides a clear separation of input and output and hence the order of execution is more clear.  So the `for_geometry` part is evaluated first, **then** the `do`, which is maybe enough?

Some other considerations:
  - Would the names of these variables need to be user specified?
    `for_geometry() ... do(my_count_name, my_index_name)`
    or, specify variable names as strings:
    `for_geometry() ... do(count_name="my_count", index="my_index") { if (my_index == 1) foo(); ... }`
    If the names are not user-specifiable, then nesting of `for_geometry` would not be possible.
  - Maybe it doesn't need to force a loop, as long as the `count` can be communicated in some way?
    So something more like the following:
 ```openscad
with_geometry() {
  convex_decomposition() ...;
} do {
  for(i=[0:1:$count-1]) {
    perform_some_operations()
      geometry(i);
  }
}
```