openscad/openscad #1564: rotate_extrude with positional parameters give misleading warning

kintel, 2016-01-21:


Battlevaria, 2021-12-05:
Just saw this, too, for the first time. I haven't used rotate_extrude a lot, but it hasn't turned up before. It happens when I specify the parameters wrong. Rather than `rotate_extrude(45) ...`, it should be `rotate_extrude(angle=45)`.

jordanbrown0, 2021-12-06:
It seems like it would be an improvement to check the type of the argument.

At a minimum, a string could yield the existing warning while a non-string could yield a type mismatch error.

Better, though, might be to have a number treated as the angle parameter so that positional arguments would appear to work with the current function signature.

jordanbrown0, 2025-03-09:
The old file-import options are gone now, so `rotate_extrude()` **does** expect the angle parameter as its first positional parameter.

closed 2025-03-09 by jordanbrown0