openscad/openscad #1566: preview of 2D thickness needs to be improved

MichaelAtOz, 2016-01-24:
Like #1202 preview of 2D thickness needs to be improved.
Current 2D objects can be confused with 3D objects.
[image]
So a thickness like 0.001 would be better.
Questionable weather it should sit at Z=0 or centered. I think Z=0.
Also debatable whether 2D preview should show the same colour as 2D rendered objects.

nophead, 2016-01-24:
2D preview should look as close as possible to 2D render, otherwise it isn't a preview.

doug-moen, 2016-01-25:
2D preview and 2D render look different, and I agree they should look the
same.

2D objects can't be exported as STL, because they aren't 3D. The
Z-thickness of a 2D object in preview/render is just a presentation issue,
it doesn't affect export. So the question becomes, what's the best way to
present 2D objects in the preview/render views?

I think Michael's ideas about presentation are okay. Give 2D objects a very
small Z-thickness (if zero thickness isn't feasible for some reason). They
should sit at Z=0, not at Z=-1 as in the current render view. Giving them a
distinct colour also makes sense.

nophead, 2016-01-25:
Yes the F6 view is nice because it looks completely different, so you can't
mistake it for 3D. Can preview be the same, i.e. green with red edges?

F5 isn't currently at Z = -1, it's actually Z = -0.1, which is weird.

kintel, 2016-01-25:
@nophead See #1202

GilesBathgate, 2016-02-13:
1. Preview (F5) mode gives 2D primitives a "thickness" but the following shows that its not necessary.

```c#
polyhedron([[0,0,0],[0,10,0],[10,10,0],[10,0,0]],[[0,1,2,3]]);

translate([10,10,0])square(10);
```

2. In Render (F6) mode you can get the square to display by commenting out the polyhedron, or the polyhedron to display by commenting out the square, but never both.

doug-moen, 2016-02-13:
>    1. In Render (F6) mode you can get the square to display by commenting

@GilesBathgate: that call to polyhedron() is bad. It returns a non-manifold
mesh which will cause CGAL assertions if you try to operate on it further.
There has been discussion elsewhere about fixing polyhedron to report bad
arguments.

But yes, its good to point out that preview is already capable of
displaying 2D objects with no thickness.

GilesBathgate, 2016-02-13:
@doug-moen What assertions?

Example 1)
```c#
hull(){
polyhedron([[0,0,0],[0,10,0],[10,10,0],[10,0,0]],[[0,1,2,3]]);
polyhedron([[0,0,10],[0,10,10],[10,10,10],[10,0,10]],[[0,1,2,3]]);
}
```

Example 2)
```c#
difference(){
polyhedron([[0,0,0],[0,10,0],[10,10,0],[10,0,0]],[[0,1,2,3]]);
polyhedron([[2.5,2.5,0],[2.5,7.5,0],[7.5,7.5,0],[7.5,2.5,0]],[[0,1,2,3]]);
}
```
Its not really 2D, and its not really 3D, I call it 2.5D ;)  

Anyway yes the point was that preview can display 2D objects.

nophead, 2016-02-13:
Example 1 doesn't use CGAL. You can tell because you only get

Top level object is a 3D object:

Facets: 12

and show edges doesn't work.

Example 2 gives

Simple: no

Vertices: 8

Halfedges: 16

Edges: 8

Halffacets: 2

Facets: 1

Volumes: 1

WARNING: Object may not be a valid 2-manifold and may need repair!

So CGAL doesn't like it. Sometimes it throws exceptions and sometimes it
just sets Simple:no which triggers the warning.

GilesBathgate, 2016-02-13:
@nophead Actually its OpenSCAD that doesn't like it. I would know, the whole 2D subsystem of RapCAD is build this way. Nef polyhedron can handle boolean set operations on non-manifold polyhedra, that's what makes it so bloody complicated (and slow!).

nophead, 2016-02-13:
Well is is CGAL that throws the exceptions and OpenSCAD that fails to catch them I think.

GilesBathgate, 2016-02-13:
@nophead Yeah that's another problem that I have hopefully fixed now in CGAL: https://github.com/CGAL/cgal/pull/451

nophead, 2016-02-13:
Or rather it catches them and displays them to the user rather than saying something helpful like "operation on non-manifold polyhedron".

GilesBathgate, 2016-02-13:
@nophead No, the "Object may not be a valid 2-manifold and may need repair!" message is generated by checking Nef's N.is_simple() No exception is thrown. Like I said Nef supports operations on these type of primitives, RapCAD takes advantage of this, OpenSCAD choose not to.

nophead, 2016-02-13:
Yes but often you get an exception when CGAL operates on non manifold
geometry from polyhedron or import, just not in your simple example.

GilesBathgate, 2016-02-13:
@nophead True, in-fact @kintel and I recently tried to get to the bottom of this. He has a patch that removes zero-area triangles, which seems to fix a lot of these problems. (I understand its still in testing though)

nophead, 2016-02-13:
IIRC, self intersections and zero area holes also cause exceptions.

GilesBathgate, 2016-02-13:
@nophead Yeah self intersections:

```c#
polyhedron([[0,0,0],[0,10,0],[10,10,0],[10,0,0]],[[0,1,3,2]]);
```
Are definitely bad. Arguably my previous examples are "bad" too, you can't export them, or do anything much with them in OpenSCAD so they should be a warning, error or otherwise filtered out like @doug-moen said. In RapCAD you can linear/rotate extrude them. You can also use the triangulate() module to "fix" self intersections, but the result is a bit unobvious ;)

nophead, 2016-02-13:
>ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion
violation! Expr: pe_prev->is_border() ||
!internal::Plane_constructor::get_plane(pe_prev->facet(),pe_prev->facet()->plane()).is_degenerate()
File:
/data/OpenSCAD/libraries-mingw64-master/mxe-w64/usr/x86_64-w64-mingw32.static/include/CGAL/Nef_3/polyhedron_3_to_nef_3.h
Line: 251

So you say CGAL can handle non-manifold geometry but is it a special subset
of non-manifold? And how are boolean ops defined on things that aren't
proper solids?

GilesBathgate, 2016-02-13:
@nophead The error there is converting from CGAL's Polyherdon3 representation to NefPolyhedron3, So the problem isn't that that shape cannot be represented, its that the construction is wrong. 

Consider the following:

```c#
linear_extrude(1)
polygon([[0,0],[0,10],[10,10],[10,0]],[[0,1,3,2]]);
```

If you could triangulate my self intersecting polyhedron in OpenSCAD, then it could be represented.
It would just end up being the above though in 2.5D

GilesBathgate, 2016-02-13:
@nophead In anwser to your question the boolean operations work as one would expect, the intersection of two lines is a point, the intersection of a sphere and a plane is a circle. etc.

GilesBathgate, 2016-02-13:
[image]

nophead, 2016-02-13:
Lines, points and planes are well defined geometric objects so boolean ops
make sense with them and solids.

difference() {
   linear_extrude(1)
      polygon([[0,0],[0,10],[10,10],[10,0]],[[0,1,3,2]]);

    cylinder();
}

Produces a mess though. It seems to me that to define what boolean ops
should do on arbitrary non-manifold objects would get very complicated very
quickly.

GilesBathgate, 2016-02-13:
hmm, Yet this works:

```c#
difference() {
    union() {
      polyhedron([[0,0,0],[0,10,0],[5,5,0]],[[0,1,2]]);
      polyhedron([[10,0,0],[5,5,0],[10,10,0]],[[0,1,2]]);
	}
    cylinder();
}
```

doug-moen, 2016-02-13:
nophead said: "So you say CGAL can handle non-manifold geometry but is it a special subset of non-manifold? And how are boolean ops defined on things that aren't proper solids?"

I haven't learned RapCAD or CGAL yet, but CGAL supports Nef Polyhedra. And Nef Polyhedra can represent things that are "proper solids" but which are not representable by STL meshes, because the mesh isn't 2-manifold. This is a pretty subtle distinction, but it matters because these non-STL-representable solids are 3D printable. They can't be represented by STL, but they can be represented by 3MF, and probably by AMF (which uses the same mesh representation as 3MF).

The Nef Polyhedron representation is closed with respect to the CSG operations of union, intersection and difference. (That's not true for STL 2-manifold meshes.)

Here are some examples of solids that are Nef- and 3MF-representable, but not STL representable. You can construct these solids by starting with convex polyhedra (proper solids), then using CSG operations:
* two cubes that touch each other along an edge or at a vertex, without intersecting
* a non-convex polyhedron with two arms that reach out, then touch each other along an edge or at a vertex, without intersecting

The reason you can represent these solids in 3MF, but not in STL, is because of the difference in the mesh representation. In 3MF, there is a list of vertices, then there is a separate list of triangles, where each triangle is specified using indexes into the vertex list. To represent these funny solids, you need to store multiple copies of the same vector in the vertex list, so that each copy has a different index. In the case of two cubes sharing an edge `<v1,v2>`, you'll have two copies of the vertices v1 and v2. Each cube will use distinct vertex indices for that shared edge. The end result is that the mesh *is* 2-manifold, if you define 2-manifold in terms of vertex indices, and not in terms of underlying vertex equality.

GilesBathgate, 2016-02-13:
@doug-moen There is no limitation on what STL can represent vs what AMF can represent, ultimately these file formats represent triangles. STL just uses a list of vertexes and repeats those vertexes for each triangle, whereas 3MF and AMF represent a list of vertexes separately and index those vertexes from a list of triangles.

The problem there isn't that they cannot be represented it rather that to get them into STL OpenSCAD first converts them from a NefPolyhedron3 to a Polyhedron3. Its the Polyhedron3 that incurs the Manifold restriction.

Coincident points are not a problem in STL as you can imagine they exist for any pair of touching triangles.

doug-moen, 2016-02-13:
@GilesBathgate:

Let's consider an example:
```
cube(10);
translate([10,10,0]) cube(10);
```
OpenSCAD says `WARNING: Object may not be a valid 2-manifold and may need repair!`

This warning message is correct for STL, but it would not be correct for 3MF (or possibly for AMF) if we had correct code for exporting these kinds of solids to these file formats, in accordance with the 3MF standard. (I don't have full information about AMF. AMF draft version 0.45 explicitly doesn't support these kinds of solids, the published version of AMF is paywalled, and the 3MF standard explicitly does support these kinds of solids.)

The problem is that the exported STL mesh would not be 2-manifold. The STL file would contain 4 triangles sharing the same edge, ([10,10,0], [10,10,10]). Part of the definition of 2-manifold is that each edge is shared by exactly 2 faces.

However, it is possible to construct a 3MF file for this shape, where the mesh is 2-manifold. The 3MF standard is explicit about this. The reason that the 3MF mesh is 2-manifold, is because vertex equality is defined by comparing vertex indices, instead of by comparing [x,y,z] coordinates, and you can have two vertexes with different indexes but the same coordinates.

GilesBathgate, 2016-02-13:
Here is your non manifold shape in STL https://gist.github.com/GilesBathgate/79450ae7f51a990f3883

nophead, 2016-02-13:
>This is a pretty subtle distinction, but it matters because these
non-STL-representable solids are 3D printable.

Although something might come out of a 3D printer it isn't the non-manifold
thing represented by the model because all real solids are manifold. You
will either get two separate cubes or two merged cubes. You won't get two
cubes sharing a single edge as there is no reality to that. Theoretically
they could share a line of atoms but then the junction would one atom
thick, whereas it has zero thickness in the model.

GilesBathgate, 2016-02-13:
I think what most people care about is whether they can load these files into their slicer. The above loads into Slic3r and it seems to think its manifold :/ there is no indication of it having to repair it either.

nophead, 2016-02-13:
Yes but the result of slicing and printing is ambiguous. They will get one
of two possibilities. Better to not allow non-manifold solids to be
exported or imported, Then you either have to have two separate cubes in
the file (not STL obviously) or two overlapping cubes, but not two sharing
an edge. They can be two separate objects with coincident edges but not a
single object with a zero thickness middle. I.e. it is fine to have two
cubes touching on an edge but they should not be unioned. That should give
an error.

nophead, 2016-02-13:
I once helped out at a show printing some give away objects. The files I
was given were non manifold but I didn't notice. When I sliced them and
printed them I though they looked a bit weak but I printed dozens before I
realised they were not what the other guys were printing. Different slicers
gave different results.

kintel, 2016-02-14:
@GilesBathgate See #1580 for an issue dealing with the edge flip mesh repair idea.

doug-moen, 2016-02-14:
A big problem with STL is that there is no standard. There is no authoritative statement of what constitutes a bad STL file, and no statement of how a good STL file should be interpreted. I think the absence of a standard is what makes it possible for @GilesBathgate and @nophead to publicly disagree on whether the STL Giles posted above is valid or not. Likewise, if two slicers interpret the same STL files different ways, then you could say that the STL file is bad, because its ambiguous, or you could say there is a bug in one of the slicers. With no standard to rely on, it's ambigous where the problem is.

For example, there was a recent forum post describing different behaviour between Slic3r and Simplify3D on an STL file containing two intersecting volumes. http://forum.openscad.org/dxf-and-stl-import-rendering-issue-td15979i20.html#a16026

This powerpoint presentation proposes a standard algorithm for interpreting "weird" STL files with arbitrary intersections. It resolves the lack of agreement between Slic3r and Simplify3D mentioned above: http://www.cs.berkeley.edu/~sequin/TALKS/2014_Dagstuhl_WindingNumber.ppt

I've come to appreciate the 3MF standard, because not only does it define what it means for a mesh to be valid, but also (unlike the AMF standard) it defines how meshes are to be interpreted, and in particular, how they are interpreted when they contain multiple 2-manifold surfaces that are nested, that intersect, or that self-intersect. In fact, it standardizes the algorithm proposed in the above powerpoint, which 3MF calls the "positive fill rule", and which the powerpoint calls "The Winding Number Paradigm" (the inside of the model is characterized by a positive winding number).

I suppose it would be nice if OpenSCAD were to use The Winding Number Paradigm as its standard for interpreting weird STL files and converting them into valid solids that can then be operated on using CGAL CSG operations. Ditto for the polyhedron() command.

MichaelAtOz, 2016-02-14:
> You will either get two separate cubes or two merged cubes. You won't get two cubes sharing a single edge as there is no reality to that. Theoretically they could share a line of atoms but then the junction would one atom thick, whereas it has zero thickness in the model.

Given the position of the electron clouds of the two cubes it is a quantum question. Schrodinger's cubes...until you slice them you won't know...

GilesBathgate, 2016-02-14:
@doug-moen Well you are right that the STL standard is weakly defined. However even strong standards are open to mis-interpretation, (whether it be deliberate or accidental) HTML for example!

The problem with the two cubes example is that its not obvious how to determine which is the outside of the surface and which is the inside. This is because at the place where four triangles share an edge, if one pair of triangles are connected then the inside is connected to the outside. If the correct pairs of triangles are connected then the outside is a single surface.

STL solves this problem using facet normals, and 3MF/AMF solve this problem the way you said, by allowing each edge to only share two facets. By encoding which points belong to which triangles this effectively specifies which triangles are connected together.

Looking at the output from Slic3r more carefully it slices this weird STL as two separate cubes, so it would be "weak" as @nophead  points out. If instead it treated it as a single surface it would create a complete outline all the way round and would be stronger.

[image]

nophead, 2016-02-14:
It won't be weakly joined when printed with FFF on a good machine, the
diagram above will print two separate cubes because of the inevitable
corner radius you get with FFF. On a machine with a bit of backlash they
might get weakly welded together but on a perfect machine with finite
filament width they will be separate. On a DLP resin machine they might
overlap by one pixel because if both walls had the same numerical value for
X they should light the same column of pixels.

I still don't see any requirement for OpenScad to handle non-manifold
solids. If you want multiple volumes that touch each other they should be
kept separate. I.e. not unioned and not exported to a single STL but could
be exported to AMF or multiple STLs. Or with the current OpenScad you can
simply separate them by a micron or overlap them by a micro to make the STL
manifold and non ambiguous and your intentions explicit. Exactly touching
should be an error. That will educate the designer to make things manifold
and stop non-manifold STL files getting to Thingiverse where they cause
people who aren't the designer of the object problems.

GilesBathgate, 2016-02-14:
@nophead I never really expected to be able to export non-manifold primitives. My point is that boolean operations on these them do appear to work with CGAL provided you can construct them properly:

Example 1) works
```c#
module thing(){
cube(10);
translate([10,10,0])cube(10);
}

difference(){
thing();
translate([9,9,0])cube([2,2,10]);
}
```

Example 2) does not works.
```c#
difference() {
polyhedron([[0,10,10],[0,0,10],[10,0,10],[10,10,10],[10,0,0],[10,10,0],[20,10,0],[20,10,10],[10,20,10],[20,20,10],[10,20,0],[0,10,0],[0,0,0],[20,20,0]],[[0,1,2,3],[2,4,5,3],[3,5,6,7],[8,3,7,9],[10,5,3,8],[5,11,0,3],[11,12,1,0],[1,12,4,2],[4,12,11,5],[6,5,10,13],[7,6,13,9],[13,10,8,9]]);

translate([9,9,0])cube([2,2,10]);

}
```

nophead, 2016-02-14:
OK thanks I understand now.

Boolean ops on manifold solids are not closed because when you union things
that share a vertex or an edge but not a face you get something like the
above. Nef Polyhedra are a special subset of non-manifoldness that allow
boolean ops to be closed. So basically the can have shared vertices and
edges because you can make those with boolean ops on manifold solids. They
are not representable by manifold triangle meshes so STL import and
polyhedron(...) don't work. Possibly STL import could work using the face
normals and polyhedron could be fixed by considering the winding order.

Presumably other forms on non-manifoldness are not allowed, such as flipped
faces, holes and self intersections. What about internal faces?

GilesBathgate, 2016-02-14:
@nophead Yes that's a much clearer explanation.

I don't know what other forms are not allowed, I have never tried constructing any of those examples. You could not do it through OpenSCAD or RapCAD easily either because both use Polyhedron3 => NefPolyhedron3 in the construction. 

Perhaps the only way to bypass this pipeline would be to export something to the .NEF file format, manually change it, and then re-import. Perhaps that would help us get to the bottom of issue https://github.com/openscad/openscad/issues/802

MichaelAtOz, 2017-02-06:
Well, that was fun re-reading all that...

I'm repurposing this issue:

Feature request: Set 2D preview to 0.001 thickness. Forget about color, that can be another issue at another time.

kintel, 2017-02-08:
@MichaelAtOz We can start a branch for that. We need to sort out the inevitable z fighting issues, but it's trivial to whip up a starting point for testing.

kintel, 2017-02-08:
See #1933

jordanbrown0, 2024-01-16:
The "incomplete polyhedron" example is how 2D objects should be shown.  They might not be polyhedra per se under the covers - they would remain 2D objects - but they should be rendered as zero thickness.  Not 1, not a very small number, zero, made non-zero only by the resolution of the display - what in the 1-D world would be called a hairline.  No matter how much you scale up or zoom in, they should remain zero thickness.

The problem with displaying them that way is that the previewer does 2D boolean operations using 3D mechanisms.  It relies on the 2D object being a 3D object with non-zero thickness.  Instead, it should do the "true" 2D operations to yield the new 2D object, without preview trickery, as (for instance) linear_extrude does.

GilesBathgate, 2024-01-16:
@jordanbrown0 (no color support but...)
[image]

jordanbrown0, 2025-09-01:
Continuing from my comment last year... we could just fully render 2D objects before previewing them, as linear extrusion does before extruding.  The problem there is that a 2D render loses color information.