openscad/openscad #1572: assertion violation

istergiou, 2016-01-27:
scad file:
`
module s_cube(
	x=20,
	y=20,
	z=30,
	angle=45
) {
	tx=z*cos(angle);
	polyhedron(
	  points=[ [x,y,0],[x,0,0],[0,0,0],[0,y,0], // the four points at base
	  		   [x+tx,y,z],[x+tx,0,z],[tx,0,z],[tx,y,z] // the four points at top
	         ], 
	  faces=[ 
	  		[0,3,2,1],
	  		[0,1,5,4],
	  		[1,2,6,5], 
	  		[2,3,7,6], 
	  		[3,0,7,4], 
	  		[4,5,6,7] 
	  		]
	  );	
}

module a() {
	s_cube(x=10,y=10,z=30,a=10);
	mirror([1,0,0]) s_cube(x=10,y=10,z=30,a=10);	
}
a();
`

Error message returned in OpenSCAD version 2016.01.27:
`
/Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD  -o output/scube.stl scube.scad
Error !!!!!!!!!!!!!!!!!!!!!!!
CGAL error: assertion violation!
Expression : pe_prev->is_border() || !internal::Plane_constructor<Plane>::get_plane(pe_prev->facet(),pe_prev->facet()->plane()).is_degenerate()
File       : /Users/kintel/code/OpenSCAD/openscad/../libraries/install/include/CGAL/Nef_3/polyhedron_3_to_nef_3.h
Line       : 251
Explanation:
Refer to the bug-reporting instructions at http://www.cgal.org/bug_report.html
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation!
Expr: pe_prev->is_border() || !internal::Plane_constructor<Plane>::get_plane(pe_prev->facet(),pe_prev->facet()->plane()).is_degenerate()
File: /Users/kintel/code/OpenSCAD/openscad/../libraries/install/include/CGAL/Nef_3/polyhedron_3_to_nef_3.h
Line: 251
Error !!!!!!!!!!!!!!!!!!!!!!!
CGAL error: assertion violation!
Expression : pe_prev->is_border() || !internal::Plane_constructor<Plane>::get_plane(pe_prev->facet(),pe_prev->facet()->plane()).is_degenerate()
File       : /Users/kintel/code/OpenSCAD/openscad/../libraries/install/include/CGAL/Nef_3/polyhedron_3_to_nef_3.h
Line       : 251
Explanation:
Refer to the bug-reporting instructions at http://www.cgal.org/bug_report.html
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation!
Expr: pe_prev->is_border() || !internal::Plane_constructor<Plane>::get_plane(pe_prev->facet(),pe_prev->facet()->plane()).is_degenerate()
File: /Users/kintel/code/OpenSCAD/openscad/../libraries/install/include/CGAL/Nef_3/polyhedron_3_to_nef_3.h
Line: 251
Current top level object is empty.
`

Does not fail if one s_cube is shown.

istergiou, 2016-01-27:
there was an error in the faces, it should be
`
	  faces=[ 
	  		[0,3,2,1],
	  		[0,1,5,4],
	  		[1,2,6,5], 
	  		[2,3,7,6], 
	  		[3,0,4,7], 
	  		[4,5,6,7] 
	  		]
`

Apologies, it was working when one polyhedron was shown and that confused me believing the polyhedron was correct.

closed 2016-01-27 by istergiou

t-paul, 2016-01-27:
Have a look at https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_is_my_model_showing_up_with_F5_but_not_F6.3F which shows how to check polyhedrons.