openscad/openscad #1573: Add $frame, $frames and $fps for animate

clothbot, 2016-01-28:
Prompted by this discussion:

http://forum.openscad.org/Product-Video-produced-with-OpenSCAD-td15783.html

I've often found times when I wish I could access the animate frame number, total number of frames and frames-per-second as integers directly instead of having to parametrize+scale+round the current $t=[0.0,1.0] to accomplish the same.

t-paul, 2021-01-31:
http://forum.openscad.org/Simple-question-is-there-a-variable-for-number-of-steps-in-animation-td31807.html

Also suggesting `$time` relative to start of animation.

jordanbrown0, 2025-09-01:
Rather than more and more top-level $ variables, perhaps there should be a $animation object with frame, frames, fps, time members.

But:  should it be possible for the program to *set* the FPS and number of frames, to control the defaults?

jordanbrown0, 2025-09-02:
Indeed... variables imply set-ability, and $ variables are a mismatch for global values because of their scope.

closed 2026-11-06 by github-actions[bot]