openscad/openscad #1574: Fails silently on circle();

TT44bb, 2016-01-31:
I wanted to add a 2d circle, so typed:
circle(radius=5);
and then spent at least 15 minutes, probably more, debugging why the hell my structure based on that circle was behaving in a strange way.
Seriously? I can hardly believe it, am I missing something? Why the hell is there no error? Is there some verbose or warning flag I should add? I couldn't find anything.

(for

kintel, 2016-01-31:
This could be generalized into: We should give warning on attempting to instantiate any module with module parameters which are not declared or used by the module in question.
The implementation for built-in modules are likely a bit different than for user-defined modules so that's probably two separate tasks.

Such warnings would be a lot more effective if displayed in context. I'll open a separate issue on context-sensitive warnings and errors as that requires parser changes.

doug-moen, 2016-01-31:
circle(radius=5) doesn't work. But OpenSCAD doesn't report errors, it
silently ignores them. Good luck if you need to debug something. Posting
your code to the forum is the workaround.

Try this: circle(r=5)

MichaelAtOz, 2016-01-31:
> This could be generalized into: We should give warning on attempting to instantiate any module with module parameters which are not declared or used by the module in question.

Noting that such use with $variables is valid to affect sub modules for example. May want that warning as a preference option, on by default for beginners.

Perhaps only warn when a default is encountered in addition to unreferenced parameter? Fits this use case. Or review the use of defaults, I can't see much use for them, and most code is unlikely to use them, I only do for debugging.

doug-moen, 2016-01-31:
"Perhaps only warn when a default is encountered in addition to
unreferenced parameter?"

This would not have addressed the problem when jpmendes tried to use the
angle= parameter of rotate_extrude, as documented in the wiki, and it was
silently ignored. (He wasn't using the unreleased DEV version of OpenSCAD.)
It would have been better to issue an error message in this case.

See:
http://forum.openscad.org/This-seems-to-be-ultra-basic-I-cannot-have-the-rotate-extrude-angle-parameter-working-tp15180.html

TakeItAndRun, 2016-02-01:
Maybe this would be something for the google summer of code:
Generate consistent error/warning messages for all (internal) calls.
So there are no more silent fails, at all.

TT44bb, 2016-02-01:
Thank you.
When opening this issue I wasn't sure if the problem wasn't entirely me after all, so I didn't generalize.
Now that I know some more I tried splitting the issues up and thinking of how they could be solved and what the downsides to these approaches would be.

1) If passing values to modules like circle(someUndefinedAndNeverUsedValue=127) is valid, then I suggest that parameters of geometric primitives should have no default value unless they are strictly optional. I don't know if there is an implicit 0 for parameters with no defaults which aren't supplied with the call or if there is an error/warning when they are missing.
An implicit 0 would already be much better than the current state, because that would likely lead to non-visible/existant objects which are easier to spot than objects with faulty dimensions.
But errors/warnings would be better.

This could break existing projects which make use of those default values. If module calls with insufficient parameters would lead to a warning (containing file/line) after the change, then I think this would be acceptable, but what do I know.

2) Allowing to pass through values like circle(someUndefinedAndNeverUsedValue=127) seems like a design choice to me rather than a bug or anything. But being used to C-like languages I still dislike the possibility and would like an option to a) throw warnings for every place it is used or b) make those warning fatal errors, preferably both on a per file basis. Per module (in it's definition) also seems useful, but more so for developers of modules than for basic users.
The warnings/errors would need to mention file/line numbers and the names of the variables which are passed but are not defined parameters. Also including the name of the module which was called that way would maybe improve readability, but is not strictly necessary.

Should we close this issue and open new ones for 1) and 2)?

doug-moen, 2016-02-01:
TakeItAndRun said: "Generate consistent error/warning messages for all
(internal) calls. So there are no more silent fails, at all."

Yes, I want the problem fixed in general, not just for specific cases.

What I want is for OpenSCAD to abort the program and print an error message
(not print a warning and keep going) when any of the following conditions
are detected:
* Too many positional arguments are passed, to a function or module, built
in or user defined. For example, abs(5,10) -- somewhat absurb, or
cube(5,10,15) -- a common typo that I was prone to as a beginner.
* Too few positional arguments are passed. For example, abs() or cube().
For a user defined function or module, if the argument does not have an
explicit default value, then it is required.
* A keyword argument with a bad name is passed. The name is bad if it isn't
defined by the function or module. The only exception is "dynamically
scoped" variable names beginning with "$".
* An argument with the wrong type is passed. For example, abs("foo") or
cube("foo").

These changes will have two benefits:
1. It's easier to find errors in a program if OpenSCAD tells you when there
is an error.
2. Right now, it's risky to add new features to OpenSCAD, because virtually
all changes are breaking changes. All extensions to the language will
change the behaviour of some working program (which may have unreported
errors, but still computes the geometry that the user wants). If OpenSCAD
reports errors by aborting the program, then we gain the ability to extend
the language without breaking existing programs. We can add new behaviour
that does something useful in cases where we previously aborted the program
and reported an error.

The corollary, of course, is that this is a breaking change.

GilesBathgate, 2016-02-13:
@doug-moen  Some compilers have a command line flag such as -Werror which turns all warnings into errors. Such a flag or user preference would make the change non-breaking.

nophead, 2016-02-13:
Do we need to worry about breaking already broken scripts? I.e. why would anybody rely on passing a value that is ignored intentionally? If I found it broke any of my scripts I would be pleased to find there was a bug and fix it. The fixed version would still work with older versions of OpenScad so I don't see a compatibility issue.

If you consider backwards compatibility on broken scripts as well as valid ones it is almost impossible to move forwards.

GilesBathgate, 2016-02-13:
@nophead You are right actually it makes more sense if anything to have a command line switch that can be used to ignore the error and treat it as a warning instead. The reason I think compilers do this is that you might want to upgrade to a new compiler to utilise other features but not have to fix all your broken code. Obviously you should fix your broken code otherwise how can you be sure the compiler isn't making assumptions about what you've written that are incorrect, but at least it gives people a more gentle migration path.

With OpenSCAD codebases, perhaps they are not as large and so it presents less of a problem. The Mendel90 codebase is the biggest I've seen ;)

nophead, 2016-02-13:
It would take no time at all to fix any extra parameters because one could simply remove them for no difference in behaviour. OpenScad seems overly cautious changing trivial things like this but quite happy to make changes that give different numerical results in the geometry that take many hours to investigate and fix or accept.

GilesBathgate, 2016-02-13:
@nophead I feel your pain, I recently suggested that parameters such as this should be made consistent. 

The options would be:
- All single letter parameters have full word overloads. e.g. add "radius" where there is currently "r"
- All full word parameters should have single letter overloads. e.g. add "c" where there is currently "center"
- Ensure that all parameters have both single letter and word overloads. e.g both of the above.

@t-paul Said this would cause confusion since you could then do ```cube(10,center=true,c=false);```

doug-moen, 2016-02-13:
Thanks Giles. I will amend my proposal so that it's an error if the same keyword parameter appears twice in a function or module call.

GilesBathgate, 2016-02-13:
@doug-moen What about: ```cube(-10);``` Should that be a warning?

doug-moen, 2016-02-13:
Thanks @GilesBathgate. Yes, `cube(-10)` should be an error. I'll amend the proposal.

doug-moen, 2016-11-20:
In the forum, @MichaelAtOz mentioned "write.scad" as an example of an old broken script (on thingiverse) that would be affected by these changes.

> this will break a. write.scad, AFAIK
> probably one of the more wide spread libraries, b. all those others which
> took write.scad as an example.
> 
> At most 'fixing' should be a warning.
>
> Write.scad extract, one of many examples:
> Call:
> ```
> writecircle(text,where+[0,0,height/2],radius-h,rotate=rotate,font=font,h=h,t=t,
>                 space=space,east=east,west=west,middle=middle,ccw=ccw);
> ```
> Definition:
> ```
> module writecircle(text,where,radius){
> ```

MichaelPFrey, 2019-01-30:
doug-moen has some relevant points, so I go trough some of them:
   * `cube(5,10,15);` creates a warning, so that is resolved,
   * `echo(abs(5,10));` results in undef - that might as well cause a warning
   * `cube();` is somewhat allowed. Some people use openscad as a voxel engine
   * `cube(10,center=true,center=false);` see #2742
   * `cylinder(r=1,r1=2,r2=3);` causes `WARNING: Cylinder parameters ambiguous, in file ./, line 1`. Note that due to d also beeing allowed, the matrix for warnings can be made very complex. To avoid that, the warning currently does not specify the precise problem.
   * `echo(abs("foo"));` causes `undef`
   * `cube("foo");` causes `WARNING: Unable to convert cube(size="foo", ...) parameter to a number or a vec3 of numbers, in file ./, line 1`
   * `cube(-10);` causes `WARNING: cube(size=-10, ...), in file ./, line 1`
So, the check for built in modules are in pretty good shape, but the built in functions are not up to the same standard. I think I will dwell a bit on that and then break down this issue into more manageable tasks.

MichaelPFrey, 2019-01-31:
write.scad: The definition has to change from
```module writecircle(text,where,radius){```
to
```module writecircle(text, where, radius, rotate=rotate, font=font, h=h, t=t, space=space, east=east, west=west, middle=middle, ccw=ccw){```

It reads funny at first, but it is consistent. middle= is the parameter. =middle is the default value from the library.

t-paul, 2020-02-16:
Fixed via #2638.

closed 2020-02-16 by t-paul