openscad/openscad #1575: Cannot build 2015.03-2 with gcc-c++ 6.0.0

hroncok, 2016-02-03:
Full build log: https://kojipkgs.fedoraproject.org/work/tasks/1021/12801021/build.log
Versions of stuff installed: https://kojipkgs.fedoraproject.org/work/tasks/1021/12801021/root.log

I think this is caused by gcc-c++ 6.0.0.

Error message:

```
/usr/include/c++/6.0.0/type_traits:143:12: error: invalid use of incomplete type 'std::conditional<true, std::is_constructible<CGAL::Plane_3<CGAL::Simple_cartesian<double> >, CGAL::Plane_3<CGAL::Simple_cartesian<double> >&&>, std::is_constructible<CGAL::Sphere_3<CGAL::Simple_cartesian<double> >, CGAL::Sphere_3<CGAL::Simple_cartesian<double> >&&> >::type {aka struct std::is_constructible<CGAL::Plane_3<CGAL::Simple_cartesian<double> >, CGAL::Plane_3<CGAL::Simple_cartesian<double> >&&>}'
     struct __and_<_B1, _B2>
            ^~~~~~~~~~~~~~~~
/usr/include/c++/6.0.0/type_traits:1115:12: note: declaration of 'std::conditional<true, std::is_constructible<CGAL::Plane_3<CGAL::Simple_cartesian<double> >, CGAL::Plane_3<CGAL::Simple_cartesian<double> >&&>, std::is_constructible<CGAL::Sphere_3<CGAL::Simple_cartesian<double> >, CGAL::Sphere_3<CGAL::Simple_cartesian<double> >&&> >::type {aka struct std::is_constructible<CGAL::Plane_3<CGAL::Simple_cartesian<double> >, CGAL::Plane_3<CGAL::Simple_cartesian<double> >&&>}'
     struct is_constructible
            ^~~~~~~~~~~~~~~~
In file included from /usr/include/c++/6.0.0/bits/stl_algobase.h:64:0,
                 from /usr/include/c++/6.0.0/bits/stl_tree.h:63,
                 from /usr/include/c++/6.0.0/set:60,
                 from src/CGAL_workaround_Mark_bounded_volumes.h:27,
                 from src/cgal.h:23,
                 from src/cgalutils.h:3,
                 from src/cgalutils.cc:3:
```

hroncok, 2016-02-03:
Reported to CGAL issue tracker as well as https://github.com/CGAL/cgal/issues/726

GilesBathgate, 2016-02-03:
Seems a bit unfair to report it to CGAL when the error is coming from CGAL_workaround_Mark_bounded_volumes.h which is not part of the CGAL project. (unless I am mistaken)

hroncok, 2016-02-03:
Closed the CGAL issue.

kintel, 2016-02-04:
I'm not sure why we wouldn't be allowed to store a Volume_handle in a set.
It looks similar to this bug: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=65760

We could write a stand-alone minimal example storing a Volume_handle of a cube or smth. in a set (or any other container) and submit that to CGAL. Perhaps that opens some doors.

Submitting this as a gcc bug would require us to reproduce without CGAL, which might be a bit of work..

kintel, 2016-02-04:
@hroncok Btw., do you know of any easy way to get a VM image with a minimal gcc-6 distro?

GilesBathgate, 2016-02-04:
Just for reference [CGAL_workaround_Mark_bounded_volumes.h](https://github.com/openscad/openscad/blob/master/src/CGAL_workaround_Mark_bounded_volumes.h)

hroncok, 2016-02-07:
@kintel Would Docker work? You can get the rawhide image. https://hub.docker.com/r/library/fedora/tags/

hroncok, 2016-02-07:
Or https://fedoraproject.org/wiki/Releases/Rawhide#Nightly_images

djdelorie, 2016-03-29:
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=70437

hroncok, 2016-04-05:
This has been fixed in upstream gcc.

t-paul, 2016-04-05:
Nice, thanks for the follow-up.

closed 2016-04-05 by t-paul

hroncok, 2016-04-10:
And now in Fedora. Thanks for all suggestions and help.