openscad/openscad #1576: Variables defined within a group() {} doesn't work

kintel, 2016-02-04:
From the [mailing list](http://forum.openscad.org/dxf-and-stl-import-rendering-issue-tp15979p16057.html):

```
a = 1;

group() {
    a = 2;
    b = 3;
    echo(a,b);
}

echo(a);
```

WARNING: Ignoring unknown variable 'b'.
ECHO: 1, undef
ECHO: 1

I expected group to force { ...} to make an new scope but it does something weird instead. However union() does force a new scope:

```
a = 1;

union() {
    a = 2;
    b = 3;
    echo(a,b);
}

echo(a);
```

ECHO: 2, 3
ECHO: 1

runsun, 2016-02-08:
Could be very useful. Thx.

It seems many other group-ish commands work, too, like hull, translate,  or
even color()

thehans, 2018-03-04:
Handled by #2323, closing.

closed 2018-03-04 by thehans