openscad/openscad #1578: Subtraction behaves in weird way

MarkoRamius, 2016-02-11:
I had an issue when I tried to write a parametric script. After following my initial suspicion, which was completely off, I think I narrowed the faulty behaviour down to the minus sign in mathematical expressions.

Test the following example script. If I insert `r=A` into the call of sphere(), I get a sphere, however, if I insert `r=B`, then the viewing area remains blank.
The difference between A and B is the number of minus signs. Is this behaviour intended, and if so is it documented?

```
X = 3.0;
Y = 4.2;

A = (X -- 2.1*Y); // works
B = (X - 2.1*Y); // does not work

// insert A or B
sphere(r=A);
```

I am using version 2015.03-1 .

nophead, 2016-02-11:
B = (3.0 - 2.1 * 4.2) = 3.0 - 8.82 = -5.82.  You can't have a sphere with a
negative radius.

A is 3.0 - -8.82 which is 3.0 + 8.82 = 11.82, so you get a sphere.

So it all looks correct. What values for A and B were you expecting?

MarkoRamius, 2016-02-11:
1. I guess I am being stupid

closed 2016-02-11 by MarkoRamius