openscad/openscad #1580: Simple mesh repair

kintel, 2016-02-14:
This is a meta issue tracking various issues related to STL import and export, which could be fixed by simple mesh repair by using edge flips to eliminate zero area triangles.

See the [edgeflip branch](https://github.com/openscad/openscad/tree/edgeflip) for an initial, unvalidated test implementation.

TODO:
* [ ] Validate STL output
* [ ] Test STL input
* [ ] Collect issues that fully or partially overlap this issue
* [ ] Move, rename, or otherwise organize tests to fit this issue

GilesBathgate, 2016-02-14:
@kintel  I found that instead of flipping an edge removing the two shortest edges of the triangle also worked. I am trying to envisage the strengths and pitfalls of both approaches.

nophead, 2016-02-14:
I think, as ever, removing degenerate triangles is more complicated than it
seems. I found a paper on it here:
https://www.graphics.rwth-aachen.de/media/papers/slicing1.pdf

There are basically two types: caps and needles.

GilesBathgate, 2016-02-14:
@nophead I think needles are what I call Zero-Length edges, and Caps are what I call Zero-Area triangles. As I understand removing triangles is not simple since you have to re-orient them all with respect to their neighbouring triangles to preserve the winding order. However, CGAL's Halfedge data structure makes this a doddle. 

There are operations on the Polyhedron P such as

P.flip_edge(h) 
Which flips the edge given by h
[image]

P.join_facet(h)
Which removes the edge given by h.

If you remove only one edge you will get a quad, so you need to remove two edges to preserve triangles.

nophead, 2016-02-14:
I don't see how flipping edges preserves the shape as in your diagram the top and bottom triangles could be far from planar to each other, in which case the edge flipped version is radically different. Could be concave rather than convex for instance.

GilesBathgate, 2016-02-14:
@nophead Yes, but that's not the case with a Zero-area triangle. In a Zero-area triangle all points are collinear, so it cannot form a concave surface with its neighbouring triangle.

nophead, 2016-02-14:
I think that is true if one is a cap but not if a pair of needles, although in that case you can merge the vertices and remove two edges. Looking at @kintel's implementation it looks like it flips regardless and then looks to see if it can remove a vertex. Not sure if that is right. Shouldn't it do the former for a cap and the later for a needle?

GilesBathgate, 2016-02-14:
Well... @kintel told me that he fixes up the Zero-Length edges (or needles if you prefer) in another part of the code. In My implementation I do it all in the construction of NefPolyhedron3 https://github.com/GilesBathgate/RapCAD/commit/1ab04434009474c3fe74217f927385951a60ff25

kintel, 2016-02-14:
@nophead The latter part of the code attempts to deal with two back-to-back caps. Not sure yet if that can happen in practice. This is why we need a bunch of testing on this one.

GilesBathgate, 2016-02-14:
@kintel I have some tests (see the above commit), but these all work in the latest version of OpenSCAD. I guess you fix bad input in polyhedron() elsewhere? To use these tests with your mesh repair they would have to manually be converted into STL?

GilesBathgate, 2016-02-14:
@kintel I have created the following test with back-to-back "cap" triangles.

https://gist.github.com/GilesBathgate/7881250e87960ef79a61#file-back-to-back-stl

When you import this into OpenSCAD (master) with the following:

```c#
import("back-to-back.stl");
cube(1);
```

You get the error as expected:

ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: pe_prev->is_border() || !internal::Plane_constructor::get_plane(pe_prev->facet(),pe_prev->facet()->plane()).is_degenerate() File: /usr/include/CGAL/Nef_3/polyhedron_3_to_nef_3.h Line: 263 

I haven't tried the same test with the edge flip branch yet...

nophead, 2016-02-14:
Netfabb thinks it has a hole, rather than degenerate triangles. To fix it one must close the hole first and then remove degenerate triangles (that it still doesn't show). The result then gains two vertices and four more edges. It seems very confused by such a simple file!

GilesBathgate, 2016-02-14:
@nophead My edge removal algorithm in RapCAD seems to work with this example, however I think in creating the example I have realised one pitfall, It creates a concave pentagon!

GilesBathgate, 2016-02-14:
Another thing... with this scad https://gist.github.com/GilesBathgate/7881250e87960ef79a61#file-back-to-back-scad Set d=1 and e=0, you get internal facets, which causes a cgal assertion error.

GilesBathgate, 2016-02-14:
@kintel Oh I forgot to mention I don't think the following code will ever execute:

```c++
if (h->next()->vertex() == h->vertex()) 
 ...
```

You are just comparing the Vertex_handles, where I think you need to instead do:

```c++
if (h->next()->vertex()->point() == h->vertex()->point()) 
 ...
```

GilesBathgate, 2016-02-14:
I've realised that the STL was an unfair test since there was a co-incident point, and so it suffers from the issue as pointed out by @doug-moen on issue https://github.com/openscad/openscad/issues/1566 

I've updated the STL and scad file accordingly (see the diff)

kintel, 2016-02-17:
@GilesBathgate I've been testing a bit: 
* CGAL (4.6.3) seems to be relatively ok with receiving zero area triangles. The simplest test case I've found is the STL referenced from https://github.com/openscad/openscad/blob/master/testdata/scad/bugs/issue945e.scad
* Your breaking ```back-to-back``` example is, as you pointed out, not manifold when read from STL and the resulting Polyhedron_3 ends up not being closed (this likely depends on how the Polyhedron is built and possibly on order of faces). Doing an edge flip on a non-manifold is a bit more tricky - haven't attempted that yet.
* Edge flipping is more work than expected, as we may need to run this in multiple passes as edge flips could create new zero area triangles, possibly infinitely. I've got some test code but need more tests. If you have any good tests in mind I'd be happy to give it a spin. See ```issue1580*``` in https://github.com/openscad/openscad/tree/master/testdata/scad/bugs

stefano2734, 2026-09-20:
Is this solved or improved after 10 years?
CGAL 4.6.3 is very old to actual CGAL 6.2.1.

jordanbrown0, 2026-09-20:
> Is this solved or improved after 10 years?

No.