openscad/openscad #1582: polygon() discards points

engblom, 2016-02-15:
Somewhere around 60-80 points, polygon() begin to discard points and the higher it goes, the less points will be drawn. For example 1000 points becomes just 8 points.

Here is a ready example with 1000 points that becomes just 8 points:

P=[[0, 0], [0.002002, 0.003003], [0.004004, 0.00600597], [0.00600601, 0.0090089], [0.00800801, 0.0120118], [0.01001, 0.0150145], [0.012012, 0.0180172], [0.014014, 0.0210196], [0.016016, 0.024022], [0.018018, 0.0270241], [0.02002, 0.030026], [0.022022, 0.0330277], [0.024024, 0.0360291], [0.026026, 0.0390302], [0.028028, 0.042031], [0.03003, 0.0450315], [0.032032, 0.0480316], [0.034034, 0.0510313], [0.036036, 0.0540307], [0.038038, 0.0570295], [0.04004, 0.060028], [0.042042, 0.0630259], [0.044044, 0.0660233], [0.046046, 0.0690203], [0.048048, 0.0720166], [0.0500501, 0.0750124], [0.0520521, 0.0780076], [0.0540541, 0.0810021], [0.0560561, 0.083996], [0.0580581, 0.0869892], [0.0600601, 0.0899818], [0.0620621, 0.0929736], [0.0640641, 0.0959646], [0.0660661, 0.0989549], [0.0680681, 0.101944], [0.0700701, 0.104933], [0.0720721, 0.107921], [0.0740741, 0.110908], [0.0760761, 0.113894], [0.0780781, 0.116879], [0.0800801, 0.119863], [0.0820821, 0.122847], [0.0840841, 0.125829], [0.0860861, 0.12881], [0.0880881, 0.13179], [0.0900901, 0.13477], [0.0920921, 0.137748], [0.0940941, 0.140725], [0.0960961, 0.1437], [0.0980981, 0.146675], [0.1001, 0.149649], [0.102102, 0.152621], [0.104104, 0.155592], [0.106106, 0.158562], [0.108108, 0.16153], [0.11011, 0.164498], [0.112112, 0.167464], [0.114114, 0.170428], [0.116116, 0.173391], [0.118118, 0.176353], [0.12012, 0.179314], [0.122122, 0.182273], [0.124124, 0.18523], [0.126126, 0.188186], [0.128128, 0.19114], [0.13013, 0.194093], [0.132132, 0.197045], [0.134134, 0.199995], [0.136136, 0.202943], [0.138138, 0.205889], [0.14014, 0.208834], [0.142142, 0.211777], [0.144144, 0.214719], [0.146146, 0.217658], [0.148148, 0.220596], [0.15015, 0.223533], [0.152152, 0.226467], [0.154154, 0.2294], [0.156156, 0.23233], [0.158158, 0.235259], [0.16016, 0.238186], [0.162162, 0.241111], [0.164164, 0.244034], [0.166166, 0.246955], [0.168168, 0.249874], [0.17017, 0.252791], [0.172172, 0.255706], [0.174174, 0.258619], [0.176176, 0.26153], [0.178178, 0.264439], [0.18018, 0.267346], [0.182182, 0.27025], [0.184184, 0.273152], [0.186186, 0.276052], [0.188188, 0.27895], [0.19019, 0.281845], [0.192192, 0.284739], [0.194194, 0.28763], [0.196196, 0.290518], [0.198198, 0.293404], [0.2002, 0.296288], [0.202202, 0.29917], [0.204204, 0.302049], [0.206206, 0.304925], [0.208208, 0.307799], [0.21021, 0.310671], [0.212212, 0.31354], [0.214214, 0.316406], [0.216216, 0.31927], [0.218218, 0.322132], [0.22022, 0.32499], [0.222222, 0.327846], [0.224224, 0.3307], [0.226226, 0.33355], [0.228228, 0.336398], [0.23023, 0.339244], [0.232232, 0.342086], [0.234234, 0.344926], [0.236236, 0.347762], [0.238238, 0.350596], [0.24024, 0.353428], [0.242242, 0.356256], [0.244244, 0.359081], [0.246246, 0.361904], [0.248248, 0.364723], [0.25025, 0.367539], [0.252252, 0.370353], [0.254254, 0.373163], [0.256256, 0.375971], [0.258258, 0.378775], [0.26026, 0.381576], [0.262262, 0.384374], [0.264264, 0.387169], [0.266266, 0.389961], [0.268268, 0.392749], [0.27027, 0.395534], [0.272272, 0.398316], [0.274274, 0.401095], [0.276276, 0.403871], [0.278278, 0.406643], [0.28028, 0.409411], [0.282282, 0.412177], [0.284284, 0.414939], [0.286286, 0.417697], [0.288288, 0.420453], [0.29029, 0.423204], [0.292292, 0.425952], [0.294294, 0.428697], [0.296296, 0.431438], [0.298298, 0.434176], [0.3003, 0.43691], [0.302302, 0.43964], [0.304304, 0.442367], [0.306306, 0.44509], [0.308308, 0.447809], [0.31031, 0.450525], [0.312312, 0.453237], [0.314314, 0.455945], [0.316316, 0.45865], [0.318318, 0.46135], [0.32032, 0.464047], [0.322322, 0.46674], [0.324324, 0.469429], [0.326326, 0.472114], [0.328328, 0.474796], [0.33033, 0.477473], [0.332332, 0.480146], [0.334334, 0.482816], [0.336336, 0.485481], [0.338338, 0.488142], [0.34034, 0.490799], [0.342342, 0.493453], [0.344344, 0.496102], [0.346346, 0.498746], [0.348348, 0.501387], [0.35035, 0.504024], [0.352352, 0.506656], [0.354354, 0.509284], [0.356356, 0.511908], [0.358358, 0.514527], [0.36036, 0.517142], [0.362362, 0.519753], [0.364364, 0.52236], [0.366366, 0.524962], [0.368368, 0.52756], [0.37037, 0.530153], [0.372372, 0.532742], [0.374374, 0.535326], [0.376376, 0.537906], [0.378378, 0.540481], [0.38038, 0.543052], [0.382382, 0.545618], [0.384384, 0.54818], [0.386386, 0.550737], [0.388388, 0.553289], [0.39039, 0.555837], [0.392392, 0.55838], [0.394394, 0.560918], [0.396396, 0.563452], [0.398398, 0.56598], [0.4004, 0.568504], [0.402402, 0.571024], [0.404404, 0.573538], [0.406406, 0.576047], [0.408408, 0.578552], [0.41041, 0.581052], [0.412412, 0.583546], [0.414414, 0.586036], [0.416416, 0.588521], [0.418418, 0.591001], [0.42042, 0.593475], [0.422422, 0.595945], [0.424424, 0.59841], [0.426426, 0.600869], [0.428428, 0.603323], [0.43043, 0.605773], [0.432432, 0.608217], [0.434434, 0.610656], [0.436436, 0.613089], [0.438438, 0.615518], [0.44044, 0.617941], [0.442442, 0.620358], [0.444444, 0.622771], [0.446446, 0.625178], [0.448448, 0.62758], [0.45045, 0.629976], [0.452452, 0.632367], [0.454454, 0.634753], [0.456456, 0.637133], [0.458458, 0.639507], [0.46046, 0.641876], [0.462462, 0.64424], [0.464464, 0.646598], [0.466466, 0.64895], [0.468468, 0.651297], [0.47047, 0.653638], [0.472472, 0.655974], [0.474474, 0.658303], [0.476476, 0.660628], [0.478478, 0.662946], [0.48048, 0.665259], [0.482482, 0.667565], [0.484484, 0.669866], [0.486486, 0.672162], [0.488488, 0.674451], [0.49049, 0.676734], [0.492492, 0.679012], [0.494494, 0.681284], [0.496496, 0.683549], [0.498498, 0.685809], [0.500501, 0.688063], [0.502503, 0.690311], [0.504505, 0.692552], [0.506507, 0.694788], [0.508509, 0.697017], [0.510511, 0.699241], [0.512513, 0.701458], [0.514515, 0.703669], [0.516517, 0.705874], [0.518519, 0.708073], [0.520521, 0.710265], [0.522523, 0.712452], [0.524525, 0.714632], [0.526527, 0.716805], [0.528529, 0.718973], [0.530531, 0.721134], [0.532533, 0.723288], [0.534535, 0.725436], [0.536537, 0.727578], [0.538539, 0.729713], [0.540541, 0.731842], [0.542543, 0.733964], [0.544545, 0.73608], [0.546547, 0.73819], [0.548549, 0.740292], [0.550551, 0.742388], [0.552553, 0.744478], [0.554555, 0.746561], [0.556557, 0.748637], [0.558559, 0.750706], [0.560561, 0.752769], [0.562563, 0.754825], [0.564565, 0.756874], [0.566567, 0.758917], [0.568569, 0.760952], [0.570571, 0.762981], [0.572573, 0.765003], [0.574575, 0.767018], [0.576577, 0.769026], [0.578579, 0.771027], [0.580581, 0.773022], [0.582583, 0.775009], [0.584585, 0.776989], [0.586587, 0.778962], [0.588589, 0.780929], [0.590591, 0.782888], [0.592593, 0.78484], [0.594595, 0.786785], [0.596597, 0.788722], [0.598599, 0.790653], [0.600601, 0.792576], [0.602603, 0.794492], [0.604605, 0.796401], [0.606607, 0.798303], [0.608609, 0.800197], [0.610611, 0.802084], [0.612613, 0.803964], [0.614615, 0.805836], [0.616617, 0.807701], [0.618619, 0.809559], [0.620621, 0.811409], [0.622623, 0.813251], [0.624625, 0.815086], [0.626627, 0.816914], [0.628629, 0.818734], [0.630631, 0.820547], [0.632633, 0.822352], [0.634635, 0.824149], [0.636637, 0.825939], [0.638639, 0.827721], [0.640641, 0.829495], [0.642643, 0.831262], [0.644645, 0.833021], [0.646647, 0.834772], [0.648649, 0.836515], [0.650651, 0.838251], [0.652653, 0.839978], [0.654655, 0.841698], [0.656657, 0.84341], [0.658659, 0.845115], [0.660661, 0.846811], [0.662663, 0.848499], [0.664665, 0.85018], [0.666667, 0.851852], [0.668669, 0.853516], [0.670671, 0.855172], [0.672673, 0.856821], [0.674675, 0.858461], [0.676677, 0.860093], [0.678679, 0.861717], [0.680681, 0.863332], [0.682683, 0.86494], [0.684685, 0.866539], [0.686687, 0.86813], [0.688689, 0.869713], [0.690691, 0.871288], [0.692693, 0.872854], [0.694695, 0.874412], [0.696697, 0.875962], [0.698699, 0.877503], [0.700701, 0.879036], [0.702703, 0.88056], [0.704705, 0.882076], [0.706707, 0.883583], [0.708709, 0.885082], [0.710711, 0.886573], [0.712713, 0.888055], [0.714715, 0.889528], [0.716717, 0.890993], [0.718719, 0.892449], [0.720721, 0.893896], [0.722723, 0.895335], [0.724725, 0.896765], [0.726727, 0.898186], [0.728729, 0.899599], [0.730731, 0.901003], [0.732733, 0.902398], [0.734735, 0.903784], [0.736737, 0.905162], [0.738739, 0.90653], [0.740741, 0.90789], [0.742743, 0.909241], [0.744745, 0.910583], [0.746747, 0.911916], [0.748749, 0.91324], [0.750751, 0.914555], [0.752753, 0.91586], [0.754755, 0.917157 [... cut ...] ], [1.95596, 0.0660233], [1.95796, 0.0630259], [1.95996, 0.060028], [1.96196, 0.0570295], [1.96396, 0.0540307], [1.96597, 0.0510313], [1.96797, 0.0480316], [1.96997, 0.0450315], [1.97197, 0.042031], [1.97397, 0.0390302], [1.97598, 0.0360291], [1.97798, 0.0330277], [1.97998, 0.030026], [1.98198, 0.0270241], [1.98398, 0.024022], [1.98599, 0.0210196], [1.98799, 0.0180172], [1.98999, 0.0150145], [1.99199, 0.0120118], [1.99399, 0.0090089], [1.996, 0.00600597], [1.998, 0.003003], [2, 0]];

polygon(P);

t-paul, 2016-02-15:
Closing as duplicate of #999. Workaound would be to scale the values up so it's not only fractions.

closed 2016-02-15 by t-paul

nophead, 2016-02-15:
This works

scale(0.001) polygon([for(p = P) p * 1000]);