openscad/openscad #1583: Generalized ranges

doug-moen, 2016-02-17:
The goal is to make ranges more useful. We'll simplify and generalize the language, by making the existing range notation just an alternate way of specifying a list.
1. Let's modify the behaviour of ranges, so that they behave like lists in all contexts.
2. Let's support the range syntax `a:b` and `a:b:c` as a 'generator' element of a list comprehension.

To modify the behaviour of ranges, let's make the following changes. Given `R=[0:3]`:
* `len` returns the number of elements in the range's sequence. Eg, `len(R)==4`
* `R[i]` returns the i'th element of the range's sequence. Eg, `R[3]==3`
* `concat` treats the range as if it were a list. Eg, `concat(R,[10,20])==[0,1,2,3,10,20]`
* in general, all built-in operations treat a range as if it were a list

List comprehension syntax is extended so that one of the comma-separated elements of a list comprehension can be a range specifier. For example, `[1,1:3,5,8]` == `[1,1,2,3,5,8]`.

GilesBathgate, 2016-02-17:
I have tried out some of these ideas in RapCAD, specifically that they behave like lists in all contexts. 

Another application for a range that I thought might be quite interesting was Interval mathematics, which I also experimented with. 

This could be used for example to represent a value with a tolerance i.e. ```5±1``` would be ```[4:6]``` Although that syntax is horrible, the ability to work out accumulative errors in these values could be quite useful in CAD.

GilesBathgate, 2016-02-17:
Interval arithmetic has the following rules:
when x = [a:b] and y = [c:d] then
* x+y = [a+c:b+d]
* x-y = [a-d:b-c]
* x*y = [min(a*c,a*d,b*c,b*d):max(a*c,a*d,b*c,b*d)]
* x/y = [min(a/c,a/d,b/c,b/d):max(a/c,a/d,b/c,b/d)]

doug-moen, 2016-02-17:
It's good that RapCAD has already implemented this extension. Do you have
any further comments, based on your experience of using this feature?

Interval values (for interval arithmetic) would need to have a different
syntax from ranges in OpenSCAD, for compatibility reasons. Please create a
separate feature request for interval values, if you want to push this
proposal any further.

GilesBathgate, 2016-02-17:
Well my comments would be that after initially making it just convert from a range directly to a list, and then using it in operations like len works fine, but if you give it an insanely big range, it will take ages:

```c#
len([0:1000000000000]);   // Zzz
```

so generators is the way to go.

likewise for indexing 

```c#
r=[0:5000:10000000000];
r[3];
```
Should not create a list of that many elements then take the third.

With respect to interval math, I think you can do it in a compatible way, since there are currently no operations defined for ranges.

```c#
[3:4]+[5:6] == undef;
```

GilesBathgate, 2016-02-17:
Here is a bit of a conundrum. What should ```len([-3,3]);``` give?

In ruby:
```ruby
(-3..3).size
```
gives 7

in python:
```python
len(range(-3,3))
```
gives 6

nophead, 2016-02-17:
That is because Python ranges are half open so range(-3,3) is [-3,-2,-1,0,1,2]

GilesBathgate, 2016-02-17:
Oh yeah! For some reason I thought ```for i in range(-3,3): print i``` would print 7 values.

GilesBathgate, 2016-02-21:
I am a bit confused about the proposed syntax consider:
```
1:2:3
```
which of the following is this to be treated as:
```
[1:2:3]

[[1:2]:3]

[1:[2:3]]
```
consider:
```
1:2:3:4
````
of the following how is this to be interpreted:
```
[[1:2:3]:4]

[[1:2]:[3:4]]

[1:[2:3:4]]

[1:[2:3]:4]
```

This is especially important in the context of the ternary operator:
```
a?1:2:3:4
```

Does it mean:
```
a?[1:2]:[3:4]

a?1:[2:3:4]

a?[1:2:3]:4
```

doug-moen, 2016-02-21:
@GilesBathgate Sorry about the confusion. `[a:b]:c` and `a:b:c:d` are not legal syntax in my proposal. I'll try to be ultra precise this time.

I'm proposing to generalize range syntax by merging it into the existing list comprehension syntax. There will no longer be any distinction between list literals and range literals. I'll make this precise by showing a BNF grammar.

Right now, in the dev snapshot, Torsten has generalized the list comprehension syntax, by integrating the [a,b,c] list literal syntax with list comprehension syntax. There is no longer any sharp distinction between a list literal and a list comprehension. A list literal now consists of a comma separated sequence of *generators*:
```
list_literal ::= "[" generators "]"
generators ::= *empty* | generator_list | generator_list ","
generator_list ::= generator | generator_list "," generator
```
Each generator specifies a sequence of zero or more values, and the list is constructed by concatenating the results of running each generator in sequence.

The simplest case of a generator is an expression, which specifies a single value:
```
generator ::= expression
```

A generator can be a `for` loop:
```
generator ::= "for" "(" identifier "=" expression ")" generator
```

A generator can be conditional. In the dev snapshot, we now support conditionals with an `else` clause.
```
generator ::= "if" "(" expression ")" generator
generator ::= "if" "(" expression ")" generator "else" generator
```

Generators may be qualified by `let`, which introduces local variables:
```
generator ::= "let" "(" variable_bindings ")" generator
```

In the dev snapshot, generators may now be parenthesized, which is useful for grouping in the case of nested conditionals.
```
generator ::= "(" generator ")"
```

In the dev snapshot, the `each` operator converts a list value to a generator:
```
generator ::= "each" expression
```
The argument to `each` evaluates to a list L, and each element of L is added to the list under construction. 

The above syntax is what we support now in the dev snapshot. My proposal is to generalize this a bit further, so that range specifiers are generators:
```
generator ::= expr ":" expr
generator ::= expr ":" expr ":" expr
```
Each expression argument to the range operator must evaluate to a number, that hasn't changed.

This means that range literals of the form `[a:b]` and `[a:b:c]` are now just a special case of a list literal.

For example,
```
[1, 2:5] => [1,2,3,4,5]
[for (x=[1:3]) 1:x] => [1,1,2,1,2,3]
```

@GilesBathgate Note that in `expr:expr` and `expr:expr:expr`, the individual arguments to the range operator are expressions, not generators. This means `(a:b):c` is a syntax error. Generator syntax is not legal outside of list literals.

t-paul, 2016-02-21:
The code change to implement that proposal is pretty trivial (like moving a couple of lines around in the `parser.y`). The main question is what to do about backward compatibility. Essentially it does remove the existence of the range type.

GilesBathgate, 2016-02-21:
@t-paul The syntax seems a bit confusing to me. You can optionally omit []'s but its not clear where.

```
[for (x=[1:3]) 1:x] //Correct
[for (x=1:3) 1:x] //Incorrect
[for (x=[1:3]) [1:x]] //Correct
[for (x=[1:3]) x>1?1:x:3 ] //Incorrect
[for (x=[1:3]) x>1?[1:x]:3 ] //Correct
```
What's real advantage of this?

t-paul, 2016-02-21:
I agree it's a bit confusing, the rule is pretty simple though. Anywhere between the `[` `]` (or additionally the `,` for multiple generators). The `for (...)` is just a prefix to the generator, so the first example is basically `[ 1 : x ]`.

GilesBathgate, 2016-02-21:
@doug-moen It would be useful if you update the OP to include which branch these changes refer to.

GilesBathgate, 2016-02-21:
@t-paul One thing you should watch out for regarding backwards compatibility is that I think this will break nesting:

```
a=[[1:3],[2:5]];

for(i=a[0])
    echo(i);
```
If I've understood correctly this is equivalent to:
```
a=[1,2,3,2,3,4,5];

for(i=a[0])
    echo(i);
```
Both scripts give a different result however.

t-paul, 2016-02-21:
No, those are not the same, either `[1:3, 2:5] == [1,2,3,2,3,4,5]` or `[[1:3],[2:5]] == [[1,2,3], [2,3,4,5]]`. But the point still stands for every place where ranges are explicitly used, e.g.:

current:
```
echo([0:2:6][2]); // ECHO: 6
echo([0:2:6].step); // ECHO: 2
```
new:
```
echo([0:2:6][2]); // ECHO: 4
echo([0:2:6].step); // ECHO: undef
```

doug-moen, 2016-02-21:
@GilesBathgate said "The syntax seems a bit confusing to me. You can optionally omit []'s but its not clear where."

Not quite. If you omit []'s then you remove a level of nesting in the list that you are generating. The syntax is consistent in this. There are no optional []'s.
```
[for (x=[1:3]) x] => [1,2,3]
[for (x=[1:3]) [x]] => [[1],[2],[3]]
[for (x=[1:3]) 1:x] => [1,1,2,1,2,3]
[for (x=[1:3]) [1:x]] => [[1],[1,2],[1,2,3]]
```

GilesBathgate, 2016-02-21:
@doug-moen, Right I realised this hence my follow up post. I am still wondering which branch you are talking about though when you refer to "dev snapshot"

t-paul, 2016-02-21:
All [those changes](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/WIP) @doug-moen mentioned should be in `master` but need to be enabled in Preferences->Features.

GilesBathgate, 2016-02-21:
@t-paul cool so we are talking about the same code, but @doug-moen 's interpretation of the parser he gave in BNF, is somewhat different from what is in parser.y

Is he proposing ```vector_expr``` no longer is going to have to have ```expr``` rule?

t-paul, 2016-02-21:
No, it would just move the two range expressions into `list_comprehension_elements` (see [patch](http://pastebin.com/0hF5Pwrn)).

Code-wise that simply reuses the `each` implementation that unwraps the following expression, so the implementation of `[ 1, 2 : 5 ]` is basically the same as `[ 1, each [ 2 : 5 ] ]`. Of cause that's not the optimized way to do things, but then that's implementation details :-) ...

Essentially that change would make the range type invisible to the user when looking at evaluation results as they are always unwrapped before they can be used anywhere.

GilesBathgate, 2016-02-21:
@t-paul Yeah, I get it now, thanks for sharing the actual code instead of "something else".

What threw me was that ```generator ::= expression``` rule which I interpreted as ```list_comprehension_elements ::= expr```

doug-moen, 2016-02-21:
@t-paul It's cool that you implemented the entire feature with such a small code change.

As for backwards compatibility, this doesn't change any behaviour that is documented in the manual. Ranges are only documented to be used as the target of a `for` loop. The *range*`.step` syntax you mentioned is new to me: it isn't documented, and I haven't seen it before in other people's code.

GilesBathgate, 2016-02-22:
@doug-moen I think the proposed grammar allows ```[true?1:2:3:4]``` I am not saying that's a bad thing. The right associativity of the ternary operator means that can be interpreted correctly as ```[(true?1:2):3:4]```

doug-moen, 2016-02-22:
@GilesBathgate said "I think the proposed grammar allows `[true?1:2:3:4]`"

I tested the development snapshot, and it works the way you say it does. I don't like it, but I don't have an alternative right now.

GilesBathgate, 2016-02-22:
@doug-moen I realised you've always been able to do that, which is why I deleted the comment.

GilesBathgate, 2016-02-22:
@doug-moen I don't see an issue with ```.step``` since as you say its undocumented, and possibly not used in the wild. The question is whether someone would want to be able to get at that data or indeed ```.start``` ```.end``` (which also exist). Its a shame that OpenSCAD currently has the index operator return this data too. I assumed it returned undef currently.

MichaelAtOz, 2016-02-23:
Don't forget .x .y .z are now in the wild.

doug-moen, 2016-02-23:
"Don't forget .x .y .z are now in the wild."

Not a problem. These don't work with ranges right now, so there's no conflict with my feature request.

cad-sc, 2016-02-23:
@doug-moen It would break

```
[1:3].x==undef;
```

I am not saying that's a bad thing, but it could still be something that someone was depending on.

GilesBathgate, 2016-02-23:
@doug-moen It would break

[1:3].x==undef;
I am not saying that's a bad thing, but it could still be something that someone was depending on.

doug-moen, 2016-02-23:
@GilesBathgate One of the problems with the OpenSCAD design is that virtually all language changes are breaking changes. Adding new operator syntax is generally safe, but extending the behaviour of an existing builtin operator, function or module, or adding a new function or module, is a breaking change. Bug fixes are usually breaking changes.

Pretty much all of the new features going into the 2016 release are, technically, breaking changes.

So the only thing we can really do is ask: is the code we are breaking something that a significant number of scripts are relying on? Also, we need to document the changes, so that users can make an informed decision when upgrading.

I've already given my reasons for why I think this change is relatively safe.

I agree that there may be code that is deliberately abusing undocumented behaviour in range values, but I doubt that there are very many scripts like that out there. Maybe a handful out of 100,000. That code will need to be updated, and I don't see an alternative right now unless we want to stop changing the language. The same criteria applies to the other changes planned for 2016.

GilesBathgate, 2016-02-23:
@doug-moen Yes I understand that. But you dismissed what @MichaelAtOz said rather than addressing it.

doug-moen, 2016-02-23:
Okay, I see what you are saying now. Sorry, I didn't mean to do that, I just didn't see the relevance of his comment at the time.