openscad/openscad #1584: type test functions

doug-moen, 2016-02-17:
Let's add the following built-in functions for testing the type of a value:
* `is_bool(X)` returns `true` if X is true or false, otherwise it returns `false`
* `is_int(X)` returns `true` if X is an integer, otherwise `false`. Sample implementation: `X == floor(X)`
* `is_num(X)` returns `true` if X is a number, otherwise `false`. Note that NaN is Not a Number, so `is_num(0/0)` returns `false`.
* `is_string(X)` returns `true` if X is a string, otherwise `false`
* `is_list(X)` returns `true` if X is a list, otherwise `false`. If we implement Generalized Ranges #1583, then `is_list` will return `true` if X is a range.

These functions will be more robust than the current idioms that some people are using, like `abs(X)==undef` to test if X is a number. See the forum discussion http://forum.openscad.org/Determining-what-data-type-a-variable-is-holding-td16111.html

In issue #1574, it is proposed that OpenSCAD report errors if bad arguments are passed to functions or modules. This change may break code that passes a bad argument to a function to test the type of a value, but if we provide type test functions, then there is a correct way to write this kind of code.

An `is_range` function is not proposed, due to Generalized Ranges #1583.

t-paul, 2016-02-17:
What's the benefit of having a list of separate test functions instead of a `typeof()` that lots of languages use?

Is it good to have values that have no function returning `true`? With the proposed list of tests, it's not possible to distinguish ranges and NaN.

As there is no `int` in OpenSCAD, I would not pretend that type exists.

doug-moen, 2016-02-17:
A type is a set of values over which a set of operations is defined. This is an abstract concept that is independent of how the language is implemented. My goal is to provide a useful high level abstraction, not to expose low level implementation details.

The distinction between integers and real numbers is a standard part of mathematics, and the distinction is part of OpenSCAD. The "integer" type mainly arises in connection with lists. The index `i` in `X[i]` is an integer. The `len` function returns an integer. The vertex indices in the faces argument of `polyhedron` are integers. The convexity argument of polyhedron is an integer, and it is documented that way in the manual. So you can't say there are no integers in OpenSCAD.

These functions can be used to define other, more fine grained type predicates like this:
```
function is_vec3(X) = is_list(X) && len(X)==3 && is_num(X[0]) && is_num(X[1]) && is_num(X[2]);
```
That's the type of a point or vector in 3-space.

Types overlap. Integer is a subtype of Number, and Vec3 is a subtype of List. In this context, a typeof function makes no sense. A "typeof" function only makes sense if you are probing low level implementation details, which might be useful for language implementers who are debugging a problem, but that is not the point of this feature request.

One use case for these functions is performing domain and range checks within function and module definitions. Or testing that the top level parameters in a model script are valid. They would be quite useful when used with an 'assert' operation, but that's a separate feature request. Another use case is writing polymorphic code that does different things base on the type of an argument. For example, two different code paths depending on whether an argument matches is_num or is_vec3.

The reason that is_num returns false for NaN is because NaN is out of bounds for all arithmetic operations, so it really is Not a Number. It's true that in the OpenSCAD implementation, numbers and NaN are represented using the same type code (Value::NUMBER), but that's a low level implementation detail that isn't relevant to this feature. By contrast, a typeof function that is meant to expose the underlying implementation would return "number" in response to the NaN value.

GilesBathgate, 2016-02-20:
@doug-moen You have ```is_num```, which returns true also true if its an int. In other words anything that passes the ```is_int``` check will pass the ```is_num``` check?

In the same way, why can't I check that something explicitly ```is_range```, despite it also returning true with ```is_list``` (Given https://github.com/openscad/openscad/issues/1583)

doug-moen, 2016-02-20:
@GilesBathgate Yes, all integers are numbers. Int is a subtype of Num. The reason Int is a subtype of Num is because there are primitive operations that are only well defined if they are passed integers as arguments; they aren't well defined over the more general class of numbers. I gave a bunch of examples earlier, such as the convexity parameter of polyhedron.

The explicit goal of #1583 is to make lists and ranges interchangeable. There aren't any operations on ranges that don't also accept lists. There aren't any operations on lists that don't also accept ranges. So one is not a subtype of the other: they are operationally equivalent.

Given this, one perfectly valid way to implement ranges is for the range constructor to return a list. There are performance issues when huge lists are used, but this likely doesn't happen much in practice, and the important thing is that this implementation provides the correct behaviour. All scripts still work, still compute the same geometry.

There are multiple implementations of OpenSCAD out there. If #1583 is implemented, we can imagine that some of those implementations might use different internal representations for lists and ranges, while other implementations might use the same internal representation. Yet these implementations would be compatible with one another.

This trick of using multiple different internal representations for lists, depending on how the list is constructed and used, is actually quite common in implementations of dynamically typed languages. I've been looking closely at high performance implementations of Javascript and Lua to get ideas on how to implement OpenSCAD more efficiently, and they all do this. A common trick is to have different internal representations for list-of-float and list-of-integer. As a user, you have no idea that the implementation is doing this, and you shouldn't need to care, but it makes programs run faster and use less memory. "Range" is just another internal representation for lists, given #1583.

Giles, you and Torsten are both language implementors. You both know exactly what is going on inside the language implementations that you maintain. And from your perspective, it is perfectly reasonable to have a way to query the internal representation of a value. Torsten called this operation `typeof`. It returns the internal type code associated with a value. I can see how this operation would be useful to language implementors, for debugging problems in the underlying implementation. For example, there is a fancy implementation of lists that uses multiple internal representations, and you'd like to verify that the representation you expect is being constructed in a particular unit test script. However, the problem is that this operation returns different results in different implementations of OpenSCAD. Any code that uses it is not portable. 

With #1583, `[1,2,3]` and `[1:3]` are both lists. They have identical behaviour in all contexts. There is no way to distinguish them from within the OpenSCAD language. At the implementation level, we may use different representations, but this shouldn't matter to users. Your suggestion of adding an `is_range` function will break the #1583 semantics, because you want `is_range([1,2,3])==false` and `is_range([1:3])` to be true. This conflicts with the goals of this feature request. A method for probing the internal representation of a value and exposing implementation-specific details has different use cases and requirements than #1584, and should be put into a different feature request.

GilesBathgate, 2016-02-20:
@doug-moen I am a bit confused then. Under https://github.com/openscad/openscad/issues/1583 I suggested that we could enhance ranges by implementing addition subtraction etc. You said this would break compatibility since currently: ```[1:3]+[4:5]==undef``` What you are suggesting is that they are semantically identical to lists, in which case ```[1:3]+[4:5]==undef``` would be semantically identical to ```[1,2,3]+[4,5,6]``` which is equal to ``` [5, 7, 9]```. Which would break compatibility.

I don't see why ranges and vectors should have the same meaning in all contexts for example ```cube([1:3]);```  has limited use, I would expect that to be a warning or error under https://github.com/openscad/openscad/issues/1574#issuecomment-177988896 

In other contexts it does make sense and I agree that ranges can be treated as if they were vectors.

Basically if your argument is that lists and ranges are the same thing then I concur that ```is_range``` wouldn't make much sense but I disagree that they are the same thing. You also proposed ```is_vec3``` which seems to suggest that lists and vectors are different things, where there is currently no distinction.

doug-moen, 2016-02-21:
@GilesBathgate Yes, #1583 "Generalized Ranges" proposes to make ranges semantically identical to lists.

The range notation is used to specify an ordered sequence of numbers, which you can iterate over. `[1:5]` is an ordered sequence of 5 numbers. It's just a short form for `[1,2,3,4,5]`.

This becomes more obvious when you consider my proposal to further extend the list literal syntax. 
* With Torsten's latest changes to the dev branch, the previously sharp distinction between list literals and list comprehensions has been erased. Now, a list literal is a comma separated sequence of expressions. Each expression specifies either a single element, or a list of elements using `for` notation. For example, `[1,for(i=[2:4])i]` is `[1,2,3,4]`.
* In #1583, I propose to take this idea further by erasing the distinction between range literals and list literals. Now, a list literal is a comma separated sequence of expressions, where an expression is allowed to be a range specifier like `a:b` or `a:b:c`. For example, `[0,1:3]` is `[0,1,2,3]`.

Since #1583 eliminates the distinction between range and list literals, it makes no sense to claim that ranges are different from lists. We would just be building a trap for users to fall into.

You would like to support interval arithmetic, which requires that we introduce some notion of an interval value. That's fine, please create a new feature request.

An interval is not a range. A range is a ordered finite sequence of numbers, with each number in the sequence separated by a step value. An interval is quite different: interval(x,y) is ALL of the numbers from x to y. There is no step. Intervals are continuous. It makes no sense to iterate over an interval. The proposed `len(r)` operation on ranges makes no sense for intervals, ditto the `r[i]` indexing operation. The operations on intervals are quite different from the operations on ranges. We would need a new syntax for interval values which is different from ranges.

GilesBathgate, 2016-02-21:
I did not say I would like to support interval arithmetic, I said I have experimented with it in RapCAD and was merely sharing my Ideas. 

You've focused on what intervals, are, and are not, but have completely missed the point I was making. Quite simply https://github.com/openscad/openscad/issues/1583 breaks the following:

```[1:2]+[3:4]==undef```

You haven't addressed the other points I made. @doug-moen This is an interesting discussion, so I would be interested to hear your further thoughts.

doug-moen, 2016-02-23:
Sorry about the delayed response.

1. My motivation for #1583 is to make the language simpler and more powerful. Simplicity comes from eliminating the range type, and making range notation just another way to form a list. Power comes from the fact that range values now work in all the same contexts that lists work. It's weird that a for loop can iterate over a range, but len(range) returns undef. That's all fixed. Eliminating weird special cases means greater simplicity and expressive power. List notation becomes more powerful by making range notation just another kind of list comprehension, which can be combined with the other list comprehension operators.

2. You are correct that it would be possible for Range to be a subtype of List, where all of the list operations work on Ranges as expected, but there is an is_range function that distinguishes ranges from lists.

3. This makes things more complicated.
  * The language definition becomes more complicated. We now have to document the situations where a list literal produces a range value, vs the situations where a list literal produces a (non-range) list value. For example, [2:4] is a range, but [1,2:4] is a list, but [c?(a:b):(de)] is a range, and so on.
  * You also suggested that maybe range values should not work as lists in all contexts. For example, cube([1:3]) should be an error. This also means that Range is not a true subtype of List, so maybe that's a separate proposal. It would be very complicated to document, on a case by case basis, the situations where ranges behave like lists, vs the situations where range values do not behave like lists. I don't know what the criteria would be.
  * The implementation of #1583 becomes much more complicated. The very simple implementation that Torsten wrote will no longer work, and now a large number of source files will need to be changed to (a) treat range values as lists in all contexts, even though they have a different internal type or (b) support all those case-by-case decisions where ranges do and do not behave like lists.

4. What justifies this increased complexity? That's what I don't know. There needs to be a killer application to justify adding so much complexity to #1583, and I don't see it.

GilesBathgate, 2016-02-24:
> Simplicity comes from eliminating the range type, and making range notation just another way to form a list.
That may be true. Its not universally true. This is the strongly typed vs weakly typed argument. You could argue that everything should be a string, or sequence of bytes (which would be silly). I am not going to argue that point because I don't think there is a right answer.

> Power comes from the fact that range values now work in all the same contexts that lists work.
I already have that feature in rapcad, its not a prerequisite that they BE the same thing to be treated the same way.

> You are correct that it would be possible for Range to be a subtype of List.
Yeah so its clear that the argument was never about what's possible, rather what you think users may or may not be able to understand.

>The language definition becomes more complicated. We now have to document the situations where a list literal produces a range value.
Yes I agree. But that's a side effect of https://github.com/openscad/openscad/issues/1583 Documenting what is_range does is simple. "It returns true if the argument is a range.". 

> It would be very complicated to document, on a case by case basis, the situations where ranges behave like lists.
You just document the type of the parameters that the modules expect. A cube expects a vec3, multmatrix expects a mat4x4. (perhaps we should add ```is_mat4x4```)

>The very simple implementation that Torsten wrote will no longer work, and now a large number of source files will need to be changed.
Torstens change makes things work. It leaves a lot of dead code referring to the ```RangeType``` class. Virtually every built in module switches on the values type and specifically checks for ```Value::RANGE```. This would ideally all be removed otherwise the codebase becomes a mess to maintain.

> What justifies this increased complexity?
You are right, for OpenSCAD this is a much more complicated language feature. For rapcad its relatively simple, because its already been designed with this in mind.

Anyway I've changed my mind, if you don't want to add these features to openscad, then who am I to argue. It was more a case of understanding, why you thought it was not possible. I think you acknowledged that it is possible but 1. Too complicated to implement in the OpenSCAD codebase (which is a shame), and 2. Confusing to users, (which is a bit condescending). But fair enough, i've already implemented the feature in rapcad and moved on.

For OpenSCAD what might be more useful is an ```is_seq()``` function that would return true if the list is a sequence of numbers:

```scad
is_seq([1,2,3])==true;
is_seq([1:3])==true;
is_seq([1,3,2])==false;
is_seq([1,2,2,2,2])==false;
```

MichaelPFrey, 2019-02-05:
* is_num(X)
Is inf a number...? well, for practicable purposes it has to be false, which philosophically can be explained by "inf is a concept not a number".

GilesBathgate, 2019-02-05:
I treat 1/0, -1/0 and 0/0 as undef and so `is_num(1/0)`, `is_num(-1/0)` and `is_num(0/0)` all return false.

Openscad treats 1/0 as inf, -1/0 as -inf and 0/0 as nan, so there is an opportunity to do something different. But as a use case if you really want to check the boundary condition, you would probably use:

function is_a_sensible_number(x) = is_num(x) && x > -100000 && x < 100000;

GilesBathgate, 2019-02-05:
Furthermore `echo(1e309);` yeilds 
`ECHO: `
whereas
`echo(1e308);`
yeilds
`ECHO: 1e+308`
perhaps `is_num(1e+309)` should be false, since it dosen't seem to be a useable number in Openscad.

doug-moen, 2019-02-05:
In my copy of OpenSCAD, echo(1e309) yields inf. Has this changed in the latest dev snapshot?

Yes, inf is a number. It is legal as input to most numeric operations.
   1 / inf == 0
   pow(1, inf) == 1
   pow(1, "abc") == undef

In pretty much all dynamically typed programming languages, there is an equivalent to is_num, and inf is always classified as a number, in all the languages I know.

GilesBathgate, 2019-02-05:
It was from OpenSCAD 2015.03-2.

nophead, 2019-02-05:
I think inf needs to be a number as it is valid input for atan, etc.

MichaelPFrey, 2019-02-06:
Good arguments, I will adjust the implementation so that inf returns true for is_number.
For is_integer, inf has to return false.

I will think about adding an is_finite check, as this is the test I typically use.

MichaelAtOz, 2019-06-27:
Are these hiding in the wiki somewhere, or did that just slip though the non-manifold cracks?

t-paul, 2019-06-30:
Closed via #2762.

No, they were hiding in the "Other" sections. I've moved them into a separate section listed on the main page. Also the cheatsheet is updated where they did not make it over from the snapshot file to the release one.

closed 2019-06-30 by t-paul