openscad/openscad #1588: Crash on startup (OSX 10.10.5)

dcousens, 2016-02-23:
Crash dump:

``` bash
Process:               OpenSCAD [29475]
Path:                  /Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD
Identifier:            org.openscad.OpenSCAD
Version:               2015.03-3 (2016.02.17)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           OpenSCAD [29475]
User ID:               501

Date/Time:             2016-02-23 14:33:47.520 +1100
OS Version:            Mac OS X 10.10.5 (14F1605)
Report Version:        11
Anonymous UUID:        A3791889-9858-9926-FCFF-FBCBA566AF9D

Sleep/Wake UUID:       94C052CA-439F-4672-9FDA-AFC4823545BF

Time Awake Since Boot: 130000 seconds
Time Since Wake:       1600 seconds

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Application Specific Information:
*** Terminating app due to uncaught exception 'SUNoFeedURL', reason: 'You must specify the URL of the appcast as the SUFeedURL key in either the Info.plist or the user defaults!'
terminating with uncaught exception of type NSException
abort() called

Application Specific Backtrace 1:
0   CoreFoundation                      0x00007fff937c403c __exceptionPreprocess + 172
1   libobjc.A.dylib                     0x00007fff8c40b76e objc_exception_throw + 43
2   CoreFoundation                      0x00007fff937c3eed +[NSException raise:format:] + 205
3   Sparkle                             0x00000001019555a4 -[SUUpdater feedURL] + 340
4   Sparkle                             0x00000001019536e8 -[SUUpdater initForBundle:] + 757
5   Sparkle                             0x00000001019533d1 +[SUUpdater updaterForBundle:] + 205
6   Sparkle                             0x00000001019532de +[SUUpdater sharedUpdater] + 64
7   OpenSCAD                            0x0000000100be75b9 _ZN18SparkleAutoUpdaterC2Ev + 73
8   OpenSCAD                            0x0000000100d6fdea _Z3guiRSt6vectorISsSaISsEERKN5boost10filesystem4pathEiPPc + 1338
9   OpenSCAD                            0x0000000100d7377f main + 10127
10  OpenSCAD                            0x0000000100be7564 start + 52

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff8e191286 __pthread_kill + 10
1   libsystem_c.dylib             	0x00007fff90a189ab abort + 129
2   libc++abi.dylib               	0x00007fff8e970a21 abort_message + 257
3   libc++abi.dylib               	0x00007fff8e9989d1 default_terminate_handler() + 267
4   libobjc.A.dylib               	0x00007fff8c40f7d6 _objc_terminate() + 103
5   libc++abi.dylib               	0x00007fff8e9960a1 std::__terminate(void (*)()) + 8
6   libc++abi.dylib               	0x00007fff8e995b30 __cxa_throw + 121
7   libobjc.A.dylib               	0x00007fff8c40b898 objc_exception_throw + 341
8   com.apple.CoreFoundation      	0x00007fff937c3eed +[NSException raise:format:] + 205
9   org.sparkle-project.Sparkle   	0x00000001019555a4 -[SUUpdater feedURL] + 340
10  org.sparkle-project.Sparkle   	0x00000001019536e8 -[SUUpdater initForBundle:] + 757
11  org.sparkle-project.Sparkle   	0x00000001019533d1 +[SUUpdater updaterForBundle:] + 205
12  org.sparkle-project.Sparkle   	0x00000001019532de +[SUUpdater sharedUpdater] + 64
13  org.openscad.OpenSCAD         	0x0000000100be75b9 SparkleAutoUpdater::SparkleAutoUpdater() + 73
14  org.openscad.OpenSCAD         	0x0000000100d6fdea gui(std::vector<std::string, std::allocator<std::string> >&, boost::filesystem::path const&, int, char**) + 1338
15  org.openscad.OpenSCAD         	0x0000000100d7377f main + 10127
16  org.openscad.OpenSCAD         	0x0000000100be7564 start + 52

Thread 1:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff8e192232 kevent64 + 10
1   libdispatch.dylib             	0x00007fff935c626e _dispatch_mgr_thread + 52

Thread 2:
0   libsystem_kernel.dylib        	0x00007fff8e19194a __workq_kernreturn + 10
[... 288 lines cut ...]
Stack                              10.1M
VM_ALLOCATE                        17.2M
__DATA                             18.6M
__IMAGE                             528K
__LINKEDIT                         79.3M
__TEXT                            166.9M
__UNICODE                           552K
mapped file                        67.4M
shared memory                         4K
===========                      =======
TOTAL                             464.7M

Model: MacBookAir7,2, BootROM MBA71.0166.B08, 2 processors, Intel Core i5, 1.6 GHz, 4 GB, SMC 2.27f2
Graphics: Intel HD Graphics 6000, Intel HD Graphics 6000, Built-In
Memory Module: BANK 0/DIMM0, 2 GB, DDR3, 1600 MHz, 0x80AD, 0x483943434E4E4E384A544D4C41522D4E544D
Memory Module: BANK 1/DIMM0, 2 GB, DDR3, 1600 MHz, 0x80AD, 0x483943434E4E4E384A544D4C41522D4E544D
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x117), Broadcom BCM43xx 1.0 (7.15.166.24.3)
Bluetooth: Version 4.3.6f3 16238, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0128G, 121.33 GB
USB Device: Internal Memory Card Reader
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Air, Apple Inc., 27.2
```

Anderson69s, 2016-02-23:
Same problem here. But I found a solution :

kintel, 2016-02-23:
Should be fixed now - try to re-download the -03 binaries.

closed 2016-02-23 by kintel