openscad/openscad #1591: Objects created by different fail to render or render incorrectly under preview.

christakahashi, 2016-02-26:
Simple program:

```
mm = 25.4;
conn_height = 10;
$fn = 30;

difference() {
  cylinder(d=0.5*mm,h=conn_height);
  cylinder(d=0.213*mm+0.4,h=conn_height+1);
}
```
generates the follwing preview/rendered output:
[image]

I've been having similar rendering problems for months now.  I think it may be related to my upgrade to windows 10, but I wasn't paying attention close enough to know for sure.  I'm using version 2015.03-2.

kintel, 2016-02-26:
I would try two things:
1. Enable Edit->Preferences->Advanced->Force Goldfeather
2. Look at Help->Library Info, search for "OpenGL Version", google your version number and see if there is a newer version of the OpenGL drivers available

christakahashi, 2016-02-26:
Goldfeather same to fix it (thanks!).  I'm using intel graphics with the latest drivers.  my other machine has amd graphics.  I'll double check for the issue and new drivers there.

Neon22, 2016-02-26:
Its exactly the problem. The Intel graphics drivers have a problem. Goldfeather fixes it.
Only Intel drivers.

christakahashi, 2016-02-27:
Interesting.  can future versions of OpenSCAD detect which driver your using and turn on goldfeather?  Is there a disadvantage to using goldfeather (ie why not have it on by default)?

Neon22, 2016-02-27:
IIRC a release has not been made since the Intel HD graphics driver problem was found. Hopefully Intel will fix it sooon but no luck yet.
I am not in a position to indicate policy about this for OpenSCAD...

doug-moen, 2016-02-27:
My understanding is that Goldfeather is more robust but a bit slower for
previewing.

msrd0, 2016-10-02:
This bug is not only on Intel driver. I'm running OpenSCAD 2015.03.03 on ArchLinux with NVIDIA drivers 370.28 and encounter the exact same issue. The NVIDIA driver is currently the latest stable driver available in the repositories.

Turning on Force Goldfeather actually doesn't remove this bug. It the looks like reported in https://github.com/openscad/openscad/issues/1409#issuecomment-240608165

kintel, 2016-10-02:
Could anyone test if the new OpenCSG fixes this?
We could close #1409, which has very similar symptoms.

donbright, 2016-10-03:
cannot reproduce bug with opencsg 1.4.2 ubuntu linux 16

msrd0, 2016-10-03:
@donbright I can reproduce the bug with opencsg version 1.4.0 and 1.4.2

When I try to execute this code:

```python
color("Green")
difference() {
	cube([3, 3, 1]);
	translate([1.5,1.5,0])
		cylinder(r=1, h=1, $fn=100);
}
```

I get this image when using preview (Goldfeather doesn't change anything):

[image]

Using render I see what I expected:

[image]

Note that viewing the preview object from different angles makes the area sometimes greener and sometimes emptier

t-paul, 2016-10-03:
@msrd0 I don't think that has anything to do with the bug report. This very much looks like Z-fighting caused by coincident faces and happens because the graphics card can't decide which of the 2 things to render as they have the same depth value.

See FAQ: [What are those strange flickering artifacts in the preview?](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#What_are_those_strange_flickering_artifacts_in_the_preview.3F)

msrd0, 2016-10-03:
@t-paul Making the cylinder inside bigger resolves the strange effects. But those effects exist in the example in the first post as well; translating the second cylinder by `[0,0,-0.5]` fixes this problem as well.

t-paul, 2016-10-03:
Right, the translation will fix the Z-fighting.

The issue screenshots show it's mainly about showing the separate parts depending on the viewing angle instead of the difference it's supposed to show. I can't seem to reproduce that though (Debian/testing, nouveau/mesa-12.0.3, SCS algorithm, both release 2015.03-1 and nightly build 2016.10.01.nightly (git 7e0935d)).

kintel, 2016-10-04:
@christakahashi @Neon22 Are any of you, who could reproduce the original issue, able to test against an OpenCSG-1.4.2 build?

donbright, 2016-10-05:
windows snapshots updated to opencsg 1.4.2

donbright, 2016-10-05:
I can **_Confirm_** this bug under Windows 10 with intel graphics , following library info

```
OpenSCAD Version: 2015.03-2
System information: Windows 8 (v6.2.0.0) 4 CPUs 7.92 GB RAM
Compiler: GCC "5.2.0" 64bit
Boost version: 1_57
Eigen version: 3.2.5
CGAL version, kernels: 4.5, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.0
Qt version: 5.5.1
QScintilla version: 2.8.4
MingW build: MingW64
GLib version: 2.44.1
Application Path: C:/Users/don/Downloads/OpenSCAD-2015.03-2-x86-64/openscad-2015.03-2
Documents Path: C:\Users\don\Documents
Resource Path: C:/Users/don/Downloads/OpenSCAD-2015.03-2-x86-64/openscad-2015.03-2
User Library Path: C:/Users/don/Documents/OpenSCAD/libraries
User Config Path: C:\Users\don\AppData\Local/OpenSCAD
Backup Path: C:/Users/don/Documents/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
C:/Users/don/Documents/OpenSCAD/libraries
C:/Users/don/Downloads/OpenSCAD-2015.03-2-x86-64/openscad-2015.03-2\libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
GLEW version: 1.12.0
OpenGL Version: 4.4.0 - Build 10.18.15.4274
GL Renderer: Intel(R) HD Graphics
GL Vendor: Intel
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Using QGLWidget
```

I can also confirm latest snapshot shows this bug is **_fixed_**. 

```
OpenSCAD Version: 2016.10.04 (git 7e0935d)
System information: Windows 8 (v6.2.0.0) 4 CPUs 7.92 GB RAM
Compiler: GCC "4.9.4" 64bit
Boost version: 1_60
Eigen version: 3.2.5
CGAL version, kernels: 4.6.3, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.7.0
QScintilla version: 2.8.4
MingW build: MingW64
GLib version: 2.44.1
Application Path: C:/Users/don/Downloads/OpenSCAD-2016.10.04-x86-64/openscad-2016.10.04
Documents Path: C:\Users\don\Documents
Resource Path: C:/Users/don/Downloads/OpenSCAD-2016.10.04-x86-64/openscad-2016.10.04
User Library Path: C:/Users/don/Documents/OpenSCAD/libraries
User Config Path: C:\Users\don\AppData\Local/OpenSCAD
Backup Path: C:/Users/don/Documents/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
C:/Users/don/Documents/OpenSCAD/libraries
C:/Users/don/Downloads/OpenSCAD-2016.10.04-x86-64/openscad-2016.10.04\libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
GLEW version: 1.12.0
OpenGL Version: 4.4.0 - Build 10.18.15.4274
GL Renderer: Intel(R) HD Graphics
GL Vendor: Intel
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
```

christakahashi, 2016-10-05:
@kintel I was able to re-reproduce (by onece again turning off goldfeather) the bug.  Then I switch to the latest development snapshot on the website (2016.10.4, 64bit windows version) and cannot reproduce the bug.  So for my case it seems fixed.

lws-team, 2017-05-31:
This is still broken for me using today's master OpenSCAD, on an Intel graphics box with Fedora 26 Alpha.  The distro version is 2015-03-3.

Goldfeather makes no difference.

Set the view to "orthogonal" makes the problem go away, even on the 2015 version.

kintel, 2017-05-31:
@lws-team Does Fedora 26 Alpha ship OpenCSG-1.4.2 ?

lws-team, 2017-05-31:
@kintel Yes, exactly that version.

kintel, 2017-05-31:
@lws-team That sounds like a slightly different problem. Could you paste Help->Library info here?
If you have any other information about the nature of this issue and possibly how similar/different it is from the originally posted issue, that would also be helpful.

lws-team, 2017-05-31:
```
OpenSCAD Version: 2015.03-3
System information: Linux 4.11.0-2.fc26.x86_64 #1 SMP Tue May 9 15:24:49 UTC 2017 x86_64 Fedora 26 (Workstation Edition) 8 CPUs 31.31 GB RAM
Compiler: GCC "7.0.1 20170219 (Red Hat 7.0.1-0.9)" 64bit
Boost version: 1_63
Eigen version: 3.3.2
CGAL version, kernels: 4.9, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 4.8.7
QScintilla version: 2.10
MingW build: No
GLib version: 2.51.2
Application Path: /usr/bin
Documents Path: /home/agreen/.local/share
Resource Path: /usr/share/openscad
User Library Path: /home/agreen/.local/share/OpenSCAD/libraries
User Config Path: /home/agreen/.config/OpenSCAD
Backup Path: /home/agreen/.local/share/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
/home/agreen/.local/share/OpenSCAD/libraries
/usr/share/openscad/libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
/usr/share/fonts
/usr/share/X11/fonts/Type1
/usr/share/X11/fonts/TTF
/usr/local/share/fonts
/home/agreen/.local/share/fonts
/home/agreen/.fonts
/usr/share/fonts/aajohan-comfortaa
/usr/share/fonts/abattis-cantarell
/usr/share/fonts/adobe-source-han-sans-cn
/usr/share/fonts/adobe-source-han-sans-tw
/usr/share/fonts/blender
/usr/share/fonts/default
/usr/share/fonts/dejavu
/usr/share/fonts/gdouros-symbola
/usr/share/fonts/gnu-free
/usr/share/fonts/google-android-emoji
/usr/share/fonts/google-crosextra-caladea
/usr/share/fonts/google-crosextra-carlito
/usr/share/fonts/google-droid
/usr/share/fonts/google-noto
/usr/share/fonts/google-noto-emoji
/usr/share/fonts/jomolhari
/usr/share/fonts/julietaula-montserrat
/usr/share/fonts/khmeros
/usr/share/fonts/liberation
/usr/share/fonts/lklug
/usr/share/fonts/lohit-assamese
/usr/share/fonts/lohit-bengali
/usr/share/fonts/lohit-devanagari
/usr/share/fonts/lohit-gujarati
/usr/share/fonts/lohit-gurmukhi
/usr/share/fonts/lohit-kannada
/usr/share/fonts/lohit-odia
/usr/share/fonts/lohit-tamil
/usr/share/fonts/lohit-telugu
/usr/share/fonts/mathjax
/usr/share/fonts/naver-nanum
/usr/share/fonts/opensymbol
/usr/share/fonts/paktype-naskh-basic
/usr/share/fonts/paratype-pt-sans
/usr/share/fonts/sil-abyssinica
/usr/share/fonts/sil-mingzat
/usr/share/fonts/sil-nuosu
/usr/share/fonts/sil-padauk
/usr/share/fonts/smc
[... 210 lines cut ...]
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_light_max_exponent
GL_NV_packed_depth_stencil
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_OES_EGL_image
GL_OES_read_format
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
```

I used it intensively today without problems using 'orthogonal'.

Without it, objects that are clipped by using `difference()` do not reliably preview-render as clipped.  For example I have a cylinder and a cube positioned over one half of it.  Depending on the preview angle, the clipping either occurs or is missed off, showing the whole cylinder.

I did notice one other problem... when objects have alpha, the preview render hides objects that are logically occluded.  But of course you can see they are missing actually since the occluding object is transparent.

Other than that it's very cool.

gadgetmind, 2017-07-17:
I'm seeing exactly the same as lws-team and orthogonal fixes it for me, which is a pretty good work around so thanks!

Note that I have opencsg.so version 1.4.2 installed, and ldd shows that OpenSCAD is opening this version, but I see this.

OpenSCAD Version: 2017.07.16.nightly (git a6e9883)
System information: Linux 4.4.0-83-generic #106-Ubuntu SMP Mon Jun 26 17:54:43 UTC 2017 x86_64 Ubuntu 16.04.2 LTS 4 CPUs 7.80 GB RAM
Compiler: GCC "5.3.1 20160413" 64bit
Boost version: 1_58
Eigen version: 3.2.92
CGAL version, kernels: 4.7, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.0
Qt version: 4.8.7
QScintilla version: 2.9.1
MingW build: No
GLib version: 2.48.0
libzip version: 
Application Path: /usr/bin
Documents Path: /home/xxxxxxx/.local/share
Resource Path: /usr/share/openscad-nightly
User Library Path: /home/xxxxxxx/.local/share/OpenSCAD/libraries
User Config Path: /home/xxxxxxx/.config/OpenSCAD
Backup Path: /home/xxxxxxx/.local/share/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
/home/xxxxxxx/.local/share/OpenSCAD/libraries
/home/xxxxxxx

donbright, 2017-08-06:
the OpenSCAD library info window is printing the version of OpenCSG that OpenSCAD was compiled against. not what it is linked against.

It is possible that your machine has an openscad binary that was compiled against OpenCSG1.4.0 but, when running is being dynamically linked at runtime by the operating system dynamic linker against OpenCSG1.4.2 as 1.4.0 might have been removed from the system. 

This might cause all kinds of bizarre behavior. A typical solution would be to update the OpenSCAD package using your operating system package manager.... and hopefully the package maintainers are not shipping a version of the binary that was compiled against a version of a library that differs from what is available to the system.

lws-team, 2017-08-07:
In my case anyway, Help | Library Info says

> OpenCSG version: OpenCSG 1.4.2

and

```
$ rpm -q opencsg
opencsg-1.4.2-3.fc26.x86_64
```

It's Fedora so it's pretty unlikely they will break deps like that.  Anyway I appreciate the thought.  It may be something to do with Intel GPU I am using.

donbright, 2017-08-07:
would you want to  dump your gpu info? here's mine following instructions here  

http://www.binarytides.com/linux-get-gpu-information/

```
don@serebryanya:~/src/openscad/bin$ lspci -vnn | grep VGA -A 12
00:02.0 VGA compatible controller [0300]: Intel Corporation Broadwell-U Integrated Graphics [8086:1606] (rev 09) (prog-if 00 [VGA controller])
	Subsystem: Hewlett-Packard Company Broadwell-U Integrated Graphics [103c:80c1]
	Flags: bus master, fast devsel, latency 0, IRQ 45
	Memory at c2000000 (64-bit, non-prefetchable) [size=16M]
	Memory at b0000000 (64-bit, prefetchable) [size=256M]
	I/O ports at 6000 [size=64]
	Expansion ROM at <unassigned> [disabled]
	Capabilities: <access denied>
	Kernel driver in use: i915
	Kernel modules: i915

00:03.0 Audio device [0403]: Intel Corporation Broadwell-U Audio Controller [8086:160c] (rev 09)
	Subsystem: Hewlett-Packard Company Broadwell-U Audio Controller [103c:80c1]

```

lws-team, 2017-08-07:
```
00:02.0 VGA compatible controller [0300]: Intel Corporation Iris Pro Graphics 580 [8086:193b] (rev 09) (prog-if 00 [VGA controller])
	Subsystem: Intel Corporation Device [8086:2064]
	Flags: bus master, fast devsel, latency 0, IRQ 139
	Memory at db000000 (64-bit, non-prefetchable) [size=16M]
	Memory at 40000000 (64-bit, prefetchable) [size=1G]
	I/O ports at f000 [size=64]
	[virtual] Expansion ROM at 000c0000 [disabled] [size=128K]
	Capabilities: [40] Vendor Specific Information: Len=0c <?>
	Capabilities: [70] Express Root Complex Integrated Endpoint, MSI 00
	Capabilities: [ac] MSI: Enable+ Count=1/1 Maskable- 64bit-
	Capabilities: [d0] Power Management version 2
	Capabilities: [100] Process Address Space ID (PASID)
	Capabilities: [200] Address Translation Service (ATS)
	Capabilities: [300] Page Request Interface (PRI)
	Kernel driver in use: i915
	Kernel modules: i915

```

lws-team, 2017-08-07:
FWIW this is all you need to reproduce it here (and perspective mode).

Play with the render angle in preview, it will flip between being correct and the whole 360 degrees of the cylinder being shown.

```
	difference() {
		difference() {
			translate ([0, 0, 10])
			color( "gold", 0.5)
			cylinder(d = 26.4, h = 16 + 28 + 2);
				
			translate ([0, 0, 9.9])
			color( "gold", 1.0)
			cylinder(d = 23.4, h = 16 + 28 + 2.2);
		}

		translate ([141, 0 , 0]) cube(282, center = true);
	}
```

donbright, 2017-08-07:
confirmed. 

https://www.youtube.com/edit?video_id=Gn6IWCcyAgg

lws-team, 2017-08-07:
Thanks for reproducing it... I meddled with the size of the cube axes (as [x, y, z] ) and that affects it.  If you copy and paste the cube outside the difference() so you can see it, it seems the 'bad view' is related to a cube vertex intersecting it somehow.

donbright, 2017-08-07:
confirmed in latest git master as well. ubuntu 16

nophead, 2017-08-07:
I think that has always been a feature of OpenCSG. The large cube that cuts
the cylinder must fit into the clipping volume for all of its faces it to
be drawn. As soon as you zoom in too far bits stops being drawn and less of
the cylinder is removed.

The surprising thing is that it works in orthogonal mode though. So maybe
it is a bug that can be fixed rather than a limitation. Presumably the
faces of the cube should be clipped to the view instead of being skipped
entirely when they don't fit.

gadgetmind, 2017-08-07:
It may be a feature, but I can sit with my Windows machine (Intel), my Linux desktop (AMD), and Linux notebook (Intel), side-by-side and it all works fine on Windows in perspective mode but I have to set orthogonal mode on Linux to avoid the issues.

gadgetmind, 2017-08-07:
My Windows machine is running Windows 2017.04.05 (git d22fa32), which I think is an experimental version with multi-threaded render. Linux is on 2017.06.03.nightly.

nophead, 2017-08-07:
I am on Windows with an Intel card but it has always been a feature of
OpenSCAD that the differenced objects need to fit in the view for me. I
always render my top level objects to avoid such issues.

donbright, 2017-08-07:
there is a similar issue back in  openscad 2014 release

 https://www.youtube.com/watch?v=a4PkXqfd6Po

however it is not as noticeable because the camera has to be "zoomed in" quite a bit before the problem occurs. 

it may be related to some camera settings that were changed between 2014 and now. I was watching the 'distance' number on the openscad window camera information area, and the problem seems to happen at around the same distance in the 2014 and current version. 

however the appearance of the object on screen is different, at the same distances.  these two screenshots show the issue. in the OpenSCAD-2014 at distance around 260 the object appears much closer than it does in OpenSCAD-current.  

[image]

[image]

in other words, maybe people didnt notice this problem as much in the past because people generally are more used to 'clipping bugs' when they are zoomed in extraordinarily close to an object... but when the object is far away they dont expect the clipping bugs.

donbright, 2017-08-07:
yes the camera settings and GL rendering has been completely changed since 2014... look at GLView.cc and Camera.h Camera.cc ... for example,  gluPerspective is used in the new openscad for ordinary interactive viewing, but it wasnt in the old openscad. also, glFrustum is not being used anymore.

lws-team, 2017-08-07:
From my noob point of view coming fresh to openscad, the only real object is the cylinder. I don't expect the interescting cube dimensions to ever matter, since it's just there to perform a logical operation on the cylinder.  Since it's part of a difference(), you should be able to clip the "chopping" cube to the bounding box of the cylinder and eliminate the excessive parts beforehand?

donbright, 2017-08-07:
Yes, that is the way it works in Render mode (F6) when 3d coordinates are calculated.

However OpenCSG preview mode is a bit more complicated as it is not calculating a complete set of 3d coordinates and bounding boxes, but rather making clever use of OpenGL features to quickly simulate a 3d appearance.  Some insight can be gained by switching to **ThrownTogether** viewing mode, which shows negative boxes as green/purple. In this mode it becomes more clear at what point the camera enters into the 'negative box' and the GL drawing gets confused.

kintel, 2017-08-08:
@lws-team @gadgetmind I think both of you are seeing a different issue than the one originally reported.

@lws-team: Your issue is due to a common technique called _view frustum culling_, where we'll clip the camera frustum against a front- and back clipping plane in order to optimize the limited z buffer bit depth. Large negative objects will definitely mess up your _preview_ rendering from angles where your camera is close to or inside a large negative object. This effect is also dependent on GPUs as different GPUs/drivers will implement different z buffer depths. Orthogonal mode is implemented differently than perspective mode so this effect will likely surface in a slightly different way.

@gadgetmind: You didn't post an example, but if you're seeing exactly the same effect as @lws-team, I suspect you're experiencing the same effect.

Unless people disagree, I feel like this issue should still be closed, and we should perhaps file a new issue concerning nicer behavior of large negative objects being clipped

nophead, 2017-08-08:
The fact it works fine in orthogonal mode seems like perspective mode could be fixed. Without looking at the code it seems the difference is faces are culled instead of clipped in perspective mode.

gadgetmind, 2017-08-08:
@kintel - I haven't created a simple test case, but I agree that I'm seeing the same as lws-team. Turning on orthogonal mode, reducing the size of the negative object, or moving away from the object, all cause the rendering to work correctly. And as @nophead says, that it works correctly in orthogonal (and on Windows for me, explain that!) suggested there is some low hanging fruit here.

Note that I use large negative objects as I'm "slicing" a large and complex object into multiple parts for printing (Fireflash on Thingiverse) and want to do this with cubes that defined planes with one part intersected and one differenced, so it's not just me being lazy.

gadgetmind, 2017-08-08:
@kintel - I haven't created a simple test case, but I agree that I'm seeing the same as lws-team. Turning on orthogonal mode, reducing the size of the negative object, or moving away from the object, all cause the rendering to work correctly. And as @nophead says, that it works correctly in orthogonal (and on Windows for me, explain that!) suggested there is some low hanging fruit here.

Note that I use large negative objects as I'm "slicing" a large and complex object into multiple parts for printing (Fireflash on Thingiverse) and want to do this with cubes that defined planes with one part intersected and one differenced, so it's not just me being lazy.

Oh, and apologies for leaping on someone else's bug.

thehans, 2019-01-28:
> The fact it works fine in orthogonal mode seems like perspective mode could be fixed. Without looking at the code it seems the difference is faces are culled instead of clipped in perspective mode.

Unfortunately its not that simple. 
When you define the camera in GL, you have to specify near and far clipping planes.  
The thing is that [glOrtho](https://www.khronos.org/registry/OpenGL-Refpages/gl2.1/xhtml/glOrtho.xml) has no problem with a zNear being negative, behind the viewer (so openscad uses zNear = -zFar), while [gluPerspective](https://www.khronos.org/registry/OpenGL-Refpages/gl2.1/xhtml/gluPerspective.xml) must have a positive zNear.  

For perspective it doesn't make sense to have clipping plane behind the viewer because it would go through the focal point.  Also if you attempt to set zNear too close to 0 then it really wreaks havoc on the depth buffer.

Relevant code is here:
https://github.com/openscad/openscad/blob/9b99a39e5c7b01a46fb360054b438d9ddcd8f1c5/src/GLView.cc#L91-L119

ciaranj, 2019-05-13:
Intel HD graphics chipset on Thinkpad.   Problem described above was exhibited on 2015.03-03.  Forcing goldfeather on fixed it for me, also updating to 2019.01-RC4 works (without forcing goldfeather)

cyhuvraak, 2019-08-30:
OpenSCAD-2019.05
[image]

[image]

nophead, 2019-08-30:
I don't get the result you do, I get shimmering where the two cylinders have coincident ends.  Making the second one overlap fixes it.

[image]

cyhuvraak, 2019-08-30:
I'm sorry guys, I got this issue because of low number of polygons. Difference works perfect.

thehans, 2020-06-04:
> ... I feel like this issue should still be closed, and we should perhaps file a new issue concerning nicer behavior of large negative objects being clipped

Just read through this again and agree its long overdue being closed.

As for the *separate* issue of large negative objects being clipped, see my previous comment... I don't believe it even possible to "fix" this aspect of perspective mode while using OpenCSG.

closed 2020-06-04 by thehans