openscad/openscad #1592: precision problem in FOR loop with small step values

mostbit, 2016-02-28:
The following example shows that the for loop has problems with the precision when stating small step values:

//Example:
for (i=[3:2:5]){
    echo(i);
}
//output:
ECHO: 3
ECHO: 5

Now, when changing the step value from "2" to ".2" one gets the following result 
(You will miss the last value "5" in the output below)

//Example:
for (i=[3:.2:5]){
    echo(i);
}
// output:
ECHO: 3
ECHO: 3.2
ECHO: 3.4
ECHO: 3.6
ECHO: 3.8
ECHO: 4
ECHO: 4.2
ECHO: 4.4
ECHO: 4.6
ECHO: 4.8

Hence, the last value "5" is missing in the output.

nophead, 2016-02-28:
That is just the reality of floating point. 0.2 is a recurring fraction in binary so it can't be represented exactly. Always use integers in for loops and scale down if needed e.g.

```
for (i = [30 : 2 : 50])
    echo(i / 10);

ECHO: 3
ECHO: 3.2
ECHO: 3.4
ECHO: 3.6
ECHO: 3.8
ECHO: 4
ECHO: 4.2
ECHO: 4.4
ECHO: 4.6
ECHO: 4.8
ECHO: 5
```

But also realise that two objects 0.2 units wide might not be the same as one 0.4mm wide.

doug-moen, 2016-02-28:
Since we use binary floating point, the only fractional values that can be
represented exactly are those with a power of 2 as the denominator. So, 0.2
can't be represented exactly, but 1/4 (0.25) works, and 1/8 (0.125) works.

This is a frequently asked question, and needs to be documented before this
is closed. I will update the language manual in the wiki.

doug-moen, 2016-02-28:
I ran some experiments, and there is a code change we can make so that ranges have better numerical behaviour.

I discovered that 0.2 * 15 == 3 exactly, but if I repeatedly add 0.2 to itself 15 times, then I get a number that is slightly larger than 3.

My suggestion is to precompute the number of elements in the range using multiplication and division. Internally, we'll iterate an integer loop variable i from 0 to len(range)-1, and for the i'th element of the range, we'll return start + i * step.

This change will cause [3:.2:5] to work correctly. It won't fix all numerical problems, but in floating point libraries, the goal is always to improve the numerical behaviour, not make it perfect. Whenever I've seen this problem posted on the forum, the failing step value has usually been 0.2, so this change should significantly reduce the frequency of users having numerical issues with ranges.

nophead, 2016-02-28:
If you make some special values work but there are still some that don't it will be more confusing. Better to educate the user say don't do it as it can't be relied upon to work. A non integer step should issue a warning, it throws an exception in Python. In fact they all have to be integers.
```
>>> range(3,5,0.2)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: range() integer step argument expected, got float.
```

doug-moen, 2016-02-28:
Yes, I am updating the wiki to recommend that step values like 0.2 should not be used.

My proposed code change will not create confusion, or make OpenSCAD less predictable or reliable. It will have the opposite effect. Right now, [0:0.2:2] works correctly, but [3:0.2:5] fails. That's pretty confusing, right? With my proposed fix, these will both work.

There is no downside to improving the correctness and reliability of the range code.

nophead, 2016-02-28:
I think there is. Simply floating point should not be used in ranges or loop indices. It is better to teach people this than to make it look like it works but then fail in more obscure circumstances or in other languages. Python teaches people not to use floats in ranges. If people learn that first they probably never expect it to work on OpenScad. As so many people make this mistake it seems OpenScad must be their first introduction to floats. An experienced programmer would never expect this to work.

doug-moen, 2016-02-28:
You are right that Python only supports integer ranges. However, that restriction isn't very backward compatible.

In OpenSCAD, step values like 0.5 and 0.25 work fine, and don't need to be avoided. In the updated language manual, I say: "You should avoid step values that cannot be represented exactly as binary floating point numbers. Integers are okay, as are fractional values whose numerator is a power of two. For example, 0.25 (1/4) and 0.125 (1/8) are safe, but 0.2 (2/10) should be avoided. The problem with these step values is that your range may have too many or too few elements, due to inexact arithmetic."

Haskell has always supported floating point ranges, and the Haskell equivalent of [3: 0.2: 5] works the way the original poster expects.

I just checked the Haskell implementation. Given [start:step:end], with Haskell semantics the list terminates when the elements become greater than end + step/2 for positive step, or when they become less than end + step∕2 for negative step. That means, if you don't know about floating point inexactness and didn't read the manual, then ranges behave they way you expect, in any "normal" case*.

*There are extreme cases where ranges can't work as expected, even if they are restricted to integers. These cases likely will never arise in practice. For example, this causes an infinite loop:
   big = pow(2,54);
   for (i=[big:big+1]) echo(i);

nophead, 2016-02-29:
The Haskell method will execute the correct number of iterations but the thing that might surprise the naive is the final value might not be quite the same as the one in the range.

I still think it is better to teach people not to use floats in ranges.

Your last example only doesn't work if 64 bit floats are used to represent integers, as in OpenScad. It would work with 64 bit ints, or 80 bit floats.

doug-moen, 2016-02-29:
If we use the Haskell method to compute the correct number of iterations, and if we avoid repeated addition (which is inaccurate) in favour of multiplication to compute each range element, then that will work pretty well, and will produce more accurate values than what Haskell produces. For example, [3:0.2:5] will produce a final value of exactly 5, whereas in Haskell the final value is slightly > 5.

We should also impose some restrictions on range parameters to avoid infinite loops and sequences of identical values due to numerical imprecision.

I think this approach does the best job of fixing bugs, meeting the expectations of naive users, and providing backward compatibility.

GilesBathgate, 2016-03-01:
I am pleased to find the OP example works in rapcad!

kintel, 2016-11-21:
@doug-moen Since you've been thinking about this: Do you have a hunch about what existing uses of fractional step values we might break if we make this change?

Ma-XX-oN, 2016-11-22:
The major problem is that the error is cumulative, which shouldn't be the case.

Anything that contains cumulative errors should be limited.  It would be far better, even though it might be slightly slower, to use division and multiplication when calculating the current iteration value and use the count to determine if the max iteration has been achieved, when either start, step or end are not integer values.  This will limit the cumulative error and result in less headaches.  Thus, internally, the count is an integer, but the value as seen by the user at each iteration is `start+step*count`.

doug-moen, 2016-11-22:
My implementation doesn't have cumulative error. It looks like this:

    double first, last, step;
    ...
    double delta = round((last - first)/step);
    double countd = delta < 0.0 ? 0.0 : delta + 1.0;
    if (countd < 1'000'000'000.0) {
        unsigned count = (unsigned) countd;
        for (unsigned i = 0; i < count; ++i)
            lb.push_back(Value{first + step*i});
    }

So I decided to sacrifice speed for accuracy. The interpreter overhead is
far greater than the loss in speed, so I doubt it matters.

kintel, 2016-11-22:
This would break:
`[1:2:6]`

-> Yields [1,3,5] today, will yield [1,3,5,7] with Haskell-style ranges out of the box.

In Haskell itself:
`[1, 3..6]  => [1,3,5]`
`[1.0, 3.0..6.0]  => [1.0,3.0,5.0,7.0]`

I expect this to be a relatively common construct due to our existing range implementation.

GilesBathgate, 2016-11-22:
I don't know if this helps but my approach in rapcad was to replace all floats with arbitrary precision numbers (Gmpq from CGAL). If you want to investigate this route you could try rapcad and see if the results/performance are desirable.

doug-moen, 2016-11-22:
@GilesBathgate wrote: I don't know if this helps but my approach in rapcad was to replace all floats with arbitrary precision numbers (Gmpq from CGAL).

I think that the Rapcad use of rational numbers is brilliant, and might be a good choice for OpenSCAD. I think it would eliminate some of the problems that people have with inexact floating point arithmetic, and given the slowness of the OpenSCAD interpreter, the performance difference should not matter.

However, for my next generation OpenSCAD project, I'm obsessed by performance, and I want to compile performance critical parts of a script into machine code. In that context, I decided that I'd have to standardize on 64 bit floating point arithmetic, and not follow Rapcad.

doug-moen, 2016-11-22:
@kintel wrote This would break `[1:2:6]` -> Yields `[1,3,5]` today, will yield `[1,3,5,7]` with Haskell-style ranges out of the box. ... I expect this to be a relatively common construct due to our existing range implementation.

Haskell-style ranges will cause a behaviour change that will fix bugs in some already deployed scripts, and break other scripts. If the number of broken scripts that are fixed outnumber the working scripts that are broken, then it is a good change.

For your example, you picked special values that demonstrate a worst case scenario. I'm not sure why you think this is a common construct. The last value in a range is normally the desired endpoint, not the desired endpoint + 0.5 * the step.

Thingiverse contains a lot of parameterized OpenSCAD scripts. Once a user starts plugging new parameters using the customizer, they may run into bugs due to floating point imprecision; the existing author may not have tested edge cases that users will try. So hypothetically, Haskell-style ranges could fix existing bugs on Thingiverse. I don't know how to test this, I don't have a large corpus of Thingiverse scripts to grep through.

kintel, 2016-11-22:
@doug-moen I'll look into it. I mentioned it since our test framework contains such a breaking range.

GilesBathgate, 2016-11-22:
@doug-moen Compiling performance critical parts of a script into machine code is an admirable goal. However my focus would be improving the performance of the geometry processing. If the performance of the geometry processing exceeds the performance of the current OpenScad script interpretation, then it would be worthwhile focusing on script compilation. IMHO

doug-moen, 2016-11-22:
@GilesBathgate Yes, that's the end goal of my project, to make it easier to
define new geometric primitives, and to make those primitives run faster.
So I'm going to move as much of the geometry processing into the OpenSCAD
language as I can, and make it run fast.

GilesBathgate, 2016-11-22:
@doug-moen I don't know what you mean by that. Currently the implementation turns the script results into an geometry tree so the pipeline is for example: Script -> AST -> interpretation -> CSG tree -> CGAL Geometry processing. I don't understand how you are proposing that the CGAL geometry processing be done as part of the script interpretation. Anyway I will be interested to see what you develop. If you are interested in the work in progress I did to improve Geometry processing its in my https://github.com/GilesBathgate/RapCAD/tree/MEPP-integration branch.

doug-moen, 2016-11-22:
@GilesBathgate: I didn't know about MEPP, thanks for the link.

The system I'm building looks superficially like OpenSCAD on the outside, but has an entirely different internal architecture. I want to define primitives like union() in the OpenSCAD-like language, then compile them into GPU code. My goal is that stuff renders really quickly, and it's much easier to implement new geometric primitives. So I'm not using CGAL. My system will initially be based on functional representation, and then I'll evolve from there once I have a working prototype.

Ma-XX-oN, 2016-11-22:
@kintel wrote: This would break [1:2:6] -> Yields [1,3,5] today, will yield [1,3,5,7] with Haskell-style ranges out of the box. ... I expect this to be a relatively common construct due to our existing range implementation.

@doug-moen wrote: Haskell-style ranges will cause a behaviour change that will fix bugs in some already deployed scripts, and break other scripts. If the number of broken scripts that are fixed outnumber the working scripts that are broken, then it is a good change.

That is a Haskell-style range result?  That is very counter intuitive.  I would never have expected the result to go beyond the last value specified.  I can see that creating many bugs, and potentially slow down rendering in my own stuff, as I try and not have duplicate object when doing a polar copy rotating or go beyond a limit when generating an array copy.

kintel, 2016-11-22:
@Ma-XX-oN Haskell determines the iteration count by using `round((end-start)/step)`, so it will overshoot with half a step as the worst case as that will be rounded up.

Ma-XX-oN, 2016-11-22:
@doug-moen 

> double first, last, step;
>     ...
>     double delta = round((last - first)/step);
>     double countd = delta < 0.0 ? 0.0 : delta + 1.0;
>     if (countd < 1'000'000'000.0) {
>         unsigned count = [...]
What is with the `countd < 1'000'000'000.0` and why are you generating a vector of items?  That seems very wasteful.

GilesBathgate, 2016-11-22:
@doug-moen Right I understand now. So you are going down a similar route to [ImplicitCad](http://www.implicitcad.org/).

GilesBathgate, 2016-11-22:
@doug-moen Also see: https://christopherolah.wordpress.com/2011/11/06/manipulation-of-implicit-functions-with-an-eye-on-cad/

Ma-XX-oN, 2016-11-22:
@kintel, very strange and still not intuitive.  There will have to be additional checks inside the loop further slowing down things.

I'd think that something like this would be better:

	double first, last, step, epsilon;
        ...
	double delta = (last - first) / step;
	double countd = delta < 0.0 ? 0.0 : delta;
	bool close_enough = abs((first + delta * step) - last) < epsilon
	unsigned int count = (unsigned int) countd - 1;
	for (unsigned i = 0; i < count; ++i) 
	{
		functor(i*delta);
	}
	if (close_enough)
	{
		functor(last);
	}

doug-moen, 2016-11-22:
@Ma-XX-oN:

In C++14, the `'` character is an optional digit separator in numerals. It makes long numerals easier to read by grouping the digits. One of my goals for this project is to learn C++14.

My approach to coding is to write the simplest code that works. First make it correct, then later make it fast. Once I get to the performance stage, I'll use profiling to determine where the real performance bottlenecks are. I'm not at that stage yet, so I haven't tried to optimize ranges.

I have implemented the "generalized range" proposal, #1583. In that context, it was easier for ranges to fill up a vector.

Python 2 works the same way: range(10) returns a list of 10 values. This didn't get optimized until Python 3. Haskell ranges also work the same way: [1..10] returns an ordinary list with 10 values. But Haskell ranges are as efficient as OpenSCAD ranges because lists are lazy, and elements are only generated on demand. So once I get to the performance stage, maybe I'll implement lazy lists. That would also optimize list comprehensions containing `for` loops. But I don't know yet.

doug-moen, 2016-11-22:
@GilesBathgate Thanks, yes I've read the ImplicitCAD source code, and it is definitely an inspiration. Although there are some things I'm planning to do differently. You can't define new implicit functions inside the scripting language. There's no modular way to add new implicit functions even inside the Haskell code: you have to modify a central algebraic type definition. There are bugs in the graphics kernel that can't be fixed without rewriting it from scratch. The STL generation algorithm is antiquated, slow, doesn't handle edges and vertices, and better choices are available today.

nophead, 2016-11-22:
How are edges handled correctly? Do you have to search for them using
numerical derivatives or something?

Ma-XX-oN, 2016-11-22:
@doug-moen ,

Yeah, sorry, that's not what I meant.  Why do you have an test for that expression?  Is there a reason that you would have a count delta that large and why would it be a problem if it was?

doug-moen, 2016-11-22:
@nophead: "How are edges handled correctly? Do you have to search for them using
numerical derivatives or something?"

That is my understanding, yes. I'm going to start by using the Dual Contouring algorithm, based on positive things that other people have said about it on the internet. Then, if necessary, I'll mine research papers for techniques to improve performance or eliminate artifacts.

doug-moen, 2016-11-22:
@Ma-XX-oN: it's easy to create test cases where the count goes all the way to infinity. Or it could just blow up to a float that is too big to fit into an integer, or too big to increment. (For a large enough integer in floating point, x+1.0 == x.) So I needed to cap the iteration count, and 1 billion was just an arbitrary choice. OpenSCAD hard codes a limit of 10,000, I thought it should be bigger. Probably should add a comment to the code.

doug-moen, 2016-12-19:
I found source code for the Matlab implementation of ranges.
https://gist.github.com/Juanlu001/7383894

Based on a code review, it appears to fix the bug without breaking backward compatibility as much as the Haskell implementation does. Integer and noninteger ranges are treated as separate cases, and floating point edge cases are treated with a lot more care than in my implementation. I've verified that [1:2:6] Yields [1,3,5], compatible with OpenSCAD, instead of [1,3,5,7] as the Haskell implementation does.

thehans, 2020-03-23:
I just submitted PR #3253 to fix this.  I'd had it in mind for quite some time and finally got around to implementing it.  It uses [`std::nextafter`](https://en.cppreference.com/w/cpp/numeric/math/nextafter) after division of `(end-begin)/step`  which accounts for the floating point imprecision.  The result gets cast to `uint32_t` and the individual iteration values are returned as `begin + i * step` rather than performing cumulative additions.

thehans, 2020-05-23:
Fixed by #3253

closed 2020-05-23 by thehans