openscad/openscad #1593: Export colors to .amf files.

terrylm, 2016-02-28:
As AMF files support color, and as OpenSCAD has partial support for color, and as 3D printing is about to move to multi-color (mixing nozzles), it would seem to make sense to export color values to the AMF files.

On the same note, for the last reason, it might also make sense for OpenSCAD to move to AMF as it's main file format, instead of STL?

bobc, 2016-04-11:
Hmm, deja vu again :)

Once #1608 is done, then multi-color AMF, STL etc becomes quite possible. 

There isn't really a "main format", the conversion to STL is the last step of export, it is not used as an internal representation. So all you could do is change the order of entries on the "Export" menu.

terrylm, 2016-04-19:
Interesting, I am glad to hear that last part. #1608 look quite long, so will read it another time, as I am quite tired...

kyrreaa, 2016-08-24:
Any progress on this?

jschobben, 2020-02-20:
Let me risk a bit of necromancy and shameless plugging:

Recently I put a script on github that allows exporting a given model as AMF/3MF, with colors preserved. Perhaps it can be helpful, until OpenSCAD can handle this natively.

It can be found here: https://github.com/jschobben/colorscad

xtemp09, 2022-12-23:
Any progress?

tpimh, 2023-06-15:
Also this can be beneficial for WRL export for the purpose described in #1304

t-paul, 2023-06-15:
Agreed that it would be nice for other formats. Considering the dead state of AMF, I'm going to close this specific issue. There's not much chance AMF support will get updates.

closed 2023-06-15 by t-paul